home-assistant/operating-system #1: YOCTO vs buildroot

rquackenbush, 2018-03-22:
I found your product via some mentions on the resin.os gitter chat. Am I guessing correctly that you abandoned resin.os and are now using buildroot instead?

The reason I ask is I'm trying to do something similar to what you were - I'm trying to run my own supervisor on a resin image and was hoping that you had some insight.

Any information you could provide would be helpful!

pvizeli, 2018-03-25:
Yes, we work actual with a fork of resinos but that is unmaintainable. Also resinos going into a direction they we can't or want support and we need a lot more hardware support.

Next, yocto is to distributed and make a project very customizable but also very complex and need a lot of abstraction and splitting in different repositories as buildroot does.

HassioOS is a replacement for our resinos fork with focus of better embedded support and does a lot things different as resin.

pvizeli, 2018-04-11:
You can help us and use our OS. We can run any supervisor without modification

rquackenbush, 2018-04-15:
Nice - I’ll take a look!

jerryaycock, 2018-04-15:
I have new Raspberry pi 3 B+ and would love to help. 
trying to figure where to start.