home-assistant/operating-system #2: EFI backend partitions by UUID

ejoerns, 2018-03-23:
Saw your question on #barebox about using EFI with barebox state.

A working example for using a fixed partition as storage backend for state I have is:

    state: state {
            magic = <0x3f2ba231>;
            backend-type = "raw";
            backend = <&state_part>;
            backend-stridesize = <4096>;
            #address-cells = <1>;
            #size-cells = <1>;

            [...]
     }

    partitions {
            compatible = "fixed-partitions";
            #address-cells = <1>;
            #size-cells = <1>;

            state_part: state {
                    partuuid = "14367da7-c518-499f-9aad-e1f366692363";
            };
    };

The partition in my case is crated with a script calling sfdisk:

    sfdisk -X gpt /dev/sdX << EOF
    start=2048,size=18432,type=C12A7328-F81F-11D2-BA4B-00A0C93EC93B,bootable
    start=20480,size=2048,uuid=14367da7-c518-499f-9aad-e1f366692363
    [...]
    EOF

pvizeli, 2018-03-25:
Thanks a lot. I have no a working base :)
I used sgdisk.

pvizeli, 2018-05-08:
Do you know, how I can pass the devicetree to linux kernel on EFI?

I work on:
https://github.com/home-assistant/hassio-os/tree/fix-rauc

ejoerns, 2018-05-09:
This is not really supported yet, mainline. But it might be a generic use case so we should think about mainline support, too.

Our current approach is to place the dtb in the EFI partition and point the barebox-state tool to this location.

ATM, this requires a RAUC patch similar to this: https://gist.github.com/ejoerns/13f9d8b594f6816fdca0b8a4d531fc7d

pvizeli, 2018-05-09:
Thanks. I will do it like:

On amd64 I patch the barebox host utility to use the /mnt/boot/efi/barebox/state.dtb (the hassio path). So it runs like other architecture and I can use the same rootfs.

pvizeli, 2018-05-09:
@ejoerns do you have also a patch for https://git.pengutronix.de/cgit/tools/dt-utils/tree/src/barebox-state.c ?

closed 2018-05-10 by pvizeli