home-assistant/operating-system #3: Roadmap for release

pvizeli, 2018-04-17:
*In progress*

# Tasks Supervisor
- [ ] Supervisor dbus support
- [ ] Supervisor systemd/NetworkManager support over dbus

# Tasks OS
- [ ] Cleanup filesystem
- [ ] Update OS/Version data
- [ ] Automatic Rauc OTA updates

# Supported Hardware on beta release
- [x] OVF/OVA
- [ ] Raspberry Pi
- [ ] Raspberry Pi 2
- [ ] Raspberry Pi 3/+

# Supported Hardware on stable release
- [ ] Orange Pi
- [ ] Pine
- [ ] Rock64

CookieMonster87, 2018-04-18:
Is support for Asus Tinker Board S also planned or considered? :)

bcran, 2018-06-15:
I just installed the 0.6 release on my new Raspberry Pi 3 B+ and it appears to be working - so it seems the "Supported Hardware on beta release" list needs updated :)

mories76, 2018-06-27:
I have a rock64 with emmc, so I can do some testing if needed.

closed 2018-07-01 by pvizeli