home-assistant/operating-system #28: Run on rpi 3 B+

danielsjf, 2018-06-11:
I've tried a process similarly to the Hass.io getting started procedure. I basically etch the image on my SD card.

When I do this, I don't see anything happening on the screen after booting my rpi. No splash screen nor code running.

When I wait longer I also don't see any IP coming online. Am I doing the right thing or is the process different?

danielsjf, 2018-06-11:
After some further testing I can add the following details. The 64 bit build just does nothing (black screen). The 32 bit build starts with scanning for ethernet. He doesn't find anything, then checks the four USB ports. Then quite quickly I see an endless loop of "fsm 1, hsts 00000001". It is like this other issue describes: https://www.mail-archive.com/arm@openbsd.org/msg01380.html.

I've tried to flash both versions twice but I can't get it to work. I also tried two ethernet cables, but none work.

cogneato, 2018-06-12:
@danielsjf I experienced the same with both pi3 builds.

pvizeli, 2018-06-12:
Fix with last dev

pvizeli, 2018-06-12:
I reopen this while the problem is with `saveenv` command. I workarroundit now, for development version is that okay.

I ask on mailing list, if some on know the problem.

pvizeli, 2018-06-12:
There is a patch to fix this: http://git.denx.de/?p=u-boot.git;a=blobdiff;f=drivers/mmc/bcm2835_sdhost.c;h=1ce019af579fb5b1e06bb04ac816733678bb51cf;hp=96428333b0abb2d28f3d845998fa2d44a70ce9ef;hb=79fd08f7456c7d12b04ef39e51d84d9981599c3a;hpb=d295c3ec3543e697b6f9f077f52877e081db4c6f

closed 2018-06-12 by pvizeli

niemyjski, 2019-03-11:
I just ran into this exact issue this week. I think I was running hassos 2.0 or 2.1 on 0.88.2. It just wouldn't boot and I was getting this issue. I was also using the evo high endurance drive on a pi 3b+.. I can see all the partitions and everything is valid..