home-assistant/operating-system #32: OVA file not compable with QNAP Virtualization

rudders, 2018-06-17:


rudders, 2018-06-17:
The QNAP site states

> Virtualization Station supports importing and exporting VMs. The *.ova, *.ovf and *.vmx formats are supported by VMs exported from VirtualBox and VMWare; while the *.ovf and *.qvm formats are supported to export VMs for other Hypervisors or the Virtualization Station on other QNAP NAS. Virtualization Station also supports the virtual appliances which are provided by on-line VM markets such as Bitnami and VMware Virtual App.

https://www.qnap.com/en/how-to/tutorial/article/how-to-import-and-export-virtual-machines-vm-tofrom-virtualization-station-including-importing-virtual-machines-from-on-line-vm-markets

rudders, 2018-06-19:
QNAP support provided the following ....

`Notice: The *.ova and *.ovf formats are only supported by VMs (virtual machine) exported from VirtualBox v4 and VMware ESXi 5.0`

rudders, 2018-06-27:
RC1 installs on QNAP QVS but stops at “Booting from disk...” will try and get some further debug logs and update.

pvizeli, 2018-06-27:
https://github.com/home-assistant/hassos/pull/46

pvizeli, 2018-06-27:
It is up now

rudders, 2018-06-27:
Thanks. QVS doesn't support vmdk anyway it appears.
I’m having issues on VMware Fusion also and expect the vmdk will work for it so ill try and export a vmx that works for QVS from there for now.

This is not a critical issue for me btw. Just seemed like a good way to exploit Hassio on a NAS rather than my Rpi - micro SD for storage makes me very nervous.

Thanks for your efforts. I'll feedback my success or failure with Fusion.

OverloadUT, 2018-07-12:
@rudders Did you end up having any luck getting the VMDK in to a format that QNAP is happy with? I was trying to do this today and haven't figure out how to go about it.

rudders, 2018-07-12:
Unfortunately not @OverloadUT ... QVS doesn't like VMDK at all and the suggestions for creating an ISO I found looked like a great way to waste more time..... 

After a number of false starts on VirtualBox and VMWare on my Mac I kind of gave up on this route installed a Debian VM on QVS and used [this method to install hass.io](https://github.com/home-assistant/hassio-build/tree/master/install#install-hassio). Which has worked but I don't have the hassio command at the moment. It's enough for me to see how the NAS perform's with Shinobi and a couple of other add-ons I'd like to use that need a bit more power.

I'm kind of waiting for hassos to stabilise (now it's released for instance) to circle back and try again.

OverloadUT, 2018-07-12:
Okay, sounds like I did the same initial steps you did, short of doing the Linux install method. I figure I can just wait for HassOS to get a working OVA version and try again then.

rudders, 2018-07-12:
Yeah - although I wouldn't hold your breath - QVS seems to have some idiosyncracies from what I can pick up on the other forums.... I'll probably circle back to this in a week or so and give it a whirl and see how I go.

Mulders84, 2018-10-04:
Rudders i got it working on Virtualisation station. Get the newest Hassos, get it setup in virtual box(2cpu, 2 gig and so on) Don't start it.

In Virtual box:

1. change VMDK to VDI under File > virtual media manager.
2. Delete VMDK and apply VDI to the Virtual machine
3. check correct vm settings : 
-  System Motherboard > uncheck floppy > chipset change to ICH9, check I/O-apic, EFI, Hardware UTC
4. Export the VM

5. Import to Virtualisation station Don't Start VM!! (fucked it up several times)
6.  VM settings,
- General >Intel Core i7 (Hehalem.....), 2 cores, 2 gig, OS Type Generic, Version Generic
- Network > VirtIO 
-  Storage> 6 GB, Write Back, IDE !!
- CD/DVD>  deleted
- Audio> Disabled
- USB 2.0 > Enable

Start VM and youre done.. 
Good luck...

rudders, 2018-10-04:
That's awesome. Well done. I moved on from the NAS as the host and shelled out for a NUC running Proxmox in the end, moved my PLEX etc there too as the NAS CPU kind of kept limiting its usefulness. But as a HA Host it might still be a good option.. thanks for passing on your wisdom. Hopefully will help others.

Mulders84, 2018-10-08:
Aah that's nice!! I read up on a Nuc a while ago and even thought about buying a simular solution like you but eventually I decided to spend the money on a NAS and upgrade it instead of buying another device to do maintenance and update/grade. So.. 
Got a TVS- 682, upgraded to i7 6700, Noctua cooler l9 65mm, 64 GB, Corsair 450 watt power supply, 1050 nvidia card and some noctua fans. This all took a lot of time and money(waited for the sales/discounts and got lucky).

Personally i like VM's more because of the backup and flexibility waited a while but they did not offer a solution but there it is. 
Still have to work on my HA programming skills but it's fun.. Goodluck and have fun.

ill-tempered, 2018-10-21:
> Rudders i got it working on Virtualisation station. Get the newest Hassos, get it setup in virtual box(2cpu, 2 gig and so on) Don't start it.
> 
> In Virtual box:
> 
> 1. change VMDK to VDI under File > virtual media manager.
> 2. Delete VMDK and apply VDI to the Virtual machine
> 3. check correct vm settings :
> 
> * System Motherboard > uncheck floppy > chipset change to ICH9, check I/O-apic, EFI, Hardware UTC
> 
> 1. Export the VM
> 2. Import to Virtualisation station Don't Start VM!! (fucked it up several times)
> 3. VM settings,
> 
> * General >Intel Core i7 (Hehalem.....), 2 cores, 2 gig, OS Type Generic, Version Generic
> * Network > VirtIO
> * Storage> 6 GB, Write Back, IDE !!
> * CD/DVD>  deleted
> * Audio> Disabled
> * USB 2.0 > Enable
> * others> check Enable VirtIO serials
> 
> Start VM and youre done..
> Good luck...

Just saying this worked perfectly on my TS-453A!

brubbybu, 2019-02-10:
> Rudders i got it working on Virtualisation station. Get the newest Hassos, get it setup in virtual box(2cpu, 2 gig and so on) Don't start it.
>
> In Virtual box:
>
> 1. change VMDK to VDI under File > [...]

Thanks a lot!! Perfectly working!

Mulders84, 2019-03-10:
Hi all 
a little update on hassio>qnap in virtualisation station
Did a new install for a friend but it was not stable for him so i went back to square 1
it is all pretty much te same but addeded (*) for the new VDI file config and found out it was stable.
if you download the VDI file start at 2... look for the (*)  and you will be fine. 
** a little note if you have problems importing to virtualisation station use virtbox 5.2

> i got it working on Virtualisation station. Get the newest Hassos, get it setup in virtual box(2cpu, 2 gig and so on) Don't start it. (* download VDI file from hassio)
> 
> In Virtual box:
> 
> 1. change VMDK to VDI under File > virtual media manager.(* skip with vdi file)
> 2. Delete VMDK and apply VDI to the Virtual machine(* apply vdi file)
> 3. check correct vm settings :
> 
> * System Motherboard > uncheck floppy > chipset change to ICH9 (*you can leave it at piix), check I/O-apic, EFI, Hardware UTC
> 
> 1. Export the VM
> 2. Import to Virtualisation station Don't Start VM!! (fucked it up several times) (* use virtbox version 5.2 if you have problems.)
> 3. VM settings,
> 
> * General >Intel Core i7 (Hehalem.....), 2 cores, 2 gig, OS Type Generic, Version Generic
> * Network > VirtIO
> * Storage> 6 GB, Write Back, IDE !!(* you can set it at VirtIO and give 32 GB to it)
> * CD/DVD>  deleted
> * Audio> Disabled
> * USB 2.0 > Enable
> * others> check Enable VirtIO serials
> 
> Start VM and youre done..
> Good luck...

If you have this running  in Virtualisation station remeber to take snapshots or make  backups
Have Fun

closed 2019-05-16 by stale[bot]

platini76, 2019-09-06:
I have booting from Hard disk fixed... not boot... someone now how solve?