home-assistant/operating-system #33: Hardware Donation

pvizeli, 2018-06-19:
To support more device, we need more Hardware. In live in switzerland and our prise and duty are high.
If you have a device they you don't need anymore and will send us or you want donate a small amount for a device, would be nice. Feel free to extend this list of hardware.

You can send donation to my paypal Account and add a descritpion with the name of you favorite device.

## Hardware list
- ASUS Tinker Board: 70$ - 0%
- Pine64: 30$ - 0%
- Rock64: 80$ - 0%
- Intel nuc: 130$ - 0%

Paypal: pascal@vizeli.ch
For post address, send me a PM on discord.

ciotlosm, 2018-06-19:
Any chance to add some Orange Pi models as well?

pvizeli, 2018-06-19:
What kind of OrangePi make sense?

ciotlosm, 2018-06-19:
Probably Orange Pi Zero and Orange Pi PC 2. One is a H3 and one is H5. I have the Orange Pi PC 2 but don't know what is currently most popular, as most custom builds used Orange Pi Zero

danielsjf, 2018-06-19:
You should have a Patreon account!

CountParadox, 2018-06-19:
I would support a homeassistsnt patreon account tbh

dzungpv, 2018-06-20:
You can add a donate button on the home page, and anyone can send you optional 5, 10, 15, 20 ... until you can purchase all the hardware you can simple close, or just let it there. I will donate 18$ for Orange Pi Zero

ciotlosm, 2018-06-20:
If anyone wants to Donate using Revolut, you can ping me and then I can proxy them to pvizeli as well.

pvizeli, 2018-06-20:
patreon account is in progress

bboti86, 2018-06-20:
Might be far fetched but have you considered one of these: 
https://getchip.com/pages/chippro#undefined4

pvizeli, 2018-06-20:
https://www.patreon.com/pvizeli

rudders, 2018-06-20:
I’ve Patreoned $5/mnth and will let it roll for 12 months at least so use that as you choose.

CountParadox, 2018-06-20:
I patreon $10 a month, please continue your hard Dev work ❤️

I have run into a Hass.io but with USB serial devices now :( I'm hoping
it's fixed in a future version

pvizeli, 2018-06-23:
@CountParadox make a PR. I run devices over GPIO and usb/serial without problems...

olskar, 2018-07-02:
@pvizeli Will you need testing? I own a Rock64 and can test if needed :)

pvizeli, 2018-07-02:
@olskar I order now a rock64. I need no tester ;) We need porting the bootloader to our system...

bboti86, 2018-07-08:
@pvizeli: would one of these help with serial debugging: http://www.ni.com/hu-hu/shop/engineering-education/portable-student-devices/myrio-student-embedded-device/what-is-myrio.html ? 
It has an FPGA that could read serial transmission in real time and you could process that on your PC. 
Let me know

jlbphotos, 2018-07-13:
Donated 68 dollaroos to the NUC
Will add to next month if i can as well, thanks for all the hard work.

okueng, 2018-07-15:
@pvizeli  dontated 40CHF for the intel nuc via paypal!

mad-ady, 2018-07-16:
Odroid HC2 runs the exact same software as the XU4, so if it runs on XU4 it will run on HC2 as well. I have both so I can test them if needed.

pvizeli, 2018-07-16:
@mad-ady they have also a SD slot for OS?

mad-ady, 2018-07-16:
XU4 supports both SD or eMMC (has a selector switch). HC1/HC2 (and MC1) have only SD slot (and HC1/HC2 have SATA for a hard drive). Note that it still needs the SD card to boot (boodloader can't be on SATA), but rootfs can be moved to disk if needed. Not sure if uboot can read kernel/initrd/dtb from SATA - all the examples I've seen do it from the SD and after kernel boots it can mount the rootfs from SATA.

jlbphotos, 2018-07-16:
Got paid...just paid 26 Euro for the NUC hopefully that's 100% now

thanks

frog32, 2018-07-17:
Sent CHF 50 your way. Use it for whatever you need besides just boards (SD cards, ethernet cables, power adapters).

ciotlosm, 2018-07-17:
@pvizeli do you need anything extra to help with automation of deployments/testing?

gotodragon, 2018-07-17:
@pvizeli I test hassos ver 1.7, it is same with ver 1.5, no data transfer.

dzungpv, 2018-08-21:
BeagleBone Black is the board often use by professional,with fast emmc flash, any change you will make hassos work with it?

ciotlosm, 2018-08-21:
@dzungpv if you start some donation for it probably will get added to the list so others can help support it.

closed 2019-04-23 by stale[bot]