home-assistant/operating-system #52: Bluetooth BCM43xx fail

jasimancas, 2018-06-28:
I have installed the addon Bluetooth BCM43xx, but the devices dont report.

The log of the add-on:

> bcm43xx_init
Flash firmware /lib/firmware/brcm/BCM43430A1.hcd
Set BDADDR UART: b8:27:eb:60:46:d0
Set Controller UART speed to 921600 bit/s
Device setup complete

I have https://www.home-assistant.io/components/sensor.mitemp_bt/ and https://www.home-assistant.io/components/sensor.miflora/

tyborall, 2018-06-28:
I think the Pi 3 B(+?) hast another BCM5xxx.

jasimancas, 2018-06-28:
Is the 3B, not +, in hassio with ResinOS (old version) work fine, but in hassos (new version) not. Same files in Config folder and same add-ons install.

pvizeli, 2018-06-28:
Do you use the same `config.txt`?

jasimancas, 2018-06-28:
The same config folder (home assistant).

pvizeli, 2018-06-29:
Can you look into `config.txt` on boot partition of hassos and resinos and look if you change there something?

If that the same, can you copy the start.elf, fix-up.bin and bootcode.bin to hassos? Maybe the is a bug RPI firmware.

jasimancas, 2018-06-29:
Ok, look this:
Hassio:
[image]
Hassos:
[image]

pvizeli, 2018-06-30:
The mac address is strange

jasimancas, 2018-06-30:
Yes, maybe this is the problem.

danielsjf, 2018-07-01:
If it helps anything, I tried the same on my raspberry pi 3 model B+. The ip also ends in aa, but also the rest is exactly the same.

pvizeli, 2018-07-02:
I move that to add-on repository and try to fix the add-on

pvizeli, 2018-07-02:
/move home-assistant/hassio-addons

pvizeli, 2018-07-02:
I'm worng: https://www.raspberrypi.org/forums/viewtopic.php?t=183474

Is that only with 64bit or also with 32bit a problem?

jasimancas, 2018-07-02:
Both, 32 and 64

pvizeli, 2018-07-02:
I try to extract the correct serial number from device tree

pvizeli, 2018-07-02:
Okay, I provide the bugfix they was describe there in HassOS 1.2

pvizeli, 2018-07-02:
https://github.com/home-assistant/hassos/issues/62

Will be release today

pvizeli, 2018-07-02:
I was right, the problem is on add-on. On new linux version they need read the device-tree for serialnumber and not /proc/cpu. I fix that now on add-on

PR #91 (2018-07-09): Native bluetooth support
Fix: #52
closed 2018-07-02 by move[bot]