home-assistant/operating-system #72: After restart, the IP doesn't come online anymore

danielsjf, 2018-07-02:
This has probably to do with the container renaming. So I wait for the next version, but how can it get the OTA update if the container is not starting anymore?

Should I update it over USB? If so, how? Just drop the raucb file one one and plus it in before rebooting? Couldn't find instructions on this.

tyborall, 2018-07-02:
You mean it is not coming up after updating from 1.0 to 1.1 via Hassio panel? I had no problems with the update yesterday. But the next start took some time.

pvizeli, 2018-07-03:
If you device run with DHCP, it will get a new IP address. Please set a static IP or make a reservation on DHCP Server.

We "fix" that with build 1.3, that he can remember the old IP but that not mean that the DHCP Server will accept the old address.

pvizeli, 2018-07-03:
Is written to: #62

PR #75 (2018-07-05): Use dhcp client / avahi after NM
Fix: #72
closed 2018-07-03 by pvizeli