home-assistant/operating-system #74: Dev SSH Key Authentication on HassOS 1.2 not working

tyborall, 2018-07-03:
With HassOS 1.2 release I'm able to deploy authorized_keys file with USB to HassOS. After reboot SSH accepts basic connection but denies key authentication after typing username "root". I use PuTTY. The authorized_keys file is correctly formatted. With the same authorized_keys file on another Linux machine it is working, so it seems there is still some error.
I tried twice even with generating new Pubkey/Privkey.

pvizeli, 2018-07-03:
can send output of `journalctl -u dropbear.service`

tyborall, 2018-07-03:
I cannot access the instance via SSH as the key is not working :-)
Is there another way to get access to the logs?

pvizeli, 2018-07-03:
Monitor (HDMI) + keyboard or with a serial connector

tyborall, 2018-07-03:
I found out that it has to do with formatting of the file if created under Windows. But even with correctly formatted authorized_keys file I get error message when connecting as root. Dropbear SSH is started and shows no error.

tyborall, 2018-07-03:
I now tried 1024 bit key, 2048 bit key and 4096 bit key. I always checked if authorized_keys file was correctly copied to /root/.ssh/authorized_keys. I configured PuTTY every time with correct Privkey. But it is always saying "Server refused our key".

pvizeli, 2018-07-03:
output of: `journalctl -u dropbear`?

tyborall, 2018-07-03:
```
-- Logs begin at Tue 2018-07-03 14:36:30 UTC, end at Tue 2018-07-03 15:03:02 UTC. --
-- No entries --
```

tyborall, 2018-07-03:
```
-- Logs begin at Tue 2018-07-03 14:43:55 UTC, end at Tue 2018-07-03 15:06:47 UTC. --
Jul 03 15:05:37 hassio dropbear[458]: [4003] Jul 03 15:05:37 Child connection from 192.168.1.131:57727
Jul 03 15:05:37 hassio dropbear[458]: [4003] Jul 03 15:05:37 User 'root' has invalid shell, rejected
Jul 03 15:05:37 hassio dropbear[458]: [4003] Jul 03 15:05:37 Exit before auth (user 'root', 0 fails): Disconnect received

```

tyborall, 2018-07-03:
/usr/sbin/hassos-cli is not in /etc/shells. Only /bin/sh is.

pvizeli, 2018-07-04:
Fix in 1.3

closed 2018-07-04 by pvizeli

tyborall, 2018-07-19:
Is this really fixed? It's still not working for me :-(

pvizeli, 2018-07-19:
Yes, I use it actual

PR #78 (2018-07-05): Fix /usr/sbin/hassos-cli is not in /etc/shells
Fix: #74