home-assistant/operating-system #79: Timezone settings in hassos

magnusoverli, 2018-07-05:
Hi!

I am having an issue where I cannot get the correct time in my logs. The hass.io log can have this event:
`18-07-05 09:30:48 INFO (MainThread) [hassio.api.proxy] /homeassistant/api/states access from a0d7b954_appdaemon3`, but time was actually `11:30:48`. There is a two hour offset...
I have set `time_zone: Europe/Oslo` in `configuration.yaml`.
Do we have access to set the time for supervisor in another way?

pvizeli, 2018-07-05:
Supervisor and OS run in UTC by designe. All other they run on top of this are in your local time zone. Maybe we change the logging time in future but actual is nothing on roadmap.

kmcelroy, 2018-07-22:
This is throwing off all my time-based automations in Node-Red. There has to be a workaround (other than the tedious calculation of adding 7 hours to every time I want to filter)

tyborall, 2018-08-09:
@pvizeli Why not updating the timezone (automatically) from Home Assistant on the host system? Is that not possible?