home-assistant/operating-system #89: Roadmap dev-2 / rel-3

pvizeli, 2018-07-08:
## Features OS
- Allow to use data partition on a USB drive

ciotlosm, 2018-08-11:
Would be nice to have the ability to write the os on a usb device and have the ability for the rpi3 to boot from there

rogersmj, 2019-01-11:
Any updates? I just want to hang a USB hard drive on the Pi and share it via Samba as a basic networked file share.

Teri928, 2019-01-13:
I'm interested in this feature as well.

closed 2019-04-23 by stale[bot]

esciara, 2019-06-26:
Can someone confirm that this feature is now available when following the steps described in https://github.com/home-assistant/hassos/issues/164#issuecomment-472788621, following the mentioned work done on #318 ?

satreddys, 2020-02-28:
It's 2020 and this feature is not available yet. It would be better to use SD card only for boot drive and remaining partitions from external drive for OS longevity when compared with SD card life.

austwhite, 2020-06-03:
Looks like this is closed as wont fix...... Guess it is not happening

dhebard, 2020-12-10:
We really need this ability.

austwhite, 2020-12-10:
We might need to open a new feature request for this one.
It was closed in April 2019 as there were no updates and no responses after the automatic system flagged it as stale.
Maybe someone can open a new request for this feature?

dhebard, 2020-12-11:
Yes please!