home-assistant/operating-system #93: Cannot update to 1.4 from VMware ESXi hassos 1.3

mihalski, 2018-07-10:
VMware ESXi 6.5. Could not use the vmdk file so someone running the same version of ESXi who did not have this problem sent me an OVF/OVA export. This also would not import to ESXi so I loaded it in VMware Fusion and it worked. Then I exported that to an OVA and it DID import into ESXi. Worked fine but upon reboot Autoboot doesn't work and either Boot System 0 or Boot System 1 needs to be manually selected. This remains until now.

[image]

System 0 is 1.2 and System 1 is 1.3 as one time the Update button in the Host system panel successfully upgraded (through I needed to choose the OTHER Boot System than that I was currently booted into for it to start).

Now that 1.4 is available I can try to update from either the 1.2 or 1.3 version but it doesn't work either time. I have the following to provide as clues.

The Hass.io system log:
```
18-07-10 12:05:22 INFO (MainThread) [hassio.tasks] All core tasks are scheduled
18-07-10 12:05:22 INFO (MainThread) [hassio.core] Hass.io is up and running
18-07-10 12:07:19 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/1.4/hassos_ova-1.4.raucb
18-07-10 12:09:14 INFO (MainThread) [hassio.hassos] OTA update is downloaded on /data/tmp/hassos-1.4.raucb
18-07-10 12:09:14 INFO (MainThread) [hassio.utils.gdbus] Call de.pengutronix.rauc.Installer.Install on /
18-07-10 12:09:14 INFO (MainThread) [hassio.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
```
The home-assistant.log:
```
2018-07-10 22:05:06 WARNING (MainThread) [homeassistant.config] Incomplete core configuration. Auto detected elevation: 0
2018-07-10 22:05:08 WARNING (MainThread) [homeassistant.components.http] You have been advised to set http.api_password.
2018-07-10 22:12:20 ERROR (MainThread) [homeassistant.components.hassio.http] Client timeout error on API request hassos/u
pdate
```
And this:
[image]

The IMPORT FROM USB option did not work for either the update or authorized_keys as far as I could tell. But that's a secondary issue for me at the moment. I'd just like to figure out why OTA updates aren't working and why Autoboot isn't working.

An explanation of how the update process and the dual booting functions would be helpful to understanding how things SHOULD work.

Thanks in advance. I'm sure I missed some things but will be happy to provide any further information to help fix these problems.

Regards,
Michal

pvizeli, 2018-07-10:
Yes, if you boot with this menu, OTA updates want work anymore. I will fix that

closed 2018-07-10 by pvizeli

pvizeli, 2018-07-10:
The SSH key work first with 1.3 correctly.

You can install the update manual with:
```
systemctl start mnt-config.mount
cd /mnt/config
rauc install xy.raucb
```

pvizeli, 2018-07-10:
You still see only:
`18-07-10 12:09:14 INFO (MainThread) [hassio.utils.gdbus] Start dbus monitor on de.pengutronix.rauc`
as last line?

No error or other abords? They should normaly report that the update going wrong with a error message. Look like there is a bug on hass.io / will fix that later.

pvizeli, 2018-07-10:
/move home-assistant/hassio

PR #94 (2018-07-10): Fix boot menu
Fix: #93