home-assistant/operating-system #97: VMDK on Proxmox - Workiong

rbflurry, 2018-07-12:
I was able to get the VMDK to boot on Proxmox. Is this an ok place to document how?

pvizeli, 2018-07-12:
Yes please :)

rbflurry, 2018-07-12:
This is what the VM conf looks like for the VM

Contents of 100.conf
```
bios: ovmf
boot: c
bootdisk: ide0
cores: 1
efidisk0: san:100/vm-100-disk-3.qcow2,size=128K
ide0: san:100/vm-100-disk-2.vmdk,size=6G
memory: 1024
name: hass.io
net0: e1000=B6:ED:7D:9E:57:6F,bridge=vmbr0
numa: 0
ostype: l26
smbios1: uuid=8c7b4639-580f-47ef-b91b-27520a663162
sockets: 2
```

rbflurry, 2018-07-12:
Hmm. However i cant get any of the hassos commands to work.

[image]

pvizeli, 2018-07-12:
Looks like the cli don't read the hosts file. Need check this

pvizeli, 2018-07-12:
Do you have any custom network settings?

rbflurry, 2018-07-12:
No this was a basic install right off the bat.  DHCP address no other changes.

rbflurry, 2018-07-12:
Should hassio be listening for commands on port 80? or 8123? For example from my laptop I cannot access http://192.168.47.147/hassio/snapshots/reload either

pvizeli, 2018-07-13:
Yes on 8123. You can access to supervisor api with home assistant as ...:8123/api/hassio/snapshot/reload

dale3h, 2018-07-13:
I was able to get the VMDK to boot as an unRAID VM, but I cannot get the networking to work. This seems like a very similar issue as you're experiencing with Proxmox. @rbflurry Can you try to type `login` at the `hassio >` prompt, and then once at the shell prompt, run:
```shell
$ nmcli d
```
Please paste the output here.

Also, run this and see if you're getting a 500 error for pulling the landingpage:
```shell
$ docker logs hassos_supervisor
```

snozzlebert, 2018-07-14:
@dale3h What kind of network adapter type did you use?
I had similar networking issues in my HassOS Proxmox VM. It looks like the virtio network adapter is not supported (missing virtio net driver?). Running HassOS in a VM with the **e1000** network adapter does work in Proxmox.

rbflurry, 2018-07-15:
@dale3h Im using the intel NIC not virtio. I can confirm when using a Virtio network adapter I cannot access the VM.

However when using the Intel Network adapter the VM gets an ipaddress which I can access via a browser. But commands from the console do not work.

cogneato, 2018-07-15:
I just got 1.6 VMDK to work in unraid by editing the bridge interface to 'e1000' instead of 'virtio'. I setup and I am currently looking at the Welcome Home! card.

cogneato, 2018-07-15:
@rbflurry 

> But commands from the console do not work.

I would not expect them to work from the host? I may be wrong but I don't think the custom cli is available but through the ssh addon choices (core and community repos). I installed the ssh addon and the commands work as they should.

pvizeli, 2018-07-18:
https://github.com/home-assistant/hassos/pull/118

pvizeli, 2018-07-18:
Version 1.8 for OVA is in progress with support the vritio drivers.

Can anyone update the `Documentation/boards/ova.md` and update the supported hypervisor + a small howto if one is needed?

mgolisch, 2018-07-21:
yeah virtio  network works fine for me with the 1.8 image in proxmox

nickrout, 2018-07-22:
I came here looking for answers as to why the 1.8 vmdk didn't work in virtualbox. 

I needed to set the network device to "Intel PRO/1000 MT Desktop" before it would work. I haven't tried virtio, but the virtualbox default on PCnet-PCI II didn't work. This accords with the comments above about needing to use e1000, so thanks for the hint.

Hope that helps someone.

manipat, 2018-07-27:
Tried to get this working on Proxmox 5.1, used similar settings as the OP, except I used a local-lvm and the qm importdisk command (could this be an issue?)  For some reason, I'm stuck at the boot menu, I saw people recommended removing the CD drive, didn't help. the only option that works is the boot shell.

user-ks, 2018-07-31:
Read many good things about Home Assistant and hass.io,
and that was the reason that I invested in a home "virtualization lab" to run it.

**This is a brief overview of the setup:**
Intel NUC7i5DNHE, i5 with 04 cores
Proxmox VE v5.2-5 running in clustering mode (with 01 test node now)
      [Proxmox is running fine and Univention Corporate Server is running.
      Note for other interested users:
      Proxmox VE = quite a steep learning curve for non Linux pros :-(  )

As far as I understand, hassos_ova-1.8.vmdk now would inlcude working VirtIO drivers.
But I have the same problem as

user-ks, 2018-07-31:
@rbflurry 
You posted following file 100.conf earlier.

`bios: ovmf
boot: c
bootdisk: ide0
cores: 1
efidisk0: san:100/vm-100-disk-3.qcow2,size=128K
ide0: san:100/vm-100-disk-2.vmdk,size=6G
memory: 1024
name: hass.io
net0: e1000=B6:ED:7D:9E:57:6F,bridge=vmbr0
numa: 0
ostype: l26
smbios1: uuid=8c7b4639-580f-47ef-b91b-27520a663162
sockets: 2`

Would you be so kind to elaborate and document a bit more the precise steps that are needed on Proxmox to generate this succesful (?) config.
And can you please explain a little the config itself?

bagobones, 2018-08-25:
It probably doesn’t have ACPI support because raspberry pi boards don’t. 

The hassos idea is great to go bare bones but the team needs to figure out a bunch of additional stuff to add back in if their target is going to be more than raspberry pi.

I suspect they will end up with a lite and full image because there is a big difference in capabilies between PI’s and NUCs/VMs

jtru, 2018-10-04:
I got the newest OVA vmdk image to boot under vanilla qemu yesterday, and will provide a detailed guide on how to do this at some point in time.

The lack of proper support for ACPI shutdown events is a big problem for hypervisors, imo. I haven't reviewed the kernel config yet (basic ACPI support seems to be hinted at in `/proc/`), but the image's userspace includes a copy of `acpid`, which _could_ be used to implement this - it is broken as of 1.11 because it requires configuration to exist at the hard-coded pathname of `/etc/acpi/`, which the image does not provide. This should be rather easy to fix.

I was wondering if there's a particular reason why the image does not include systemd-logind, because that piece of systemd infra could also handle shutdown via ACPI.

Do any of the targeted ARM platforms support ACPI, by the way?

jtru, 2018-10-04:
I did some research this evening, and it turns out that in order to use busybox's ACPI shutdown support, most of the required tricky parts are already in place: busybox does not necessarily rely on `/proc/acpi/event` (support for which the kernel config lacks), but can receive the PWRF event via either netlink or the input layer, which the hassos kernel or the *OVA* board delivers when qemu gets `system_powerdown` on its monitor interface.

Now, what's still missing to get this to work as intended are the following things:

1. a proper configuration directory
2. a service unit to start it automatically
3. a helper script to perform the shutdown, located in the config dir

I opened a branch in my fork of this repo that's supposed to implement these via commit https://github.com/jtru/hassos/commit/29325b8ec45630d90f3392fa6526b9673909ead3 - *however* my changeset is completely untested, since I'm not familiar with the bootstrapping process for the required build environment that is required to bake new hassos OVA images. If @pvizeli could please provide docs on how to do that (an informal transcript of a shell session, where the end result is a working build environment, would suffice).

Other than that, I'd appreciate if you could let me know if my proposed changes look sensible :)

jtru, 2018-10-08:
FTR: I managed to build the OVA image and test my changes, with the result looking promising enough to risk a pull request ;)

codertux, 2018-10-10:
@jtru I also want to make some changes to make it easier to deploy on a headless KVM host, but I'm having trouble building it. If you figured it out, can you update the documentation or share some basic steps here?

jtru, 2018-10-10:
@codertux I sat down this evening and wrote a quick guide on how I got started with modifying hassos - you can find it at https://github.com/jtru/hassos/blob/getting-started-with-devel-guide/Documentation/getting_started_development.md

If @pvizeli finds it any good, I'll be happy to create a pull request to include it in the project.

codertux, 2018-10-11:
@jtru that's very much. That was basically what I was doing but apparently building in docker on MacOS fails. Wasted about a day on that. When I built it on a ubuntu system it worked like a charm!

jtru, 2018-11-19:
@etique57: That piece of documentation was merged thanks to @pvizeli and is available at https://github.com/home-assistant/hassos/blob/dev/Documentation/getting_started_development.md - it does not, however, concern itself with getting the image to play nice on the proxmox platform specifically.

You still might find it useful for that purpose - if not, please report back where and why you get stuck :)

sanjeewasam, 2018-12-17:
> Read many good things about Home Assistant and hass.io,
> and that was the reason that I invested in a home "virtualization lab" to run it.
> 
> **This is a brief overview of the setup:**
> Intel NUC7i5DNHE, i5 with 04 cores and 32GB RAM
> Proxmox VE v5.2-5 running in clustering mode (with only 01 cluster node for the moment)
> [Proxmox is running fine and e.g. Univention Corporate Server is running.
> Note for other interested users:
> Proxmox VE = quite a steep learning curve for non Linux pros, even wit a GIU :-( ]
> 
> As far as I understand, hassos_ova-1.8.vmdk now would include working VirtIO drivers.
> That is good news, if only I could get it working ...
> 
> Given that I run Proxmox VE v5.2-5, **by default the VMs now run from Thin LVM storage**. But it also means that the hass.io .vmdk image must first be converted into Proxmox VE .raw format. Such can relatively easy be done with the qemu-img convert command.
> (see [https://docs.openstack.org/image-guide/convert-images.html](url))
> (Note: the new –and higher performance– Thin LVM VM storage is a logical block volume. And as a consequence, it does not support running the VM from a _file_ such as .vmdk or qcow2.)
> 
> For us Intel NUC+Proxmox _ambitioners_, it would be a serious advantage to have an (bootable) ISO image of hass.io instead of the .vmdk image.
> 
> After a few days playing (and struggling) with the new Proxmox virtualization+containerization setup, still no luck of getting hass.io up-and-running, because it doesn't boot (yet).
> (With e.g. Univention Corporate Server, the process went very successful, because it's much easier in Proxmox to create a VM beginning from an ISO...)
> 
> Wanted to share some background for people interested in running hass.io on Intel NUC + Proxmox.
> But if anyone could provide some winning tip to succeed in booting hass.io, be welcome :-)

Did you manage to get .vmdk to work on Promox? I created a VM and installed Debian without GUI and then got hass.io installed. It works great. I have Lenovo Tiny M93P (core i5 dual core gen4, 8GB ram and 256SDD) and work great for Hass.io and I have windows 7 also running all the time for Blue Iris cam server supporting 10 cameras. Both work perfectly without any issue in my small machine. Much much better expeirince than hass.io in Pi 3+ and having surface pro 1 for Blue Iris. CPU is not even 30% used and Memory usage is for both VMs and Promox <6GB

bagobones, 2018-12-17:
I opened a separate issue. The VirtIO network driver appears to work in 1.8 but it doesn’t appear to be possible to use the VirtIO disk controller to boot.

Alex9779, 2018-12-28:
I also tried running the current ova image on a proxmox host, lastest patches (5.3-5) and got it working though I had a few problems with the descriptions I found above and elsewhere so here a step-by-step list of what I did:
1. Create a new VM on the promox host.
2. In "Hardware" select the disk, "Detach" then "Remove".
3. In "Options" change "BIOS" to "OVMF".
4. In "Hardware" add "EFI Disk" to the machine.
5. I also removed the CDROM from the "Hardware", didn't try with it so no idea if that's needed.
6. Download latest vmdk image from https://github.com/home-assistant/hassos/releases to my proxmox host.
7. "gunzip" that file on my proxmox host. 
8. Import that file to the machine: `qm import <vmid> hassos_ova-2.4.vmdk <storageid>`, no need to convert or do something with it at least if you use a block based storae like LVM or ZFS like me, you can also do conversion with that command directly, read up here https://pve.proxmox.com/pve-docs/qm.1.html
9. In "Hardware" "Attach" the "Unused Disk" as "SATA" or "IDE", didn't get the difference yet, both work, "SCSI" does not!
10. In "Options" change first entry in "Boot Order" to the disk you attached.

Alex9779, 2018-12-28:
Jus a note to resources, I gave the machine 8GB of RAM and 4 core (1 socket), after installation and setup and running the configurator add-on the machine uses around 16% of that ram, 1.3GB...

bestlibre, 2019-01-10:
Seems to work. I'm on the preparing hassio screen. In point 8. above, it should be `qm importdisk`.

I have tried to resize the disk just after attaching it and before restart, will need to check the actual size.

tboyce021, 2019-01-10:
I just tried the latest 2.5 dev release and got it working with those instructions except I didn't add a separate EFI disk.

Resizing the disk didn't seem to work initially. I ended up creating another VM to boot an Ubuntu live CD and modified the config to load my existing HassOS disk on it. Then I used gparted to resize the data partition to fill the rest of the disk.

bestlibre, 2019-01-11:
Thanks @Alex9779 ! I have re-done the procedure with the 1.13 release and it works perfectly. In both case (last dev and last stable) the disk resize works without manual action. I have resized the disk before the first boot of hassio.

@tboyce021 for resizing the disk afterward you can use a live gparted CD (or ubuntu) and boot from it in the current VM.

tboyce021, 2019-01-11:
If I have Proxmox on a NUC, is there any reason to use the NUC image rather than the VMDK?

jvanderneutstulen, 2019-01-11:
@tboyce021 No, the NUC does not have the more efficient virtual net/disk drivers and is larger. So in virtual environment use the ova/vmdk image.

tboyce021, 2019-03-17:
Just an update regarding me earlier comment: While it boots fine without an EFI disk, it does give me a warning when starting the container along the lines of "no efidisk configured, using temporary efivars". I haven't had any issues because of this, but if you create an efidisk it won't give the warning.

closed 2019-05-23 by stale[bot]

jgthornburgh, 2020-04-23:
> 6\. Download latest vmdk image from https://github.com/home-assistant/hassos/releases to the proxmox host.

how do you do this? I have it on my computer, but can only see how to upload iso or img files to promox

tboyce021, 2020-04-23:
> > 6\. Download latest vmdk image from https://github.com/home-assistant/hassos/releases to the proxmox host.
> 
> how do you do this? I have it on my computer, but can only see how to upload iso or img files to promox

Been awhile since I did this but I believe I just had a screen, keyboard, and mouse attached directly to the host and used `wget` from the command line.

I'm about to set this all up again at a new house so I can hopefully help out then, but it might be a few days or weeks before I actually get it done.

Alex9779, 2020-04-23:
you have to download on your host or upload the file to it using sftp or something else, the GUI does not allow to upload it...