home-assistant/operating-system #100: CEC scanner addon not working on raspberry pi 3 B+

danielsjf, 2018-07-13:
I tried with 3 different Hdmi cables to scan the hdmi ports of my television. For all three I get the response "Autodetect failed"

Full log:
starting version 3.2.2 
autodetect FAILED

I have tried to reboot the raspberry pi. Tried to reboot the TV as well. For one of the cables I'm absolutely positive that it should work, since I also use it with another device.

The error is not very helpful, so I'm not certain if the problem is with Hass os or the addon. Most issues with autodetect failed seem to be solved by using another cable. I've tried this but it didnt work.

pvizeli, 2018-07-14:
Yes, that is a limit of 64bit version. Raspberry support this since 1 month but I find not time to add this into container. I will try to do this for 0.74.0

pvizeli, 2018-07-14:
/move home-assistant/hassio-homeassistant

closed 2018-07-14 by move[bot]

danielsjf, 2018-09-10:
@pvizeli I am still interested in having this functionality. At the moment the add-on isn't working for me.

nbeernink, 2018-09-14:
I'm also curious how this depends on 64-bits vs 32-bits. What can I do to help bring cec compatibility to Hassio on the Raspberry Pi 3 Model B?

```
core-ssh:~# hassio ha info
{
    "result": "ok",
    "data": {
        "version": "0.77.3",
        "last_version": "0.77.3",
        "machine": "raspberrypi3-64",
        "image": "homeassistant/raspberrypi3-64-homeassistant",
        "custom": false,
        "boot": true,
        "port": 8123,
        "ssl": true,
        "watchdog": true,
        "wait_boot": 600
    }
}
```

rarrais, 2019-04-25:
I'm also having the same issue as @danielsjf on a 32 bit system. 

`core-ssh:~# hassio ha info
arch: armv7
boot: true
custom: false
image: homeassistant/raspberrypi3-homeassistant
ip_address: 172.30.32.1
last_version: 0.92.0
machine: raspberrypi3
port: 8123
ssl: false
version: 0.92.0
wait_boot: 600
watchdog: true`

LucianoVandi, 2019-05-28:
Same issue here. The result of `hassio ha info`:

```sh
arch: armv7
boot: true
custom: false
image: homeassistant/raspberrypi3-homeassistant
ip_address: 172.30.32.1
last_version: 0.93.2
machine: raspberrypi3
port: 8123
ssl: false
version: 0.93.2
wait_boot: 600
watchdog: true
```

nbeernink, 2019-05-28:
I forgot to mention that CEC control works for me now. I replugged the HDMI cable to the receiver and hassio (on raspberry pi 3-64) can now control all the devices connected to the receiver.

askthomas, 2019-07-01:
CEC:  autodetect FAILED

Tested on Pi Zero w and Pi 3, also tested with new 2.5a USB power and new HDMI Supra cable.

Any ide? It did work, then i did a clean reinstall and then i cant get CEC to work.

/Thomas