home-assistant/operating-system #102: Installing HassOS 1.5 on VMware ESXi 6.5. Problem and solution.

mihalski, 2018-07-14:
Hi,

This is specific to the versions in the topic as I've had different experiences with HassOS 1.3 for example and not yet had the chance to try ESXi 6.7.

Problem:
Installing HassOS 1.5 directly in VMware ESXi 6.5 leads to no ip/interface on the LAN (DHCP server running on pfSense), so homeassistant cannot be installed.

Solution:
You can do the installation on an identically configured vm in VMware Fusion (or Workstation I presume) and it works fine. You can then export an OVA file and import it back into ESXi. Just make sure you've got the right compatibility settings before exporting.

In my case I chose Other Linux 3.x or greater as the OS and compatibility of vmx13.

Would be interested in hearing other people's experiences with HassOS and VMware ESXi.

Regards,
Michal

pvizeli, 2018-07-14:
Yeah the `vmdk` is only a workaround and need fine tuning on the hypervisor to run it.

I hope someone can help to make a OVA/OVF. At the end they are only a tar file with a xml inside they describe how you need setup a vm.

mihalski, 2018-07-14:
The strange thing is that HassOS 1.2 installed in ESXi ok and then updated to 1.3.. After that it would fail to update to 1.4 without error. But there was no problem with ip/interface in either case.

I wonder if reverse engineering the OVA I made can be of any use?

mihalski, 2018-07-15:
Wanted to add that I was successfully able to update from 1.5 to 1.6 on ESXi using the update function in the Hass.io System panel.

mihalski, 2018-07-15:
For the next 14 days this HassOS 1.5 OVA should be available for download. It updates to 1.6 on ESXi 6.5 for me.

https://transfer.sh/p8sTz/hassos-1.5.ova

If anyone tried it, let me know how it went.

Regards,
Michal

shad0wca7, 2018-07-19:
I'm getting this same issue installing HassOS 1.8 on ESXI 6.7. It's not getting an IP address - pfsense internally as well. I've tried the e1000 network adaptor and that's made no difference.

mihalski, 2018-07-20:
Did you try the OVA I uploaded?

I was able to transfer that (as an updated 1.7) from ESXi 6.5 to ESXi 6.7 yesterday (just upgraded) and it updated to HassOS 1.8 just fine.

gerard33, 2018-07-20:
@mihalski installed your HassOS 1.5 OVA successfully on ESXI 6.7 and was able to update it to HassOS 1.8.
Thanks!

mihalski, 2018-07-21:
Glad somebody got some use out of it as it took me about 5 hours to upload :P

nugget, 2018-07-21:
I also installed it successfully and appreciate the hours spent uploading.  One question -- is there any way to enable ssh in to the core OS (and not just the ssh addon container)?

mihalski, 2018-07-21:
You're welcome. I don't believe there is a way to ssh into the host (unless maybe you create some sort of a loopback to the console from within a container maybe?).

But in the case of being physically at the console you can log in as root with no password. So probably not advisable to make that network accessible.

I do share the desire to be able to get at the host, purely for debugging reasons.

nugget, 2018-07-21:
Logging in to the console actually does not work.  It gets you a `hassio` shell, but not a real shell.  There is an official process for the SD card images documented [here](https://developers.home-assistant.io/docs/en/hassio_debugging.html) where you place your `authorized_keys` file in the boot partition to enable host level ssh.  But I don't know how that would translate to the OVA/VMDK install.

mihalski, 2018-07-21:
Did you read what it says when you go into the hassio shell? Try typing 'login'.

That authorised_keys documentation does not apply to HassOS. There appears to be an equivalent but I'm not sure how it works. See:

https://github.com/cogneato/hassos/blob/dev/Documentation/configuration.md

Either way it works via a USB drive that you connect to the HassOS device and then import the config via the web UI.

nugget, 2018-07-21:
> Did you read what it says when you go into the hassio shell? Try typing 'login'.

Dang, you're right, I completely missed that.  Thanks.

apo-mak, 2018-07-22:
@pvizeli 
 i made ova Version 1 and 2 files.
also  ovf (for vmware compatibility).
 
in the below link .. 
https://github.com/apo-mak/hassos_VM
p.s. i am planing to update them when there is a new release of  hassOS

mihalski, 2018-07-22:
OVA files do work in VMware (I provided one in this very thread).

Method I used was to create the VM with VMware Fusion (Workstation should work the same), configure it appropriately (but never start it) and then export to OVA.

borpin, 2018-07-24:
> Would be interested in hearing other people's experiences with HassOS and VMware ESXi.

Using an older ESXi 5.5 (free). Had to carefully select the OS Type so I could then force EFI boot which gets me to the boot menu.  The final piece was this tip https://community.home-assistant.io/t/no-bootable-medium-found-system-halted/60783/9 to delete the CD ROM.

closed 2019-04-23 by stale[bot]

heapxor, 2022-08-07:
> les do work in VMware (I provided one in this very thread).
> 
> Method I used was to create the VM with VMware Fusion (Workstation should work the same), configure it appropriately (but never start it) and then export to OVA.

OVA is definitely broken and it has to be RAISED AS A BUG.
After importing OVA VM cant be snapshoted/suspended anything bc wrong /old template is used.

Using vmdk there is anothe BUG as someone didnt covert old IDE to SCSI - so it cant be extended. 
After conversion from ide to scsi - issue with uefi not bootable.

i cant believe that Both options are Seriously broken - nooone raised that as a BUG and noone care. 
is that kind of a game how not to be able run haas properly on Esxi , are u forcing all who wants to virtualise to buy hardware, thats insane!

stevegore, 2023-08-18:
Yeah my Linux server had a hardware malfunction so I tried to install HA on Windows. Not of the OVA images work. The instructions are terribly out of date as well. But if the images were working I'd at least have a shot.