home-assistant/operating-system #110: MAC ERROR: the disk you inserted was not readable

robmarkcole, 2018-07-16:
On flashing the latest image of HassioOS on Mac with etcher, on completion I see the following and cannot access the disk - any advice? When I insert the SD card into the pi HassOS does boot up, but since I cannot edit the network file via my Mac I cannot setup wifi.

[image]

pvizeli, 2018-07-16:
You need a usb stick and not edit the SDcard :) Maybe the Mac OSX don't support GPT legacy mode.

pvizeli, 2018-07-16:
https://github.com/home-assistant/hassos/blob/dev/Documentation/configuration.md

robmarkcole, 2018-07-16:
OK the [docs](https://www.home-assistant.io/hassio/installation/) need updating then as states:
```
Optional - Setup the WiFi or static IP: On the SD-card, create the network/my-network file and follow the HassOS howto.
```

pvizeli, 2018-07-16:
Where you see that? I can't find this...

robmarkcole, 2018-07-16:
https://www.home-assistant.io/hassio/installation/

cogneato, 2018-07-17:
It should just say

> Optional - To setup WiFi or a static IP, follow the HassOS howto.

The reference to SD is incorrect.

frenck, 2018-07-17:
PR has been merged. This issue may be closed now.

closed 2018-07-17 by pvizeli

smuda, 2019-07-19:
There are other references to the SD-card as well. The one I stumbled upon is
[https://www.home-assistant.io/hassio/zwave/](https://www.home-assistant.io/hassio/zwave/)

Sorry if this is obvious, but why isn't it possible to open the SD-card in a mac?

rudders, 2019-07-19:
The file system used by HASSOS on the SD card is not compatible with MacOS. A Linux VM on your Mac would be able to read it as a work around.

virgilm, 2020-07-16:
This is still a problem/confusing. Had to google it to get to this page.

The installation [page](https://www.home-assistant.io/hassio/installation/) still says:
"Optional - set up the Wi-Fi or a static IP address. There are two possible places for that:"

Linux VM is indeed a valid workaround.

tmm1, 2020-10-03:
See https://gist.github.com/miguelmota/d1d70d19fbd763afdd65c5c1375ddbde#gistcomment-2735565

h/t https://community.home-assistant.io/t/how-to-access-config-txt-in-hassio/139330/19