home-assistant/operating-system #113: Snapshot restoration issues

Billabongodysee, 2018-07-17:
When performing an upgrade following [these](https://www.home-assistant.io/blog/2018/07/11/hassio-images/) instructions, at the step where I am restoring my snapshot, I am having issues with it restoring everything.  It is failing to restore my customize, automation, iframe, and lights yaml files. It also fails to restore my nest.conf file. I can add them back manually and keep on going but it is strange that it is not putting everything back.
Thanks for all your hard work!

cogneato, 2018-07-17:
I haven't had an issue restoring Home Assistant or any items within /config or /share, but it failed to restore any addons which were not from core or community repos. Adding the missing repos manually to the new install allowed the snapshot to restore the missing addons. So somehow the repositories are not being restored.

pvizeli, 2018-07-17:
@cogneato interesting, need look into this issue.

@Billabongodysee. Could be that you come from raspbian? It can only restore if the permission of files are correctly. Haspbian/raspbian user run in a context of a other user. If you edit your config files without a hass.io add-on, they will be fails. I think that I need reset the user permission before we do a snapshot to avoid this issue.

pvizeli, 2018-07-17:
/move home-assistant/hassio

closed 2018-07-17 by move[bot]