home-assistant/operating-system #117: WiFi not working on Pi 3B+ Hassos

FlaMike, 2018-07-17:
I installed the new distribution of HASSOS over the weekend (July 7).  I created a USB stick per the instructions, named CONFIG with a directory named network and a plain-text file with no extensions called my-network, the file content is at the end of this note.  

I have not been able to get WiFi going on the Pi.

I connected a keyboard & monitor to the Pi & logged in.  I verified the file was copied correctly to /etc/NeworkManager/system-connections on the Pi.  The file protection is -RW-------

The output from nmcli d says that wlan0 is disconnected

The output from dmesg | grep mmc1 says wlan0: link is not ready

Content of my-network file:

[connection]
id=hassos-network
uuid=72111c67-4a5d-4d5c-925e-f8ee26efb3c3
type=802-11-wireless

[wifi]
mode=infrastructure
ssid=mySSID

[wifi-security]
auth-alg=open
key-mgmt=wpa-psk
pskmyPSK

[ipv4]
method=auto

[ipv6]
addr-gen-mode=stable-privacy
method=auto

tjorim, 2018-07-17:
If you just copied and pasted the content of the file over here, than there might be an issue with [wifi-security] on the last line: `pskmyPSK`. You might want to add an equals sign.

FlaMike, 2018-07-17:
My bad!  The line in the file is actually:  psk=myPSK

When I removed my real key, I lost the equal sign.  Good catch, though.

robertho94, 2018-07-18:
WiFi setup only: Format a USB-Stick with name CONFIG and copy one of the examples to the file network/hassos-wifi.

I don't know this step, try many time but can't config wifi... Just Ethernet cable work!

pvizeli, 2018-07-18:
`journalctl -u NetworkManager`

FlaMike, 2018-07-18:
Will post the log as you requested--probably tomorrow.  Today is pretty busy & I'm not sure I can get it for you as soon as I'd like to.  Thanks for your patience & help!

FlaMike, 2018-07-18:
I had a bit of free time so I gave it a whirl.  I am using the ssh add-on and get the following error message:  -bash: journalctl: command not found

Do I need to setup the developer ssh or can I only run this command using direct log in?

Sorry if I am asking something basic.

FlaMike, 2018-07-18:
Got it!  Log is below--more available if you need it, but I think this may be enough.  Please let me know.

journalctl -u NetworkManager
-- Logs begin at Tue 2018-07-17 19:02:30 UTC, end at Wed 2018-07-18 16:13:36 UTC. --
Jul 17 19:05:34 hassio NetworkManager[219]: <info>  [1531854334.8862] device (wlan0): set-hw-addr: set MAC address to B6:6A:2B:66:9E:D3 (scanning)
Jul 17 19:05:34 hassio NetworkManager[219]: <info>  [1531854334.9143] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 19:05:34 hassio NetworkManager[219]: <info>  [1531854334.9159] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 19:10:50 hassio NetworkManager[219]: <info>  [1531854650.8978] device (wlan0): set-hw-addr: set MAC address to 2A:6F:61:91:18:C6 (scanning)
Jul 17 19:10:50 hassio NetworkManager[219]: <info>  [1531854650.9234] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 19:10:50 hassio NetworkManager[219]: <info>  [1531854650.9244] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 19:16:06 hassio NetworkManager[219]: <info>  [1531854966.9136] device (wlan0): set-hw-addr: set MAC address to 2E:1A:AB:49:6C:59 (scanning)
Jul 17 19:16:06 hassio NetworkManager[219]: <info>  [1531854966.9428] device (wlan0): supplicant interface state: inactive -> disconnected
Jul 17 19:16:06 hassio NetworkManager[219]: <info>  [1531854966.9436] device (wlan0): supplicant interface state: disconnected -> inactive
Jul 17 19:21:22 hassio NetworkManager[219]: <info>  [1531855282.8976] device (wlan0): set-hw-addr: set MAC address to D6:37:CB:8A:D2:DF (scanning)
an0): supplicant interface state: inactive -> disabled531855282.9258] device (wl--More-- 
Jul 17 19:21:22 hassio NetworkManager[219]: <info>  [1531855282.9266] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 19:26:38 hassio NetworkManager[219]: <info>  [1531855598.9219] device (wlan0): set-hw-addr: set MAC address to 3E:E1:39:22:44:AF (scanning)
Jul 17 19:26:38 hassio NetworkManager[219]: <info>  [1531855598.9479] device (wlan0): supplicant interface state: inactive -> disconnected
Jul 17 19:26:38 hassio NetworkManager[219]: <info>  [1531855598.9486] device (wlan0): supplicant interface state: disconnected -> inactive
Jul 17 19:31:54 hassio NetworkManager[219]: <info>  [1531855914.9216] device (wlan0): set-hw-addr: set MAC address to 5A:F4:8F:74:B3:4C (scanning)
Jul 17 19:31:54 hassio NetworkManager[219]: <info>  [1531855914.9467] device (wlan0): supplicant interface state: inactive -> disconnected
Jul 17 19:31:54 hassio NetworkManager[219]: <info>  [1531855914.9475] device (wlan0): supplicant interface state: disconnected -> inactive
Jul 17 19:37:10 hassio NetworkManager[219]: <info>  [1531856230.9220] device (wlan0): set-hw-addr: set MAC address to 7A:89:B1:9B:E0:9D (scanning)
Jul 17 19:37:10 hassio NetworkManager[219]: <info>  [1531856230.9510] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 19:37:10 hassio NetworkManager[219]: <info>  [1531856230.9520] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 19:42:26 hassio NetworkManager[219]: <info>  [1531856546.9219] device (wlan0): set-hw-addr: set MAC address to 3A:D9:4F:B1:E7:11 (scanning)
Jul 17 19:42:26 hassio NetworkManager[219]: <info>  [1531856546.9487] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 19:42:26 hassio NetworkManager[219]: <info>  [1531856546.9498] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 19:47:42 hassio NetworkManager[219]: <info>  [1531856862.8977] device (wlan0): set-hw-addr: set MAC address to 7A:2E:28:D4:BE:24 (scanning)
Jul 17 19:47:42 hassio NetworkManager[219]: <info>  [1531856862.9235] device (wlan0): supplicant interface state: inactive -> disconnected
Jul 17 19:47:42 hassio NetworkManager[219]: <info>  [1531856862.9244] device (wlan0): supplicant interface state: disconnected -> inactive
Jul 17 19:52:58 hassio NetworkManager[219]: <info>  [1531857178.8896] device (wlan0): set-hw-addr: set MAC address to 22:C8:11:A4:53:0D (scanning)
Jul 17 19:52:58 hassio NetworkManager[219]: <info>  [1531857178.9209] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 19:52:58 hassio NetworkManager[219]: <info>  [1531857178.9221] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 19:58:14 hassio NetworkManager[219]: <info>  [1531857494.8661] device (wlan0): set-hw-addr: set MAC address to 3A:B3:93:89:EE:34 (scanning)
Jul 17 19:58:14 hassio NetworkManager[219]: <info>  [1531857494.8926] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 19:58:14 hassio NetworkManager[219]: <info>  [1531857494.8936] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 20:03:30 hassio NetworkManager[219]: <info>  [1531857810.8663] device (wlan0): set-hw-addr: set MAC address to A6:C8:0D:F5:7B:C0 (scanning)
Jul 17 20:03:30 hassio NetworkManager[219]: <info>  [1531857810.9022] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 20:03:30 hassio NetworkManager[219]: <info>  [1531857810.9038] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 20:08:46 hassio NetworkManager[219]: <info>  [1531858126.8658] device (wlan0): set-hw-addr: set MAC address to 02:A3:E9:E0:EE:9C (scanning)
Jul 17 20:08:46 hassio NetworkManager[219]: <info>  [1531858126.8906] device (wlan0): supplicant interface state: inactive -> disabled
[... cut ...]
Jul 17 22:09:54 hassio NetworkManager[219]: <info>  [1531865394.9479] device (wlan0): supplicant interface state: inactive -> disabled
Jul 17 22:09:54 hassio NetworkManager[219]: <info>  [1531865394.9488] device (wlan0): supplicant interface state: disabled -> inactive
Jul 17 22:15:10 hassio NetworkManager[219]: <info>  [1531865710.8656] device (wlan0): set-hw-addr: set MAC address to BE:5E:D2:26:8C:A4 (scanning)
Jul 17 22:15:10 hassio NetworkManager[219]: <info>  [1531865710.8933] device (wlan0): supplicant interface state: inactive -> disabled

pvizeli, 2018-07-18:
Can you provide the log after restart of system?

FlaMike, 2018-07-18:
Just restarted it.  Here is the log file:

~ # journalctl -u NetworkManager
-- Logs begin at Sun 2018-01-28 15:58:17 UTC, end at Wed 2018-07-18 19:55:05 UTC. --
Jul 18 19:51:01 hassio systemd[1]: Starting Network Manager...
Jul 18 19:51:02 hassio NetworkManager[268]: <info>  [1531943462.8491] NetworkManager (version 1.10.2) is starting... (for the first time)
Jul 18 19:51:02 hassio NetworkManager[268]: <info>  [1531943462.8498] Read config: /etc/NetworkManager/NetworkManager.conf
Jul 18 19:51:02 hassio NetworkManager[268]: <info>  [1531943462.9161] manager[0x3eb61080]: monitoring kernel firmware directory '/lib/firmware'.
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.0138] hostname: hostname: using hostnamed
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.0140] hostname: hostname changed from (none) to "hassio"
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.0156] dns-mgr[0x3eb7d130]: init: dns=default, rc-manager=file
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.0242] rfkill0: found WiFi radio killswitch (at /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill0) (driver brcmfmac)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.0258] manager[0x3eb61080]: rfkill: WiFi hardware radio set enabled
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.0259] manager[0x3eb61080]: rfkill: WWAN hardware radio set enabled
Jul 18 19:51:03 hassio systemd[1]: Started Network Manager.
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.1786] settings: loaded plugin keyfile: (c) 2007 - 2016 Red Hat, Inc.  To report bugs please use the NetworkManager mailing list.
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.2801] keyfile: new connection /etc/NetworkManager/system-connections/my-network (72111c67-4a5d-4d5c-925e-f8ee26efb3c3,"hassos-network")
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3422] manager: rfkill: WiFi enabled by radio killswitch; enabled by state file
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3427] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3429] manager: Networking is enabled by state file
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3443] dhcp-init: Using DHCP client 'dhclient'
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3530] Loaded device plugin: NMBondDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3533] Loaded device plugin: NMBridgeDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3536] Loaded device plugin: NMDummyDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3538] Loaded device plugin: NMEthernetDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3540] Loaded device plugin: NMInfinibandDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3542] Loaded device plugin: NMIPTunnelDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3544] Loaded device plugin: NMMacsecDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3546] Loaded device plugin: NMMacvlanDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3548] Loaded device plugin: NMPppDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3549] Loaded device plugin: NMTunDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3551] Loaded device plugin: NMVethDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3552] Loaded device plugin: NMVlanDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3553] Loaded device plugin: NMVxlanDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4177] Loaded device plugin: NMBluezManager (/usr/lib/NetworkManager/libnm-device-plugin-bluetooth.so)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4275] Loaded device plugin: NMWifiFactory (/usr/lib/NetworkManager/libnm-device-plugin-wifi.so)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4316] Loaded dev--More--

FlaMike, 2018-07-18:
Here is some more from the same restart.  Perhaps closer to what you're looking for.

ul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3546] Loaded device plugin: NMMacvlanDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3548] Loaded device plugin: NMPppDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3549] Loaded device plugin: NMTunDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3551] Loaded device plugin: NMVethDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3552] Loaded device plugin: NMVlanDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.3553] Loaded device plugin: NMVxlanDeviceFactory (internal)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4177] Loaded device plugin: NMBluezManager (/usr/lib/NetworkManager/libnm-device-plugin-bluetooth.so)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4275] Loaded device plugin: NMWifiFactory (/usr/lib/NetworkManager/libnm-device-plugin-wifi.so)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4316] Loaded device plugin: NMWwanFactory (/usr/lib/NetworkManager/libnm-device-plugin-wwan.so)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4393] Loaded device plugin: NMAtmManager (/usr/lib/NetworkManager/libnm-device-plugin-adsl.so)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4461] device (lo): carrier: link connected
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4529] manager: (lo): new Generic device (/org/freedesktop/NetworkManager/Devices/1)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4634] manager: (dummy0): new Dummy device (/org/freedesktop/NetworkManager/Devices/2)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4780] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/3)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4878] keyfile: add connection in-memory (e5b2792f-8e6d-3699-b72f-f6a0a2dc055c,"Wired connection 1")
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.4922] settings: (eth0): created default wired connection 'Wired connection 1'
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.5011] device (eth0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.5381] wifi-nl80211: (wlan0): using nl80211 for WiFi device control
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.5461] device (wlan0): driver supports Access Point (AP) mode
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.5588] manager: (wlan0): new 802.11 WiFi device (/org/freedesktop/NetworkManager/Devices/4)
Jul 18 19:51:03 hassio NetworkManager[268]: <info>  [1531943463.5677] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.1215] device (wlan0): set-hw-addr: set MAC address to 26:23:49:7C:00:A4 (scanning)
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2010] device (eth0): carrier: link connected
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2215] bluez: use BlueZ version 5
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2522] device (eth0): state change: unavailable -> disconnected (reason 'carrier-changed', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2746] policy: auto-activating connection 'Wired connection 1'
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2930] supplicant: wpa_supplicant running
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2937] device (wlan0): supplicant interface state: init -> starting
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2960] bluez5: NAP: added interface B8:27:EB:EF:9B:4D
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2986] policy: au--More--

FlaMike, 2018-07-18:
And a bit more.  I don't know if this is enough to help you diagnose things.  If not, please let me know & I'll copy more for you.

Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2930] supplicant: wpa_supplicant running
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2937] device (wlan0): supplicant interface state: init -> starting
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2960] bluez5: NAP: added interface B8:27:EB:EF:9B:4D
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.2986] policy: auto-activating connection 'Wired connection 1'
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.3027] device (eth0): Activation: starting connection 'Wired connection 1' (e5b2792f-8e6d-3699-b72f-f6a0a2dc055c)
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.3052] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.3068] manager: NetworkManager state is now CONNECTING
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.3111] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.3241] device (eth0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.3330] dhcp4 (eth0): activation: beginning transaction (timeout in 45 seconds)
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.4050] dhcp4 (eth0): dhclient started with pid 325
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.4507] sup-iface[0x3eb43190,wlan0]: supports 5 scan SSIDs
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.4593] device (wlan0): supplicant interface state: starting -> ready
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.4599] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio dhclient[325]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
Jul 18 19:51:04 hassio dhclient[325]: DHCPREQUEST on eth0 to 255.255.255.255 port 67
Jul 18 19:51:04 hassio dhclient[325]: DHCPOFFER from 192.168.1.1
Jul 18 19:51:04 hassio dhclient[325]: DHCPACK from 192.168.1.1
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5371] dhcp4 (eth0):   address 192.168.1.120
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5373] dhcp4 (eth0):   plen 24 (255.255.255.0)
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5374] dhcp4 (eth0):   gateway 192.168.1.1
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5377] dhcp4 (eth0):   lease time 86400
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5378] dhcp4 (eth0):   hostname 'hassio'
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5379] dhcp4 (eth0):   nameserver '192.168.1.1'
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5380] dhcp4 (eth0): state changed unknown -> bound
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5462] device (eth0): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5533] device (eth0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5557] device (eth0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.5573] manager: NetworkManager state is now CONNECTED_LOCAL
Jul 18 19:51:04 hassio dhclient[325]: bound to 192.168.1.120 -- renewal in 42863 seconds.
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.6246] manager: NetworkManager state is now CONNECTED_SITE
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.6253] policy: set 'Wired connection 1' (eth0) as default for IPv4 routing and DNS
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.6273] device (eth0): Activation: successful, device activated.
Jul 18 19:51:04 hassio NetworkManager[268]: <info>  [1531943464.6357] manager: NetworkManager state is now CONNECTED_GLOBAL
Jul 18 19:51:07 hassio NetworkManager[268]: <info>  [1531943467.3413] manager: startup complete
Jul 18 19:51:12 hassio NetworkManager[268]: <info>  [1531943472.0971] manager: (docker0): new Bridge device (/org/freedesktop/NetworkManager/Devices/5)
Jul 18 19:51:12 hassio NetworkManager[268]: <info>  [1531943472.4513] manager: (hassio): new Bridge device (/org/freedesktop/NetworkManager/Devices/6)
Jul 18 19:51:18 hassio NetworkManager[268]: <info>  [1531943478.7711] manager: (veth4950796): new Veth device (/org/freedesktop/NetworkManager/Devices/7)
Jul 18 19:51:18 hassio NetworkManager[268]: <info>  [1531943478.7897] manager: (veth3b1a8e9): new Veth device (/org/freedesktop/NetworkManager/Devices/8)
Jul 18 19:51:18 hassio NetworkManager[268]: <info>  [1531943478.8990] manager: (veth2564916): new Veth device (/org/freedesktop/NetworkManager/Devices/9)
Jul 18 19:51:18 hassio NetworkManager[268]: <info>  [1531943478.9152] manager: (veth8585a65): new Veth device (/org/freedesktop/NetworkManager/Devices/10)
[... cut ...]
Jul 18 19:51:19 hassio NetworkManager[268]: <info>  [1531943479.7782] device (docker0): carrier: link connected
Jul 18 19:51:20 hassio NetworkManager[268]: <info>  [1531943480.2640] device (veth8585a65): carrier: link connected
Jul 18 19:51:20 hassio NetworkManager[268]: <info>  [1531943480.2650] device (hassio): carrier: link connected
Jul 18 19:52:10 hassio NetworkManager[268]: <info>  [1531943530.8213] manager: (vethaa5aae3): new Veth device (/org/freedesktop/NetworkManager/Devices/11)
Jul 18 19:52:10 hassio NetworkManager[268]: <info>  [1531943530.8431] manager: (veth98da527)

closed 2018-07-19 by pvizeli

kajmaj, 2018-07-26:
Wifi is still not working on RPi 3B while using config USB. The only way is to use LAN.
Is there any way how to swap to wifi and change to static IP?

FlaMike, 2018-07-26:
I was also not able to get WiFi working with the latest release, including the fix Pascal noted above.  I am using Ethernet and a static IP address, as you inquired about.

What I did to assign a static IP address to the Pi was to use DCHP on my router.  I use the Pi's MAC address to assign a fixed IP address.  

You should be able to determine the Pi's MAC address by looking at the devices connected to the router & go from there.  Alternatively, you can use ipconfig on the Pi's command line to see ETH0's MAC address.

Hope this helps!

kajmaj, 2018-07-26:
My R Pi3 B started to act in the weird manner.
I have already assigned the static address from previous configuration - for unknown reason it does not work. Router decided to assign new IP and refusing to do what I want (but only with RPI)
I need to use wifi instead of LAN therefore I decided to revert back to 73.1 base on HassIO, but weird behavior continues - timezone is changed and seems to me that no way to set it up properly (timezone and lat, lon is already set in configuration.yaml)
It is quite frustrating. I spent whole day to find a way haow to have it working again bot no progress yet.

danielgoergen, 2018-07-31:
Have the same issue, can't get Wifi to work on latest release

kajmaj, 2018-08-01:
I have it solved :) 
Try to use this wifi config file:

```
[connection]
id=hassos-network
uuid=c07d6071-b647-4437-9c59-ff929e66526d
type=802-11-wireless

[802-11-wireless]
mode=infrastructure
ssid=my ssid
powersave=0

[802-11-wireless-security]
key-mgmt=wpa-psk
psk=my password

[ipv4]
method=auto
dns=192.168.1.1;8.8.8.8

[ipv6]
method=auto
```
You could try tu use USB during initial installation boot (it did not work for me) or do full install with LAN cable connected, import config from USB afterwards, unplug USB, unplug LAN cable reboot and you should be there :)

mitchando96, 2018-08-02:
This is great work. Would you have any ideas on how to get a wireless access point working from HassOS? I have it working great on Hassbian with RaspAP, but nothing seems to work on HassOS.

pvizeli, 2018-08-02:
That is a normal NetworkManager system. You can google how that work with NetworkManager and it will work :)

zaine68, 2018-08-12:
I'm a little confused still. Managed to find my way (finally) in to /etc/NetworkManager and in there is the config that i popped on my USB stick when I installed so in theory, there shouldn't be a problem. Reboot after reboot, it's still pulling a DHCP address from the router of 192.168.1.x. if i use the SSH setup from within Hass.io and do an ifconfig, i get this:

core-ssh:/etc# ifconfig
eth0      Link encap:Ethernet  HWaddr 02:42:AC:1E:21:02  
          inet addr:172.30.33.2  Bcast:172.30.33.255  Mask:255.255.254.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3013 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1252 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:256623 (250.6 KiB)  TX bytes:242914 (237.2 KiB)

Wireless is still not even trying to connect either. The config file its got in the /etc/NetworkManager/system-connections folder is:

[connection]
id=HassOS
uuid=9ef9d2c8-5eda-4f67-855a-b2f882e0ff6f
type=802-3-ethernet

[ipv4]
method=manual
address1=192.168.1.50/24,192.168.1.1
dns=8.8.8.8;8.8.4.4;

[connection]
id=hassos-wifi
uuid=282c187e-1aa6-494e-bc64-7a0fb760d479
type=802-11-wireless

[802-11-wireless]
mode=infrastructure
ssid=changedthisbit
powersave=0

[802-11-wireless-security]
auth-alg=open
key-mgmt=wpa-psk
psk=changedthisbit

[ipv4]
method=manual
address1=192.168.1.51/24,192.168.1.1
dns=8.8.8.8;8.8.4.4

Basically, i'm baffled but any help would be great thanks.

JoseAlcerreca, 2019-07-03:
Try `address1=192.168.1.51/24;192.168.1.1`

See https://github.com/home-assistant/hassos/issues/414

rmariboe, 2019-07-17:
According to the manual, use auth-alg=shared (for WPA) rather than auth-alg=open (for WPE):
https://developer.gnome.org/NetworkManager/stable/nm-settings.html

PR #124 (2018-07-19): fix wireless settings
Fix: #117