home-assistant/operating-system #120: Not working with Hyper-V

vb-git, 2018-07-18:
I have tried to get hassos working on Hyper-V running on Windows Server 2016 but was unable to. I will report my findings:

I used the 1.8 vmdk file which needed to be converted to VHD or VHDX to be able to used with hyperv. I tried both Starwind's V2V converter as Microsoft's Virtual machine converter. Both were able to convert to VHD / VHDX files without error.

I then created VM's, i tried both Gen 1 and 2 (hassos instructions mention UEFI needed so i think Gen 2 should be used). I attached the VHD(x) file.

In none of the cases the VM would actually boot - it would either show a blank screen with cursor blinking or it would display some UEFI booting diagnostic messages.

vb-git, 2018-07-18:
After increasing memory from 1 to 2GB i now sometimes see the hassos boot menu. The only options which "work" are the shell and System 0. The auto and system 1 don't do anything when enter is pressed. System 0 doesn't seem to progress through the boot however.

[image]

NK553, 2018-07-19:
I have the exact same problem on virtualbox, can't make it past the boot menu screen.

anisite, 2018-07-28:
Same! Not working on Hyper-V :(

mcinnes01, 2018-07-28:
I'm also having the same issue.

1liminal1, 2018-08-03:
Yeah I've just tried it as well and it won't boot on Hyper V

Love the guy giving instructions for the virtual box on a mac... Thanks, but doesn't help for those of us running a proper hypervisor lol

claymen, 2018-08-05:
Getting the same here on a Windows 10 Hyper-V (1803 build) using 1.9 of the image converted to a vhdx. 2vCPU, EFI (Gen2), Secure Boot Disabled, 3GB of RAM (No dynamic). And sadly still doesn't boot.

[image]

vb-git, 2018-08-15:
I tried it again with the latest development 2 / build 1 release but same result it seems.

claymen, 2018-08-30:
Still seeing the same issue on HassOS Release 1 build 1.10 (Stable). Converted using StarWind V2V convertor, tried both VHD and VHDX (not that that should make any difference).

lpt2007, 2018-09-25:
Same issue here, on Server 2016 and HassOS R1B1.10. Converted usin Microsoft tool. Any solution?

lpt2007, 2018-10-18:
Any progress about that?

lpt2007, 2018-10-18:
I find this, look at the end of the post.

[http://www.shafa.com/articles/AOaRqAtcjxlaI2XH.html](url)

tylerszabo, 2018-10-29:
I attempted to convert with `qemu-img convert -f vmdk hassos_ova-1.11.vmdk -O vhdx subformat=fixed hassos_ova-1.11.vhdx` and still no dice (was hoping it might have been an artifact of Microsoft Virtual Machine Converter).

lpt2007, 2018-10-31:
Any news on how to run hassos on hyper-v?

tylerszabo, 2018-10-31:
It's worth noting that Home Assistant will run in a Hyper-V virtual machine in general. I installed a guest Ubuntu 18.04 server (enable "universe" repository for the additional python dependencies) then followed the Raspbian/Debian guide [here](https://www.home-assistant.io/docs/installation/). I suspect the [python virtual env](https://www.home-assistant.io/docs/installation/virtualenv/) setup will work fairly well on any guest OS.

It might make more sense to specify the `vmdk` doesn't boot under Hyper-V.

claymen, 2018-11-01:
> It's worth noting that Home Assistant will run in a Hyper-V virtual machine in general. I installed a guest Ubuntu 18.04 server (enable "universe" repository for the additional python dependencies) [...]

I'm not sure you are in the right section. Of course it works in Hyper-V as it would under pretty much any hypervisor if you manually install your own Linux OS (or any supported OS) and then manually install the environment. This repo (and issue) is specifically about HassOS.

tylerszabo, 2018-11-01:
My apologies, I thought HassOS also covered arbitrary Raspbian environments running Home Assistant.

reharmsen, 2018-11-17:
Just treid with HassOS 1.12 inside Hyper-V 2012-R2
Still no luck, it keeps using a lot op CPU but it's not booting.

Any news on development to fix this?

apo-mak, 2018-11-20:
same here still no luck :-(

reharmsen, 2018-11-21:
To bad I do not know how to debug/investigate this problem any further (perhaps someone can point me in the right direction)

For myself I created a 'workaround' by manually installing Ubuntu server within a Hyper-V guest and installing Hass.IO there, it's not fully HassOS but at least it works for me.

reharmsen, 2018-11-28:
@pvizeli When can we expect a new release/build, in order for me to test this?

claymen, 2018-12-20:
The latest 2.4 build now boots on Hyper-V, you will need to disable secure boot (at least I had to), and use a Gen2 VM (for EFI) and the provided vhdx just works.

claytonjn, 2018-12-23:
> The latest 2.4 build now boots on Hyper-V, you will need to disable secure boot (at least I had to), and use a Gen2 VM (for EFI) and the provided vhdx just works.

Hmm, unfortunately this is not my experience, I'm seeing the same behavior as with a converted vmdk. I'm still seeing the same boot screen in the most recent screenshot you posted - after a seemingly random number of reboots it has a couple times started booting but then the docker service fails to start and the machine never fully boots. 

Gen2, Secure boot disabled, not using dynamic memory, and I also tried playing with number of cores assigned and with integration services nothing I've tried gets it to boot properly. FWIW I'm running Server 2016 Datacenter.

Codythejack, 2019-01-06:
VHDX provided by https://github.com/home-assistant/hassos/releases/download/2.4/hassos_ova-2.4.vhdx.gz is ok. You need to setup machine 2nd generation and disable secure boot by powershell command:
Set-VMFirmware "vm-machine-name" -EnableSecureBoot Off
I cannot check right now is 2GB of memory needed.

claytonjn, 2019-01-06:
No idea why I'm having so much trouble.
[image]

claymen, 2019-01-07:
> VHDX provided by https://github.com/home-assistant/hassos/releases/download/2.4/hassos_ova-2.4.vhdx.gz is ok. You need to setup machine 2nd generation and disable secure boot by powershell command: [...]

You don't have to do this via PowerShell, it is available in the UI as well. But yes you do need to disable secure boot at this stage.

> No idea why I'm having so much trouble.
> [image]

That looks like you are getting it to boot, it does take a long time on the first boot before it is functional as it runs through a lot of initial setup. If you are at least seeing the OS boot like you are, just give it some time. Although it does seem odd that you are getting Docker Failures, if I get time I'll setup a fresh copy and see if I can replicate, but on my Win 10 Hyper-V it just deployed, booted and went without issue.

With that said, I did have an issue when rebooting the VM, it went back to being stuck on the boot menu, it seemed to fix itself after changing the SCSI ID for the virtual disk to anything other than what it was on, after that it seemed to boot again.

claytonjn, 2019-01-07:
I had left it for a few hours with no progress, but after changing the SCSI ID I was able to get it to boot (although I have to change it any time it reboots). 

Still playing with it and trying to figure out how to get wget or similar working and I need to see if I can run a boot script similarly to traditional Linux.

TobiShorty, 2019-01-12:
Hi, I just experienced this issue to after updating Hassio on webif. On reboot I got stuck at the same EFI screen. 

My solution was, to get it back to work:
- check dynamic ram is Off! No matter if it was previously set on, simply turn it off. 
- change SCSI ID for your VHDD, ex from 0 to 1. 
- System will boot
- Shuutdown the Server again, and turn dynamic ram on again if you need. 
- System will also boot again with dynamic RAM. 

Im running with 512MB just fine.

pvizeli, 2019-01-13:
Maybe a problem with auto expand on bootloader? https://github.com/home-assistant/hassos/blob/dev/buildroot-external/bootloader/barebox/boot/system0#L6

We could try to use a newer barbox version... I hope U-boot will support UEFI soon like barebox and we can use only one bootloader 👍

vb-git, 2019-02-11:
I can now also report success using v. 2.8 on Hyper-V! My steps:

- download and unpack 2.8 VHDX image
- create new VM; gen2 with 1024MB memory and secure boot disabled
- attach the downloaded VHDX to the vm (don't create a new disk, only use this one)
- boot
- enjoy HASS.io

In one evening i was able to migrate my whole Ubuntu setup to hass.io. This includes MQTT server and Node-RED. I used the hass.io addons for this and everything seems to work perfectly fine. I also performed some reboots during setting up without problems. Now to see what the longer term stability is, also when HA or HASSOS updates are performed. Thanks for the continuing development!

closed 2019-02-11 by vb-git

cyberbit, 2019-02-12:
@vb-git Good to hear someone has had success! I downloaded the image, will try these steps later this week.

erwinjong, 2019-04-05:
Working fine with Gen2 and disable secure boot on W2012R2 Storage Server HyperV

theskillster, 2019-08-30:
Can confirm the VHDX image works on Hyper-v running on Windows Server 2019 Standard (evaluation copy), with 1GB RAM, no secure boot.
However I did get this error on the Hyperv logs (I assume this wasn't critical or the system would've stalled).
Description:
'HASSIO' has encountered a watchdog timeout and was reset. If the problem persists, contact Product Support for the guest operating system. (Virtual machine ID EC2BAB36-EAE3-42CA-9F91-DD3F53E0197E)

Log Name:
Microsoft-Windows-Hyper-V-Worker-Admin

Source:
Microsoft-Windows-Hyper-V-Chipset

Logged:
Fri Aug 30 2019 16:01:42 GMT+0100 (British Summer Time)

Event ID:
18600

Level:
Critical

LanceMcCarthy, 2020-02-13:
I can confirm v3.10 it works with  on windows 10 1909 (v18363) with no effort or special VM OS setup. You need to diable secure boot which is a simple checkbox (see below)

, but there was a caveat regarding secure boot.

1. Download the VHDX option ([link to all downloads](https://www.home-assistant.io/hassio/installation/))
2. Unblock the gz file before extracting (Right click -> Properties -> Unblock -> OK)
3. Extract the vhdx file
4. Open Hyper-V Manager
5. In the right pane, select "Quick Create"
6. Choose "Local File" creation option and point it to the vhdx
7. Leave all the defaults selected (or increase resources like RAM if desired)
8. In the VM's settings, disable Secure Boot:
[image]
9. Boot machine and wait for initial setup to complete.

Result:
[image]

The only issue I'm still dealing with is getting the adapter working so that hassio.local works (but that's not related to HASS). If anyone has any tips for this, I'm all ears :)

GeoffAtHome, 2020-05-05:
@LanceMcCarthy did you get your problem resolved?

GeoffAtHome, 2020-05-05:
@theskillster I am seeing a similar issue: 'Hassio' has encountered a watchdog timeout and was reset. If the problem persists, contact Product Support for the guest operating system. (Virtual machine ID C8589DC8-440E-4B87-BB5A-D4B6EC0678B7)

Did you manage to find a solution to this?

LanceMcCarthy, 2020-05-05:
@GeoffAtHome I ended up giving up on HyperV and used the VirtualBox image. Super easy install and I have superb control over USB, memory and vCPUs.

GeoffAtHome, 2020-05-06:
@LanceMcCarthy thanks. I went the other way. I started with the VirtualBox image and I am now using HyperV. Works well for me about from every now and again reporting an issue in hyper-V log. Reliable and I found it easier that VirtualBox for updating.

At one stage I was trying to use Docker on WSL but WSL needs more work. It maybe time to try this again as WSL has improved.

jogdy7827, 2020-05-20:
Hi guys, maybe somebody can help me out with this problem. I'm trying to run it on hyper v aswell. I'm just not able to reach the browser of home assistant. 
Logging in with: root will give me the opertunity to check the ip: it was 172.30.32.1 but i'm not able to connect from my pc. 
Can somebody help me? 
[image]

LanceMcCarthy, 2020-05-20:
> Hi guys, maybe somebody can help me out with this problem. I'm trying to run it on hyper v aswell. I'm just not able to reach the browser of home assistant.
> Logging in with: root will give me the opertunity to check the ip: it was 172.30.32.1 but i'm not able to connect from my pc.

@jogdy7827 
Are you using the right port? by default it's `8123`. Try: [http://172.30.32.1:8123](http://172.30.32.1:8123)

However, that IP address looks more like a loopback IP address, and a LAN IP address (i.e. 192.168.1.x) that the Hyper-V's network adapter would be assigned from the router (or wherever your DCHP is).

Double check that Hyper-V has a network adapter that is accessible from the host. Here's one (of many) posts that discuss how to do it: https://tunecomp.net/hyperv-virtual-machine-remote-access-lan-internet-external-switch/

jogdy7827, 2020-05-22:
> @jogdy7827
> Are you using the right port? by default it's `8123`. Try: http://172.30.32.1:8123
>
> However, that IP address looks more like a loopback IP address, and a LAN IP address (i.e. 192.168.1.x) [...]

@LanceMcCarthy 
Thanks for your help. I've checked my router and I needed to add the ip adres of my external switch into the DHCP binding. After that, rebooted and home asssistant with port 8123 was reachable.

scyto, 2021-01-10:
For hyper-v is there anyway to install the Linux Integration Services?

AshishMBakshi, 2021-02-02:
@theskillster @GeoffAtHome @LanceMcCarthy

Are you able to run Home Assistant OS on Hyper-V without issue now?

I've been running it on Hyper-V on Win Server 2019 for a couple months and fairly regularly see the "[Hassio VM] has encountered a watchdog timeout and was reset" error in my Hyper-V logs.

On at least a few occasions, that error appears to have caused Windows to crash, rebooting the host OS - which is a significant issue (given I run lots of other things on there).

Any idea how to fix this?

LanceMcCarthy, 2021-02-02:
>  "[Hassio VM] has encountered a watchdog timeout and was reset" error in my Hyper-V logs.
 I don't recall seeing this

> On at least a few occasions, that error appears to have caused Windows to crash, rebooting the host OS - which is a significant issue (given I run lots of other things on there).

I have never seen this. I suspect you may need to change where the VMs are stored. I do not use the detail c:/users/public location. Instead, I have another SSD (avoid HDD) where I run all my Hyperv machines from. 

If I had to take a wild guess, anything strong enough to crash the host system sounds like a hardware driver fault (it's never the kernel)

scyto, 2021-02-02:
I get these too since starting to use HASS in dec on hyper-v.

I assumed the were related to the "NMI received for unknown reason 00 on CPU 0. on the HASS console"  but that was an assumed correlation on my part. 

But I have not seen any crashes that i am aware of when either message has happened.

I am running Server 2019, all VMs stored on SSD.

scyto, 2021-02-02:
Just did some extra checking for me the watchdog timer eveoccur EXACTLY 4 minutes and 59 seconds after the VM is reported as booted in the hyper-v worker log and is *absolutely* correlated with the NMI error on the hass console.

As such I would posit that the watchodg timer was a result of the reboot after a crash, not the cause of the crash.

AshishMBakshi, 2021-02-02:
Hmm interesting - thanks. I run all my VMs off an NVME SSD, and encountered no notable unexpected host OS reboots until the Home Assistant OS VM was added. Will try to investigate further...

SixFive7, 2021-05-15:
I'm experiencing this issue on a stable host: https://github.com/home-assistant/operating-system/issues/1362.