home-assistant/operating-system #121: date/time is not correctly set on the Raspberry Pi 3B during first time power on

shiyush-git, 2018-07-19:
The Raspberry 3B does not have a clock, when first time boot the system after flashed the image to SD card,  looks like the system doesn't sync the date/time with NTP, the wrong date, caused the hassos_supervisor can't connect to the server to download the landing page(SSL error: CERTIFICATE_VERIFY_FAILED).

pvizeli, 2018-07-19:
They will be sync after the network is available with systemd timesync service. But they should ajust itself and work correctly after that?

shiyush-git, 2018-07-20:
I have waited for couple hours in powner on state with LAN cable plugged, the date/time can't be synced, which NTP server used by default?

pvizeli, 2018-07-20:
https://wiki.archlinux.org/index.php/systemd-timesyncd

DavidFW1960, 2018-07-23:
I logged an issue in Home Assistant with time - my histories have all flatlined... I think the timezone is being ignored??? This may be a HassOS issue not HA? I only just switched from ResinOS.

pvizeli, 2018-07-23:
Home Assistant is the time master. Maybe you need restart home assistant after you set the correct time zone inside config.

DavidFW1960, 2018-07-24:
I was running the new 32bit HassOS on a Pi3B and took a snapshot; went to 64bit and restored. I think the database was the issue as after I deleted the database it's all working properly now.

shiyush-git, 2018-07-25:
I just checked the timesyncd.conf, the default NTP server is time1.google.com, this probably is the reason why my Pi3 time can't not be synced, I am in China, thx gfw, no google at all.
But the tricky thing is the fallbackNTP setting looks like don't work.

closed 2018-07-31 by shiyush-git

chc189, 2018-08-05:
最后怎么解决？@shiyush-git