home-assistant/operating-system #122: USB stick missing from Hardware Info Page in VM

NK553, 2018-07-19:
Running HassOS in a Vitualbox VM with USB passthrough on (as a USB 2.0 EHCI Controller) the device does not appear in the hardware page.  From the CLI, hassio hw info does not show it either.

pvizeli, 2018-07-19:
Can you debug that on host OS, if they available there?

NK553, 2018-07-19:
During Initial VM loading, from the virtualbox log:

```
00:00:04.116633 EHCI: USB Operational
00:00:04.149572 EHCI: USB Suspended
00:00:04.149630 EHCI: Hardware reset
00:00:04.149675 EHCI: USB Operational
```

It is present and running in the logs, and shows up in a different ubuntu VM.

NK553, 2018-07-19:
@cogneato couldn't get it to work either...

[image]

NK553, 2018-07-19:
It's a USB Z-Wave stick, Aeotec Z-Stick Gen5, Z-Wave Plus USB.  It shows up in Windows device manager (HostOS) as "Sigma Designs, Inc as a com port...

It's a very common device in the community.  it worked perfectly in HassIO.

NK553, 2018-07-19:
Where can I run lsusb?  Do I need to setup keys and ssh into 22222?  I couldn't get there from the hassio cli.

NK553, 2018-07-19:
I GOT IT TO WORK!

I found the COM port in windows (COM7), forwarded that to my VM in Virtualbox as COM1, then put 

`zwave:
  usb_path: /dev/ttyS0
  polling_interval: 30000`

In my config.  Worked great.

pvizeli, 2018-07-19:
You can login on terminal of virtualbox as root. Maybe I find out what kind of serial adapter it is :)

closed 2018-07-19 by pvizeli