home-assistant/operating-system #125: RaZberry board not working, no option to disable BT on Raspberry PI 3

bboti86, 2018-07-19:
Hi Pvizeli! 

As described on the original Hass.IO installation site's Z-Wave section the RazBerry board requires the same UART channel as BT is using on the Pi 3 to function. https://www.home-assistant.io/hassio/zwave/ 
The ResinOS based Hass.IO was utilizing the config.txt file, but the newer version of the system based on Buildroot does not or maybe it is just not that easy to figure out how to change the settings. 

After some digging, assuming that I understand correctly what's happening in the image building process I found that based on the readme here: https://github.com/home-assistant/hassos/tree/dev/buildroot/board/raspberrypi 
the UART is actually used by Buildroot to provide a console interface (see remark #2). This would mean that the ReZberry's communication channel is blocked. 

Do you think that this could be the reason for what the users are describing in this topic: https://community.home-assistant.io/t/razberry-and-hassos/60283 

Unfortunately I'm not an expert of the new system. But would it be possible to disable the console that is opened by default? Do you think that the BT is disabled on the PI 3s? 

Regards:
Botond

rudders, 2018-07-24:
Have you added `dtoverlay=pi3-miniuart-bt` to the config.txt file in the root of the SD. Mine looked like this after editing which needs to be done on a computer that can read the squashfs file system. Mined looked like this after editing.
```
kernel=u-boot.bin
disable_splash=1
dtparam=audio=on
dtoverlay=pi3-miniuart-bt
```

pvizeli, 2018-07-24:
not squashfs, support only GPT with hyprid support.
You can also use the command line from host.

rudders, 2018-07-25:
@pvizeli 
> You can also use the command line from host

could you please elaborate - I didn't think you could access config.txt from the host os....

rudders, 2018-07-29:
Does your `Hassio - System - Hardware` Look similar to this?

[image]

bboti86, 2018-07-29:
It looks like this: 
[image]
You have a flashdrive attached and I have a USB microphone attached
the only difference I see is the gpiochip100, but that might be because I have the older Raspberry Pi 3

rudders, 2018-07-29:
Agreed - looks fine to me (not an expert by any means) - have you tried resetting the board on the Pi and making sure the led lights up when you power cycle - on my older 3B it was quite temperamental.

pvizeli, 2018-07-30:
You can edit the config from host with:
`vi /mnt/boot/config.txt`

closed 2018-07-30 by pvizeli

ZyberSE, 2018-07-30:
Forgot to update here, I had the 64-bit version installed, then switched to the recommended 32-bit and the exact same config worked.

rudders, 2018-07-30:
Of course, sorry I didn't think of that.

Great tip @pvizeli !!!! Needs to be documented - cost me two hours the other day that little trick...

I'll have my first go at a documentation PR

pepeEL, 2019-11-12:
Hi
I have a problem.  I use SSD Disc on USB. I edited config.txt by add this dtoverlay=pi3-miniuart-bt
After add this and reboot my RPI3b+ he dont boot my Hassio. WHen delete it it works ok.