home-assistant/operating-system #127: Home Assistant and IPv6

DidierLmn, 2018-07-21:
Hass.io Supervisor 115
HassOS 1.7
Home-Assistant 0.74
Raspberry Pi 2 B

Home Assistant does not get an IPv6 address on my router, while other devices do get an IPv6 address. 
Is IPv6 supported in HassOS?
If so, what configs/logs/... can I check to find the problem or enable IPv6 support?

pvizeli, 2018-07-23:
https://www.google.com/search?q=linux+networkmanager+ipv6&oq=linux+networkmanager+ipv6&aqs=chrome..69i57j0.5166j0j7&sourceid=chrome&ie=UTF-8

closed 2018-07-23 by pvizeli

tjorim, 2018-07-23:
Weird, my setup does get an IPv6.
Are you using any custom NetworkManager connection files?
The default profile should get an IPv6 address.

bayi, 2018-08-21:
The default profile has ipv6 in it ( look at: https://github.com/home-assistant/hassos/blob/dev/buildroot-external/rootfs-overlay/usr/share/system-connections/default ),

even adding an explicit config doesnt help here, Hassos isnt receiving an ipv6 address and ipv6 network is broken ( i see a log entries about not able to reach some of my zeroconf services because of that ).

I just restored my old Hassio installation with resin and it worked there, while Hassos isnt receiving an address.

@tjorim can u please confirm that your device has a working ipv6 ?

bayi, 2018-08-22:
@pvizeli is this enough information to reopen this issue ? 
I just reinstalled HassOS newest and it isnt receiving any ipv6 address no matter what i do.

What can i do to help you out with more information ?

widder, 2019-09-29:
What method are you using to get the IPv6 address? I tried it today and it's working fine with SLAAC