home-assistant/operating-system #128: Aeotec Zstick not recognised on ESXi 6.5 using ova

coldow, 2018-07-21:
I have a working Zwave network under Domoticz that I am trying to move to Home Assistant.

I have installed the ova on ESXi 6.5 using the link provided by mihalski for issue #102 and I have successfully updated from 1.5 to 1.8.

I use the ESXi web interface to connect my Aeotec Zstick Gen5 and my RFXCOM hub. The RFXCOM hub is recognised as /dev/ttyUSB0 and is working but the Aeotec Zstick is not.

Running lsusb I get
[image]

The Aeotec Zstick is recognised as Bus 002 Device 004.
But no device appears in the /dev directory so I can not connect to it.

From dmesg the boot process detects both USB devices and configures the RFXCOM hub but can not identify the Aeotec Zstick.
[image]

Can anyone help me please?
Thank you

mihalski, 2018-07-22:
Have you been able to test this with a rpi at all? I get the following (I also have another device on ttyACM0):
```
[152087.722845] usb 1-1.4: new full-speed USB device number 8 using dwc_otg
[152087.856112] usb 1-1.4: New USB device found, idVendor=0658, idProduct=0200
[152087.870953] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[152087.887859] cdc_acm 1-1.4:1.0: ttyACM1: USB ACM device
```

coldow, 2018-07-22:
I have tested this on an rpi3, it works correctly and attaches the Aeotec Zstick to /dev/ttyACM0 as expected and all the devices are correctly loaded.

I have also tested the ovf from @pvizeli at https://github.com/apo-mak/hassos_VM.
This has the same problem, the Aeotec Zstick is detected as a USB device but it is not identified and no /dev entry is added.

Thank you.

pvizeli, 2018-07-24:
Can you load the kernel module? Maybe that will not done automatic?

PR #132 (2018-07-28): OVA: Add support for Aeotec Zstick
Fix: #128 
Fix: #126
closed 2018-07-28 by pvizeli