home-assistant/operating-system #129: Add onewire

ezar, 2018-07-21:
I want to add onewire but I can access to boot folder.

pvizeli, 2018-07-24:
?

ezar, 2018-07-24:
> 2018-07-24 19:07:10 ERROR (SyncWorker_2) [homeassistant.components.sensor.onewire] No onewire sensor found. Check if dtoverlay=w1-gpio is in your /boot/config.txt. Check the mount_dir parameter if it's defined

cogneato, 2018-07-29:
config.txt is accessed through the SD on the hassos-boot partition, but I don't know if that is what you need here?

JakobTewes, 2018-07-31:
How to access config.txt:
  - Copy “authorized_keys” via (FAT32 formatted, named “CONFIG”) USB stick.
    USB config can be imported via web interface -> hass.io -> system -> “Import from USB”
  - SSH to Host os via Port 22222 with user “root”
  - Edit config via
      vi /mnt/boot/config.txt

closed 2019-04-23 by stale[bot]

danillab, 2019-09-25:
```sh
ssh root@hassio.local
ls /mnt/boot/
ls: /mnt/boot/: No such file or directory
```
How to mount boot?
how to edit the file /mnt/boot/config.txt ?

pvizeli, 2019-09-25:
You need enable debug access or but the SD card on your computer.