home-assistant/operating-system #130: [homeassistant.components.camera.rpi_camera] 'raspistill' was not found

hungreebr, 2018-07-23:
It's raising this error when I am trying to use the [rpi_camera component](https://www.home-assistant.io/components/camera.rpi_camera/). 

`[homeassistant.components.camera.rpi_camera] 'raspistill' was not found`

Raspberry Pi 3 B+
hassio version 1.3.1
home assistant version 0.74.0

pvizeli, 2018-07-24:
With extended Firmware from raspberry repository?

sergiofraiman, 2018-07-26:
no @pvizeli , just the plain installation from https://github.com/home-assistant/hassos/releases/download/1.7/hassos_rpi3-64-1.7.img.gz

floowsn, 2018-08-22:
Same error with 

Raspberry Pi 3 B+
HassOS 1.9
hass 0.76.2

Evanlec, 2018-12-19:
Same problem here
Raspberry Pi 3 B+
HassOS v1.13
HomeAssistant 0.84.3

There must be some way to install raspistill on HassOS?

closed 2019-04-23 by stale[bot]

Scott8586, 2019-05-11:
Reactivating.  How can we get this fixed?
HassOS/Hassio
HA 0.92.0

GrowFOUR, 2019-05-14:
Seriously! I have a Raspberry Pi 3 B+ with on board camera ribbon cable attached to genuine RPI camera V2 and I can't activate the camera rapistill because you cannot access so
 # Example configuration.yaml entry
camera:
  - platform: rpi_camera

DOES NOT WORK

schmidt91, 2019-05-17:
I have the same issue. Raspberry Pi 3 + 0.93.1.

Stebar59, 2020-06-11:
Do we really don't have a solution to have rpi_camera working?

kotwox, 2020-09-27:
does not work. is there any hope?)

marcinbauer85, 2020-10-26:
Can we get any response from the devs? Please 😢

Stebar59, 2020-10-26:
Please, please .... please.

-------- Messaggio originale --------Da: marcinbauer85 <notifications@github.com> Data: 26/10/20  12:29  (GMT+01:00) A: home-assistant/operating-system <operating-system@noreply.github.com> Cc: Stebar59 <stefano.barbay@gmail.com>, Comment <comment@noreply.github.com> Oggetto: Re: [home-assistant/operating-system] [homeassistant.components.camera.rpi_camera] 'raspistill' was not found (#130) 

Can we get any response from the devs? Please 😢

luci-11, 2020-10-29:
Logger: homeassistant.components.rpi_camera.camera
Source: components/rpi_camera/camera.py:42
Integration: Raspberry Pi Camera (documentation, issues)
First occurred: 18:10:07 (1 occurrences)
Last logged: 18:10:07

'raspistill' was not found

:( no way..

terryrankine, 2020-11-20:
#same

0.118.1

soterocra, 2021-03-18:
😢

Xitee1, 2021-06-11:
Still no solution?

agners, 2021-06-12:
No solution, see more details: https://github.com/home-assistant/core/issues/51755

RubenKelevra, 2021-07-10:
@agners I mean, can we please leave the tickets open as long as this is a limitation?