home-assistant/operating-system #131: HassOS on Synology DSM

coldi2k3, 2018-07-23:
Heyho..

tried to install it on my Synology DSM.. Got the same problem as other ppl with BIOS and UEFI.. that got fixed quickly.. 

but after that, i get to the bootmenu, i can select something and then it just freezez up or does nothing at all.. only the 4th option with the shell worked... any ideas?

schmidi198, 2018-07-27:
Hi

I have the same issue and i can confirm, that it does not work in Synology DSM with Virtual Machine Manager. See image below.

[image]

bagobones, 2018-10-28:
It is possible to boot now in 1.12 with the IDE drive and a VirtIO NIC and USB 3 enabled.. Doesn't look like VirtIO works for the disk controller however.

closed 2019-04-23 by stale[bot]