home-assistant/operating-system #133: Cannot istall HassOS to RPi 3B

kajmaj, 2018-07-26:
Hello, 
I am trying to do fresh install of HomeAssistant based on HassOS.
I flashed (via Etcher) the latest version  from [https://www.home-assistant.io/hassio/installation/](url) 
I prepared USB Drive, generated new UUID, put the connection file into the network folder, inserted both to RPi and boot RPi. 
That´s all, raspberry did not connect to preconfigured wifi, even if waiting one hour or so.
Tested with both `method=auto` and static IP. No success at all :(
I did all descibed here [https://www.juanmtech.com/home-assistant-hassos-beginners-guide/](url) up to point 2,  or in any other guide.
If I insert microSD with previous version (based on HassIO), everything goes well again, well, except several weird error reports.
This is my connection config:
```
[connection]
id=hassos-network
uuid=d69b7932-02ea-4e14-b29a-8c0939e1fcae
type=802-11-wireless

[802-11-wireless]
mode=infrastructure
ssid=my wifi name

[802-11-wireless-security]
auth-alg=open
key-mgmt=wpa-psk
psk=my password

[ipv4]
method=auto

[ipv6]
addr-gen-mode=stable-privacy
method=auto
```
Is there anything what I could do differently (in the better way)?

cogneato, 2018-07-26:
To pass your network configuration files, you will need to reboot with the USB in place or use the "import from USB" option in the UI. I prefer the UI method because you can check the logs below on the same page (hassio > system) to confirm the import. Then remove the USB, and reboot the host.

I have found that if you have ethernet physically connected and have created only a wifi connection, you need to remove the ethernet cable to reach the wifi ip address.

kajmaj, 2018-07-26:
So the process should be:
first boot with LAN cable -> full install -> hot plug USB -> import from USB -> hot unplug of USB ->  unplug LAN -> reboot ?

pvizeli, 2018-07-27:
You can also boot your device with USB. They import the network config also on startup.

kajmaj, 2018-07-27:
I boot device with USB stick plugged,  after +/-15 minutes IP scanner is not able to find it,  Fing as  well as  router does the same. i.e. no wifi connection available.
I  imported  config from usb while connected by the cable, switch off, uplugged cable and USB, boot again.
And - no wifi.
Connected to cable, everything goes well.
If I enter SD cart with HassIO, wifi works like a charm.
Am I doing sthg wrong?
Did I miss something?

kajmaj, 2018-07-27:
It look like there is no driver available

cogneato, 2018-07-27:
Can you come in to the discord channel or the community forum for support? It does work. Maybe I can help there.

<https://community.home-assistant.io/>

<https://discord.gg/c5DvZ4e>

kajmaj, 2018-07-28:
OK. Thanks

closed 2018-07-28 by kajmaj