home-assistant/operating-system #135: Bluetooth logs

danielsjf, 2018-07-27:
Since 1.5, Bluetooth moved to the OS instead of the add-on. Is there any way to generate some logs? I am still experiencing problems, but it is hard to debug. I've tried a few devices but none seem to work. At the moment, I have not other test than adding mac addresses to the configuration.yaml.

Ideally I would find a way to check if there is a connection made with any other Bluetooth device.

pvizeli, 2018-07-27:
You can access to `bluetoothctl` on host.

danielsjf, 2018-07-27:
It might be a dumb question, but how do I use it? I use ssh via the ssh addon, but just typing Bluetoothctl there is not recognised.

rudders, 2018-07-27:
Use the `ssh` instructions from https://github.com/home-assistant/hassos/blob/dev/Documentation/configuration.md and you get to ssh into the the HASSOS then by typing `login` at the `hassio>` prompt `bluetoothctl` will be available.

closed 2018-07-28 by pvizeli