home-assistant/operating-system #142: No /dev/i2c devices (i2c-dev not loaded)

termdude, 2018-08-04:
I have a 1-wire to i2c board based on i2c-bcm2708 which also requires i2c-dev to be loaded. I have managed to load i2c-bcm2708 by adding a reference to it on a CONFIG usb stick, but i2c-dev just won't load and it doesn't show any error messages. The i2c-dev module is available in HassOS.

beewebmeister, 2018-08-06:
Same issues with rpi 3b+ (both 32 and 64 bits), latest HassOS
Please help fixing it.
Thanks.

pvizeli, 2018-08-06:
Can you try a module load config?

pvizeli, 2018-08-06:
https://github.com/home-assistant/hassos/pull/147 (That should work)

pvizeli, 2018-08-07:
ping

termdude, 2018-08-07:
This is exactly what I have tried, but it hasn't worked.

beewebmeister, 2018-08-07:
I followed your advice but it does not work.
I was not able to write in /etc/module-conf.d permanently, after reboot the file disappears.
Thanks.

beewebmeister, 2018-08-08:
I decided to start from beginning with a new microSD.

I downloaded Hassos (hassos_rpi3-1.9.img.gz).
I wrote the image on a microSD (Samsung 32 EVO plus) using Etcher.
I added to config.txt the following:
`dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=pi3-disable-bt
`

then booted. After 20' the system was up.
Just these warnings:

`Timer got out of sync. Resetting
10:32 core.py (ERROR)`
`Setup of config is taking over 10 seconds.
10:32 bootstrap.py (WARNING)`

I installed the ssh (SSH server 3.7) addon from the official addons.
At this point I connected using putty to the host.
The /dev/i2c* is not present.
In the next lines the output of some common  commands:

lsmod:
>Module                  Size  Used by    Tainted: G
nf_conntrack_netlink    36864  0
nfnetlink              16384  2 nf_conntrack_netlink
bnep                   20480  2
hci_uart               36864  1
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             364544 24 bnep,hci_uart,btbcm
ecdh_generic           28672  1 bluetooth
sch_fq_codel           20480  3
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              569344  1 brcmfmac
snd_bcm2835            32768  0
snd_pcm                98304  1 snd_bcm2835
rfkill                 28672  4 bluetooth,cfg80211
snd_timer              32768  1 snd_pcm
snd                    69632  3 snd_bcm2835,snd_pcm,snd_timer
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0

cat /proc/modules:
> nf_conntrack_netlink 36864 0 - Live 0x7f520000
nfnetlink 16384 2 nf_conntrack_netlink, Live 0x7f516000
bnep 20480 2 - Live 0x7f50c000
hci_uart 36864 1 - Live 0x7f4fd000
btbcm 16384 1 hci_uart, Live 0x7f4f6000
serdev 20480 1 hci_uart, Live 0x7f4ec000
bluetooth 364544 24 bnep,hci_uart,btbcm, Live 0x7f480000
ecdh_generic 28672 1 bluetooth, Live 0x7f476000
sch_fq_codel 20480 3 - Live 0x7f46d000
brcmfmac 307200 0 - Live 0x7f347000
brcmutil 16384 1 brcmfmac, Live 0x7f32f000
cfg80211 569344 1 brcmfmac, Live 0x7f0fb000
snd_bcm2835 32768 0 - Live 0x7f0ee000 (C)
snd_pcm 98304 1 snd_bcm2835, Live 0x7f0cb000
rfkill 28672 4 bluetooth,cfg80211, Live 0x7f0a4000
snd_timer 32768 1 snd_pcm, Live 0x7f096000
snd 69632 3 snd_bcm2835,snd_pcm,snd_timer, Live 0x7f07c000
uio_pdrv_genirq 16384 0 - Live 0x7f04c000
uio 20480 1 uio_pdrv_genirq, Live 0x7f010000
fixed 16384 0 - Live 0x7f000000

dmesg |grep i2c:
>nil

ls -l /dev/i2c*:
> ls: /dev/i2c*:  No such file or directory

I added the platform htu21 to the configuration file, the only entry I inserted to the default configuration.yaml.
As expected the error 
> FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1' 

is there.

I confirm that the kernel module folder /etc/modules-load.d is NOT persistent, I added the configuration file in this folder but after rebooting it was not there.

For sure I'm missing something.
Sorry for the long post, I hope this can help you in understanding what is going wrong.

Thanks.

termdude, 2018-08-08:
I managed to get modules to load by using a config usb stick (https://github.com/home-assistant/hassos/blob/rel-1/Documentation/configuration.md#automatic). Remember to name the stick CONFIG (or it won't load), and att your module conf file in /modules. Then restart the HassOS RPI.

pvizeli, 2018-08-08:
@beewebmeister you need connect to host and not with ssh add-on. You can also use  the terminal from hdmi

beewebmeister, 2018-08-08:
Well, at the end I was able to load the i2c module.
I connected to the physical machine (I mean, keyboard, mouse and monitor) and added to the folder /etc/modules-load.d the file rpi-i2c.conf with the following content:
>i2c-dev
i2c-bcm2708

in the config.txt I added
`dtparam=i2c1=on dtparam=i2c_arm=on` 

after rebooting the /dev/i2c-1 is present.
Thanks.

Now I have got a new error:

`HTU21D sensor not detected in bus 1`
`components/sensor/htu21d.py (ERROR)`

I'll take a look to it in the next weeks; maybe a wrong wiring.
Is it possible to install i2c-tools on the host?
How can I install i2c-tools?

Thanks

termdude, 2018-08-09:
My issue has been solved, by manually adding a file in /etc/modules-load.d named rpi-i2c.conf.

When using the config USB stick (as suggested by the documentation), the content from the file in /modules/rpi-i2c.conf was copied to a file /etc/modules-load.d/* (asterisk). This way the modules weren't loaded, so I suspect a bug in the USB stick handling?

brunofosi, 2018-09-29:
> My issue has been solved, by manually adding a file in  named rpi-i2c.conf.
> 
> When using the config USB stick (as suggested by the documentation), the content from the file in /modules/rpi-i2c.conf was copied to a file /etc/modules-load.d/* (asterisk). This way the modules weren't loaded, so I suspect a bug in the USB stick handling?

how did you manage to resolve this? 
manually create `/etc/modules-load.d/rpi-i2c.conf` after booting hassos using hdmi and a usb keyboard as console? and with a CONFIG usb create the config.txt with the follwing:` dtparam=i2c1=on dtparam=i2c_arm=on dtoverlay=pi3-disable-bt` ?

termdude, 2018-09-30:
I simply ssh:d to the host (https://developers.home-assistant.io/docs/en/hassio_debugging.html) and created the file with vi (vi /etc/modules-load.d/rpi-i2c.conf).

brunofosi, 2018-10-01:
after a bit of struggling with the authorized_keys generated by Putty! i managed to ssh hassos and what i did, was rename the * `mv /etc/modules-load.d/* /etc/modules-load.d/rpi-i2c.conf` and it started working

Fusseldieb, 2018-10-12:
Adding those things in config.txt and adding that rpi-i2c.conf made a /dev/i2c-1 device appear, but Home Assistant doesn't recognize it. It throws:

```
Error while setting up platform rpi_gpio_pwm
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity_platform.py", line 128, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=hass.loop)
  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/light/rpi_gpio_pwm.py", line 77, in setup_platform
    driver = Pca9685Driver(pins, **opt_args)
  File "/usr/local/lib/python3.6/site-packages/pwmled/driver/pca9685.py", line 22, in __init__
    self._device = PCA9685(address)
  File "/usr/local/lib/python3.6/site-packages/Adafruit_PCA9685/PCA9685.py", line 74, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/Adafruit_GPIO/I2C.py", line 63, in get_i2c_device
    busnum = get_default_bus()
  File "/usr/local/lib/python3.6/site-packages/Adafruit_GPIO/I2C.py", line 55, in get_default_bus
    raise RuntimeError('Could not determine default I2C bus for platform.')
RuntimeError: Could not determine default I2C bus for platform.
```

I think I'll go back to ResinOS. Reply me if someone figured this out. Thanks.

jatlast, 2018-12-11:
Similar issue here  After upgrading Home Assistant on a Raspberry Pi 3 B+ from 0.75.3 to 0.83.3 my HTU21D sensor is no longer detected.  Any help will be much appreciated.  Thank you.

See below for details.

home-assistant.log
2018-12-11 13:56:58 ERROR (SyncWorker_9) [homeassistant.components.sensor.htu21d] Bad reading: 5
2018-12-11 13:56:58 ERROR (MainThread) [homeassistant.components.sensor.htu21d] HTU21D sensor not detected in bus 1

configuration.yaml
sensor:
# GPIO attached HTU21D Temp/Humidity sensor
  - platform: htu21d
    name: temp_humidity

# Generic Thermostat...
climate:
  - platform: generic_thermostat
    name: Apartment 1 Thermostat
    heater: switch.apartment_1_heat
    target_sensor: sensor.temp_humidity_temperature
    min_temp: 70
    max_temp: 75
    target_temp: 72
    cold_tolerance: 1
    hot_tolerance: 1
    min_cycle_duration:
      seconds: 3
    keep_alive:
      minutes: 3
    initial_operation_mode: "auto"
    # Generic
    away_temp: 65

i2cdetect -y 1
40: 40

ls -l /dev/i2c*
crw-rw---- 1 root i2c 89, 1 Dec 11 01:17 /dev/i2c-1

dmesg | grep i2c
[    5.336012] i2c /dev entries driver

./HTU21D_test
22.08C
22.61%rh

beewebmeister, 2018-12-11:
Hi,

I solved the issue using an old I2C module driver, i2c_bcm2708, instead of the new one.

You can find module and details here:

https://community.home-assistant.io/t/htu21d-on-raspberry-pi/33132/2

Let me stress the point highlighted by termdude; 

"When using the config USB stick (as suggested by the documentation), the content from the file in /modules/rpi-i2c.conf was copied to a file /etc/modules-load.d/* (asterisk). This way the modules weren't loaded".

So, hassos and hassbian teams, please fix both bugs; the wrong driver and the USB stick handling.

Thanks

jatlast, 2018-12-12:
Thank you for the reply.

However, I had already followed the directions of your suggested post when I initially set up my htu21d sensor.

/boot/config.txt
# JTB - Enable HTU21D temperature sensor (https://community.home-assistant.io/t/htu21d-on-raspberry-pi/33132)
dtoverlay=i2c1-bcm2708

ls -la i2c1*
-rwxr-xr-x 1 root root 886 Aug  7 01:53 i2c1-bcm2708.dtbo

As for how I am surviving in the meantime, I rolled back to an old version I had backed up last summer.  It took a while to bring the interfaces up to date, but at least it works and I have heat again.

Home Automation is still for the tech-savvy, I am afraid, i.e., not ready for hoi polloi.

closed 2019-04-23 by stale[bot]