home-assistant/operating-system #145: ModuleNotFoundError: No module named 'pytz.exceptions'

chc189, 2018-08-04:
HASSOS version 1.9 rpi3B(not +)

run 3 commands: ha logs/ha info/su logs ,result like this:
 
hassio > ha logs
Error decoding json invalid character 'T' looking for beginning of value: Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/__main__.py", line 377, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/site-packages/homeassistant/__main__.py", line 352, in main
    args = get_arguments()
  File "/usr/local/lib/python3.6/site-packages/homeassistant/__main__.py", line 85, in get_arguments
    import homeassistant.config as config_util
  File "/usr/local/lib/python3.6/site-packages/homeassistant/config.py", line 16, in <module>
    from homeassistant import auth
  File "/usr/local/lib/python3.6/site-packages/homeassistant/auth/__init__.py", line 6, in <module>
    from homeassistant import data_entry_flow
  File "/usr/local/lib/python3.6/site-packages/homeassistant/data_entry_flow.py", line 6, in <module>
    from .core import callback, HomeAssistant
  File "/usr/local/lib/python3.6/site-packages/homeassistant/core.py", line 38, in <module>
    from homeassistant import loader
  File "/usr/local/lib/python3.6/site-packages/homeassistant/loader.py", line 23, in <module>
    from homeassistant.util import OrderedSet
  File "/usr/local/lib/python3.6/site-packages/homeassistant/util/__init__.py", line 18, in <module>
    from .dt import as_local, utcnow
  File "/usr/local/lib/python3.6/site-packages/homeassistant/util/dt.py", line 9, in <module>
    import pytz.exceptions as pytzexceptions
ModuleNotFoundError: No module named 'pytz.exceptions'

hassio > ha info
{
    "result": "ok",
    "data": {
        "version": "0.75.0",
        "last_version": "0.75.0",
        "machine": "raspberrypi3",
        "image": "homeassistant/raspberrypi3-homeassistant",
        "custom": false,
        "boot": true,
        "port": 8123,
        "ssl": false,
        "watchdog": true,
        "wait_boot": 600
    }
}

hassio > su logs
Error decoding json invalid character '\x1b' looking for beginning of value: 18-01-28 15:59:25 INFO (MainThread) [__main__] Initialize Hassio setup
18-01-28 15:59:25 INFO (MainThread) [__main__] Setup HassIO
18-01-28 15:59:25 INFO (SyncWorker_0) [hassio.docker.supervisor] Attach to supervisor homeassistant/armhf-hassio-supervisor with version 123
18-01-28 15:59:25 INFO (MainThread) [hassio.utils.gdbus] Introspect org.freedesktop.systemd1 on /org/freedesktop/systemd1
18-01-28 15:59:26 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd1
18-01-28 15:59:26 INFO (MainThread) [hassio.utils.gdbus] Introspect org.freedesktop.hostname1 on /org/freedesktop/hostname1
18-01-28 15:59:26 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.hostname1 - /org/freedesktop/hostname1
18-01-28 15:59:26 INFO (MainThread) [hassio.utils.gdbus] Introspect de.pengutronix.rauc on /
18-01-28 15:59:26 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: de.pengutronix.rauc - /
18-01-28 15:59:26 INFO (MainThread) [hassio.host.info] Update local host information
18-01-28 15:59:26 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1
18-01-28 15:59:26 INFO (MainThread) [hassio.host.services] Update service information
18-01-28 15:59:26 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1
18-01-28 15:59:27 INFO (MainThread) [hassio.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
18-01-28 15:59:27 INFO (MainThread) [hassio.host.services] Reload local service hassos-apparmor.service
18-01-28 15:59:27 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1
18-01-28 15:59:27 INFO (MainThread) [hassio.hassos] Detect HassOS 1.9 on host system
18-01-28 15:59:27 INFO (SyncWorker_1) [hassio.docker.hassos_cli] Found HassOS cli homeassistant/armhf-hassio-cli with version 6
18-01-28 15:59:27 INFO (SyncWorker_2) [hassio.docker.interface] Attach to image homeassistant/raspberrypi3-homeassistant with version 0.75.0
18-01-28 15:59:27 INFO (MainThread) [hassio.addons.git] Load addon /data/addons/core repository
18-01-28 15:59:28 INFO (MainThread) [hassio.addons.git] Load addon /data/addons/git/a0d7b954 repository
18-01-28 15:59:28 INFO (MainThread) [hassio.addons] Load addons: 43 all - 43 new - 0 remove
18-01-28 15:59:28 INFO (MainThread) [hassio.updater] Fetch update data from https://s3.amazonaws.com/hassio-version/stable.json
18-01-28 15:59:35 INFO (MainThread) [hassio.snapshots] Found 0 snapshot files
18-01-28 15:59:35 INFO (MainThread) [__main__] Run HassIO
18-01-28 15:59:35 INFO (MainThread) [hassio.misc.dns] Start DNS port forwarding for host add-ons
18-01-28 15:59:35 INFO (MainThread) [hassio.core] Ignore Hass.io auto updates on dev channel
18-01-28 15:59:35 INFO (MainThread) [hassio.api] Start API on 172.30.32.2
18-01-28 15:59:35 INFO (MainThread) [hassio.addons] Startup initialize run 0 addons
18-01-28 15:59:35 INFO (MainThread) [hassio.core] Hass.io reboot detected
18-01-28 15:59:35 INFO (MainThread) [hassio.tasks] All core tasks are scheduled
18-01-28 15:59:35 INFO (MainThread) [hassio.core] Hass.io is up and running
18-01-28 16:00:11 INFO (SyncWorker_6) [hassio.docker] Run command 'python3 -m homeassistant -c /config --script check_config' on homeassistant/raspberrypi3-homeassistant
18-01-28 16:02:46 INFO (SyncWorker_9) [hassio.docker.homeassistant] Start homeassistant homeassistant/raspberrypi3-homeassistant with version 0.75.0
18-01-28 16:02:56 ERROR (MainThread) [hassio.homeassistant] Home Assistant is crashed!
18-01-28 16:02:56 WARNING (MainThread) [hassio.homeassistant] Don't wait anymore of HomeAssistant startup!

chc189, 2018-08-04:
after login into the HASSOS by run command login, and run docker logs containerid(homeassistant/armhf-hassio-supervisor) ,i got the result like this:

18-01-28 16:18:49 INFO (MainThread) [hassio.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
18-01-28 16:18:49 INFO (MainThread) [hassio.host.services] Reload local service hassos-apparmor.service
18-01-28 16:18:49 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1
18-01-28 16:18:49 INFO (MainThread) [hassio.hassos] Detect HassOS 1.9 on host system
18-01-28 16:18:49 INFO (SyncWorker_2) [hassio.docker.hassos_cli] Found HassOS cli homeassistant/armhf-hassio-cli with version 6
18-01-28 16:18:49 INFO (SyncWorker_3) [hassio.docker.interface] Attach to image homeassistant/raspberrypi3-homeassistant with version 0.75.0
18-01-28 16:18:50 INFO (MainThread) [hassio.addons.git] Load addon /data/addons/git/a0d7b954 repository
18-01-28 16:18:50 INFO (MainThread) [hassio.addons.git] Load addon /data/addons/core repository
18-01-28 16:18:50 INFO (MainThread) [hassio.addons] Load addons: 43 all - 43 new - 0 remove
18-01-28 16:18:50 INFO (MainThread) [hassio.updater] Fetch update data from https://s3.amazonaws.com/hassio-version/stable.json
18-01-28 16:18:52 INFO (MainThread) [hassio.snapshots] Found 0 snapshot files
18-01-28 16:18:52 INFO (MainThread) [__main__] Run HassIO
18-01-28 16:18:52 INFO (MainThread) [hassio.misc.dns] Start DNS port forwarding for host add-ons
18-01-28 16:18:52 INFO (MainThread) [hassio.core] Ignore Hass.io auto updates on dev channel
18-01-28 16:18:52 INFO (MainThread) [hassio.api] Start API on 172.30.32.2
18-01-28 16:18:52 INFO (MainThread) [hassio.addons] Startup initialize run 0 addons
18-01-28 16:18:52 INFO (MainThread) [hassio.core] Hass.io reboot detected
18-01-28 16:18:52 INFO (MainThread) [hassio.tasks] All core tasks are scheduled
18-01-28 16:18:52 INFO (MainThread) [hassio.core] Hass.io is up and running
18-01-28 16:19:08 WARNING (MainThread) [hassio.tasks] Watchdog found a problem with Home-Assistant docker!
18-01-28 16:19:08 INFO (SyncWorker_1) [hassio.docker.interface] Clean homeassistant/raspberrypi3-homeassistant docker application
18-01-28 16:19:10 INFO (SyncWorker_1) [hassio.docker.homeassistant] Start homeassistant homeassistant/raspberrypi3-homeassistant with version 0.75.0
18-01-28 16:19:20 ERROR (MainThread) [hassio.homeassistant] Home Assistant is crashed!
18-01-28 16:19:20 WARNING (MainThread) [hassio.homeassistant] Don't wait anymore of HomeAssistant startup!
18-01-28 16:19:20 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished coro=<Tasks._watchdog_homeassistant_docker() done, defined at /usr/local/lib/python3.7/site-packages/hassio/tasks.py:93> exception=HomeAssistantError()>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/hassio/tasks.py", line 107, in _watchdog_homeassistant_docker
    await self.sys_homeassistant.start()
  File "/usr/local/lib/python3.7/site-packages/hassio/utils/__init__.py", line 25, in wrap_api
    return await method(api, *args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/hassio/homeassistant.py", line 279, in _start
    await self._block_till_run()
  File "/usr/local/lib/python3.7/site-packages/hassio/homeassistant.py", line 468, in _block_till_run
    raise HomeAssistantError()
hassio.exceptions.HomeAssistantError

pvizeli, 2018-08-04:
I think your home-assistant images is broken. If you update to 0.75.1 it should be fixed

pvizeli, 2018-08-04:
/move home-assistant/hassio

closed 2018-08-04 by move[bot]