home-assistant/operating-system #159: Waiting for root filesystem on boot with VMWare

nhorvath, 2018-08-12:
Downloaded fresh 1.9 image. Created new 64 bit vm, 2 cores 1 GB ram and used the drive image downloaded. Tried to boot and got stuck at boot menu, saw another issue about deleting the CDROM. So I deleted CDROM and Floppy, then it let me pick System 0 and started to boot but gets stuck waiting for root filesystem.

[image]

mshrem, 2018-08-15:
Change the disk mode to SATA worked for me ....

pvizeli, 2018-08-15:
Yes, we should also add other driver. What is the default disk driver?

nhorvath, 2018-08-16:
Vmware is scsi by default

pronsta, 2018-08-16:
I hope SCSI will be supported as its not working had to change to IDE with proxmox.

closed 2019-04-23 by stale[bot]