home-assistant/operating-system #164: HassOS on SSD

hugodelaroza, 2018-08-18:
Is it possible ton install HassOS on an SSD disk on a Raspberry Pi 3?

Does anybody knows how?

hugodelaroza, 2018-08-18:
Sorry, I can't understand you. Is there an addon that will allow this? or the next version of HASSOS will?

mbo18, 2018-08-20:
As I understand, we will have to wait for the next HassOS release. 
Regarding SSD/eMMC, will this also work for the Asus tinker board S? It has a 16Gb eMMC soldered. To burn an image on it you just have to connect it on a computer with a USB cable and use etcher. So no offload needed.

tdolder, 2018-09-23:
Boot from SSD would be great feature.

tj31moll, 2018-09-28:
With the new update, there is amention of the odroid eMMC commit. I wonder if that means we can use SSD on pi3 bas well.

rbray89, 2018-09-28:
@tj31moll  Unfortunately there would be some effort required to allow non-SD card booting.

shyne99, 2018-11-03:
@rbray89 Can you please develop on this bit ? I was thinking of diving into this matter at some point, SD card are just unreliable. Can you please develop on what is blocking for the moment ? Thank you

rbray89, 2018-11-04:
https://github.com/home-assistant/hassos/commit/960e11dc4f7f32fa3c4937a432ee79d0fe8f1686 was merged to dev. It includes a change for the tinkerboard that should provide the potential for booting off of eMMC, however due to not having the device, the change has yet to be verified.

USB booting is unfortunately usually out of the question, as there exists a chicken-in-the-egg scenario where the USB disk drivers have to be provided at boot.

murphyslaw05, 2018-11-05:
So you are saying that no usb boot support will EVER be added to HassOS? This is a huge reliability complaint from the community.

shyne99, 2018-11-05:
Hi,
Thank you for your feedback.
As a user of SD cards on SBC in a professional way. I can assure you that SSD are way more reliable than SD card. SD card are the less reliable way to go for system partition, and good quality SD card only mitigate the issue, or postpone the inevitable partition corruption .

I think that your way of only putting boot partition o. SD card and lay the rest in SSD is a good solution. And this is actually a mandatory feature for a home automation system (again with reliability in mind)

murphyslaw05, 2018-11-05:
> @murphyslaw05 It's not possible to boot directly from USB for most embedded devices. We'll attempt to boot from eMMC and USB when available, but it isn't always possible. You should note that there ARE plans to allow users to move the data partition onto USB as @pvizeli mentioned above.
> 
> You should also be aware that good quality SD cards often have no issues over long-running installs. (read the tips here: https://www.raspberrypi.org/documentation/installation/sd-cards.md)
> 
> If this continues to be an issue for you, I'd suggest moving to a platform where we support eMMC booting, or an efi x86 install.

Thanks for the suggestions. I have a good quality card, over sized power supply, along with a ups, and do not have corruption issues. However there is no viable way to have any sort of write intensive functionality. I am not familiar enough with the OS structure to know much about the limitations, however I believe that some sort of version of external drive option was available with ResinOS. This is why myself and others are asking.

cholek3, 2018-11-08:
> > @murphyslaw05 It's not possible to boot directly from USB for most embedded devices. We'll attempt to boot from eMMC and USB when available, but it isn't always possible. You should note that there ARE plans to allow users to move the data partition onto USB as @pvizeli mentioned above.
> > You should also be aware that good quality SD cards often have no issues over long-running installs. (read the tips here: https://www.raspberrypi.org/documentation/installation/sd-cards.md)
> > If this continues to be an issue for you, I'd suggest moving to a platform where we support eMMC booting, or an efi x86 install.
> 
> Hi.
> 
> Sorry for my possible lack of knowledge, but for the RPi 3B+ it's possible to boot directly from USB when using Eg. Raspbian - I know hassio is based on another system, but isn't it somehow possible to make a Hassio release for RPi 3B+ that makes it possible to boot from USB like eg. Raspbian?
> 
> And by the way, you are doing a fantastic job. :)
> 
> /Thomas.

I think you are getting confused with terminology here. 
There are two versions of Hass.io. One is based on new system (HassOS) and other one is based on old system (ResinOS). You cannot boot from Hass.io based on HassOS, but you can boot from Hass.io based on ResinOS.

pepeEL, 2019-01-13:
Is any chance to add this feature?

szurr, 2019-01-19:
Hi
Maybe there is a way to put only db files to usb?

mbo18, 2019-01-19:
@berg seems to be working on it: https://github.com/home-assistant/hassos/pull/318

austwhite, 2019-01-31:
Waiting to see result of this one https://github.com/home-assistant/hassos/pull/318 seems to be merged already into Dev

Ayo91, 2019-02-17:
> Is there any progress with this question?
> I really would like to use Hassio (hassos version) from msata with my rpi3b+ :(
> I already lost 2 sd cards (sandisk ultras ...), now hassio is on a sandisk extreme plus, but I don't feel safe :)
> Regards
> Tamás

I have the same problem... I lost 2 sd cards too.. And they were a good brand. After that, I create automatic backups and upload to Dropbox. We need usb boot :(

tikkerei, 2019-02-19:
+1 an usb boot option would be very great, I have unused usb ssd that waits to run as an hassos hard drive

Ayo91, 2019-02-20:
Yes, my Pi has broken 2 Sandisk SD card. After some months they were corrupted.... We need this funcionality...

I have been researching and it seems a solution is install HA in other way around:[https://www.home-assistant.io/docs/installation/raspberry-pi/]( https://www.home-assistant.io/docs/installation/raspberry-pi/)

Any feedback?

jvanderneutstulen, 2019-02-20:
The main issue the data partition which gets written to a lot.  The PR from @berg should be enough to move the data partition to an external drive. Basically it's two steps:
1. Create ext4 partition with label `hassos-data` on the external drive.
2. Rename the partition `hassos-data` on the sdcard to something else.

With the RPI 3 you can (permanently) enable usb boot mode, but this will probably require changes in the uboot configuration, so a different image for usb and mmc. As an alternative it may be posssible to create a small usb-boot uboot image, so the bootloader lives on the sdcard and the other stuff is on the external disk. This principle can work with other targets as well.

austwhite, 2019-02-20:
@Ayo91 
You are correct that many other methods exist.  This is specifically for those that prefer the HassIO method using the self contained HassOS image.
🙂

@jvanderneutstulen
The boot loader is an interesting idea as it would rarely need to be anything other than a read only device.  
I wonder if maybe that is an option @Berg could work with rather than fully moving it all to a USB device.
You are right that the MMC and USB would likely need to be separate images, unless HassOS can be accept a txt config file with an option that could be toggled between USB and MMC boot modes

Ayo91, 2019-02-20:
> I wonder if maybe that is an option @berg could work with rather than fully moving it all to a USB device.
> You are right that the MMC and USB would likely need to be separate images, unless HassOS can be accept a txt config file with an option that could be toggled between USB and MMC boot modes

It will be awesome support that.

Ayo91, 2019-02-20:
> The main issue the data partition which gets written to a lot. The PR from @berg should be enough to move the data partition to an external drive. Basically it's two steps:
>
> 1. Create ext4 partition [...]

Has someone tested that? Because my HA is in live, its a monitoring system and an alarm and I dont want to break anything and stop the system. Should I migrate the content of the SD hassoss-data to the  
new partition that I will create?

> With the RPI 3 you can (permanently) enable usb boot mode, but this will probably require changes in the uboot configuration, so a different image for usb and mmc. As an alternative it may be [...]

Some instructions for do that? small usb-boot uboot image and the rest in the SSD.

Thanks.

tboyce021, 2019-02-20:
> Has someone tested that? Because my HA is in live, its a monitoring system and an alarm and I dont want to break anything and stop the system. Should I migrate the content of the SD hassoss-data to [...]

I just tried this on my NUC (running HassOS VMDK on Proxmox) and it seems to have worked. I booted to an Ubuntu live USB and used gparted to rename/relabel the old partition to `hassos-data-orig` and create the new partition on a separate disk. Then I copied the data over using
```
sudo dd if=/dev/sda8 of=/dev/sdb1 bs=4096 conv=notrunc,noerror,sync
```
(update to use your paths)

That seems to have changed the label on the new partition to match the old one (`hassos-data-orig`), so I used gparted again to relabel the new partition to `hassos-data`.

Rebooted HassOS, briefly saw a message about it resizing the disk/partition, and can now see the new size with `df -h /mnt/data`.

You may be able to do this on a running system, but this was easier for me since I could ensure everything was shut down and not writing to the disk when I copied the contents.

austwhite, 2019-02-20:
@tboyce021 
Did you try the standard image as well as the virtual disk image?
In theory they should be the same, but the virtual disk image may have some tweaks in it. 
🙂

tboyce021, 2019-02-25:
I only tried the VMDK. However, after just updating to 2.10, my HassOS VM won't boot anymore. Not sure if it's related to moving the data partition or not though.

Ayo91, 2019-02-27:
I deleted HassOs. Bad performance ever. I installed a Ubuntu server in a computer with a consumption of 25W and I'm very happy. Everything works nice and very stable.  No more SD broken.

iLLiac4, 2019-03-13:
Hi. Is this already available in the latest image?
I have tried what tboyce021 wrote but it does not boot.
I there any official statement how to achive that on hassio?

iLLiac4, 2019-03-14:
Hi.
To answer my own question :)
Yes it is possible to migrate data from sd card to ssd.

I have tested few scenarios and they do work.

1. Clean install:

- create sdcard with winimg32. 
- With partition manager (i was doing with gparted live usb, note that i doubt that any windows partition manager will work).
- Create ext4 partition on the external drive (SSD).
- Clone partition from created sd card that is named hassos-data. When inserting created sd card into gparted the program will offer you to fix the partition. Do not let gparted fix the partition!!!!
sudo dd if=/dev/sda8 of=/dev/sdb bs=4096 conv=notrunc,noerror,sync
The above command is very important!!!! (do not use of=/dev/sdb1 but sdb) Note that sda or sdb can differ from system to system.
- After cloning label partition on sd card from hassos-data to hassos-data-old. Cloned partition label has to be labeled hassos-data on SSD.
- Connect everything to RPi and it will install Hassio onto SSD.

2. Migration:
- Create ext4 on SSD with gparted.
- Clone hassos-data from your production sd card to SSD with command:
sudo dd if=/dev/sda8 of=/dev/sdb bs=4096 conv=notrunc,noerror,sync
- After clonning label sd card hassos-data to hassos-data-old.
- On SSD label of data partition has to be hassos-data.
- To extend the data partition of newly cloned ssd select from gparted menu Partition/Check and it will extend it.
- Insert sd card into rpi and newly created ssd and it will boot.

tboyce021, 2019-03-14:
> do not use of=/dev/sdb1 but sdb

Awesome find. For me, /dev/sdb1 would work fine initially, but subsequent boots would fail. With /dev/sdb, reboots seem to be working again.

FYI to others, you can add `status=progress` to the end of the `dd` command to get progress updates while it runs (assuming your version of `dd` is new enough).

bbbenji, 2019-03-14:
Somethings to note:
- Formating the new drive as ext4 is unnecessary.
- For me the original `hassos-data` partition is `/dev/sda4` NOT `/dev/sda8`.
- Resizing the new `hassos-data` partition is NOT optional. I was stuck on "Preparing Hass.io" if I didn't resize the new partition.

iLLiac4, 2019-03-15:
Ehat do you mean with this?
DO NOT CLICK UPDATE UNDER HOST SYSTEM CARD

Do you mean that the hassio can not be updated after migration to external ssd?

bbbenji, 2019-03-15:
No. I mean that until a new stable build is released, that button will downgrade you to 2.10.

tboyce021, 2019-03-15:
Maybe add (if you're on 3.0+) to clarify. I did that same thing when on a development build once. But I'm currently on 2.10 and it still works fine.

iLLiac4, 2019-03-15:
I was doing those testing on 2.10 image.
I did not see a 3.0+ images at all.

nukefrenzy, 2019-05-20:
I have followed iLLiac4's instructions for a new install after my SD card failed for the 2nd time.  However, the RPi shows no signs of life except for the red power light.  The green activity light never blinks even once.  A fresh Hassio install will work just fine.  Any idea's?

I used gparted from a USB drive to copy the data folder and followed your steps to the letter.

Jpsy, 2019-06-04:
> However, the RPi shows no signs of life except for the red power light. The green activity light never blinks even once. 

I had the exact same effect. Turned out that I had not changed the LABEL of the SD card data partition to hassos-data-old, but the NAME! Gparted allows to change both. You may have to unmount the partition before you can change the label. Unmounting is also possible through Gparted.

pepeEL, 2019-06-24:
When will be implement boot from SSD USB to image HASS.io ?

Misiu, 2019-06-25:
Raspberry PI 4 is released and many of us are wondering if this scenario will be possible some day - download HASSIO image, flash it to Pen Drive or SSD and use that instead of SD Card.

pepeEL, 2019-06-25:
I buy RPI4 and wait for it. I also can be beta tester if it is need.

esciara, 2019-06-26:
Hi guys,

Just to share my experience of trying @iLLiac4 's steps directly on a RPI. I did it from a brand new install. I guess it can work with an existing setup, but I have not tried, and my knowledge of linux is too limited to vouch for that.

### Full procedure
First I followed the [Install Hassio doc](https://www.home-assistant.io/hassio/installation/) to create my SD card with balenaEtcher (using the latest `hassos_rpi3-2.12.img` image) on a mac for a Raspberry Pi 3 Model B+ (32bit as recommended on the doc). Also did the optional part to get a static IP. And setup SSH on port 22222 as explained on the [Debugging Hassio dev doc](https://developers.home-assistant.io/docs/en/hassio_debugging.html#ssh-access-to-the-host).

Launched my RPI with the "CONFIG" USB stick and a USB drive (target for holding all my Hassio data). All starts normally and as explained in the Debugging Hassio doc, I can SSH from my mac with a `ssh root@hassio.local -p 22222` and access the host system use the 'login' command.

From there, `lsblk` gives me a picture of the partitions:
```
# lsblk -o name,mountpoint,label,size,uuid
NAME        MOUNTPOINT   LABEL             SIZE UUID
sda                                        1.9G
`-sda1                   CONFIG            1.9G C41D-13F5
sdb                                      465.8G
mmcblk0                                    7.4G
|-mmcblk0p1 /mnt/boot    hassos-boot        32M 25BC-1A86
|-mmcblk0p2              hassos-kernel      24M b3888eda-c63e-472d-a697-388c2a6d74be
|-mmcblk0p3 /                              256M
|-mmcblk0p4                                 24M
|-mmcblk0p5                                256M
|-mmcblk0p6                                  8M
|-mmcblk0p7 /mnt/overlay hassos-overlay     96M c3204cd3-41c0-456d-9305-e01c8e14d9cb
`-mmcblk0p8 /mnt/data    hassos-data       6.7G 73557961-3e93-47cd-ad4f-9c7751ac28d9
zram0       [SWAP]       hassos-zramswap 185.2M 16dad8e5-07b1-49e3-8711-ff4afbe159a6
zram1       /var         hassos-var         32M 1dba1baa-3905-463e-97e7-4ea75a057c40
zram2       /tmp         hassos-tmp         16M 45e52160-1bee-4ae8-a1c8-023cfb87da1a
```
So in my case, the partition I want to clone is `/dev/mmcblk0p8`. That is what I do:
```
# dd if=/dev/mmcblk0p8 of=/dev/sdb bs=4096 conv=notrunc,noerror,sync
```
At that point, the USB drive has the same `hassos-data` label as the SD card. I relabel the SD `hassos-data` to `hassos-data-old`:
```
# e2label /dev/mmcblk0p8 hassos-data-old
```
Checked with `lsblk` to see if all is fine, and it seems to be: (note that `sdb` seem to have changed name to `sda` in the process... pretty sure I did not change USB ports)
```
# lsblk -o name,mountpoint,label,size,uuid
NAME        MOUNTPOINT   LABEL             SIZE UUID
sda         /mnt/data    hassos-data     465.8G 73557961-3e93-47cd-ad4f-9c7751ac28d9
sdb                                        1.9G
`-sdb1                   CONFIG            1.9G C41D-13F5
mmcblk0                                    7.4G
|-mmcblk0p1 /mnt/boot    hassos-boot        32M 25BC-1A86
|-mmcblk0p2              hassos-kernel      24M b3888eda-c63e-472d-a697-388c2a6d74be
|-mmcblk0p3 /                              256M
|-mmcblk0p4                                 24M
|-mmcblk0p5                                256M
|-mmcblk0p6                                  8M
|-mmcblk0p7 /mnt/overlay hassos-overlay     96M c3204cd3-41c0-456d-9305-e01c8e14d9cb
`-mmcblk0p8              hassos-data-old   6.7G 73557961-3e93-47cd-ad4f-9c7751ac28d9
zram0       [SWAP]       hassos-zramswap 185.2M 9a2aa25f-6450-45db-a89f-5b959e36a9f6
zram1       /var         hassos-var         32M cc6cdd18-9615-43ce-b206-e8ab4232c32d
zram2       /tmp         hassos-tmp         16M 9f729146-0ac1-46e0-8b2d-85a49fded628
```
Since I am not so good in linux (so I don't really know how to check it smartly), I guess this all means that it has done things as expected. In any case, when I am starting the RPI without the USB drive, it would not work... Whereas it works just fine when the USB drive is plugged.

### Finishing notes
#### Trying to unmount `hassos-data-old`
Tried to unmount `/dev/mmcblk0p8` with `fdisk` to make sure it was not used, but `fdisk` talking to me about some complicated hybrid stuff:
```
# fdisk /dev/mmcblk0

Welcome to fdisk (util-linux 2.32.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

A hybrid GPT was detected. You have to sync the hybrid MBR manually (expert command 'M').

Command (m for help):
```
That didn't encourage me try further.

#### Extend the data partition of newly cloned `hassos-data` on USB drive
This was suggested in @iLLiac4 's steps. Is it really required? If yes, is it possible to do it without GParted?

__Any feedback welcome!__

pvizeli, 2019-06-28:
@esciara look like you know how a Linux system/partition work.

Add a script to /usr/sbin they do that automatically if you set a device as a parameter. That the expander script going wrong, is a bug. Maybe you can fix it? 

Like: https://github.com/home-assistant/hassos/blob/dev/buildroot-external/rootfs-overlay/usr/libexec/hassos-expand
Use `sgdisk` for GPT and `sfdisk` for MBR (on the main harddisk). The new Data partition Harddisk should be all-time GPT because we don't need boot from it.

esciara, 2019-07-09:
Sorry for the late reply.

@pvizeli Google is my friend :stuck_out_tongue_winking_eye: . Put what I could find together (with enough understanding of it.

> Add a script to /usr/sbin they do that automatically if you set a device as a parameter. That the expander script going wrong, is a bug. Maybe you can fix it?

Not sure I understood. Do you mean there is a bug somewhere in what I wrote? Or are you talking about adding a new feature? Can you clarify, please?
 
> Like: https://github.com/home-assistant/hassos/blob/dev/buildroot-external/rootfs-overlay/usr/libexec/hassos-expand
> Use `sgdisk` for GPT and `sfdisk` for MBR (on the main harddisk). The new Data partition Harddisk should be all-time GPT because we don't need boot from it.

I actually don't plan to touch the partitions... I prefer to do as less modifications as possible to avoid breaking things. Unless you think I should or that it should be done for it to work?

Misiu, 2019-07-12:
@pvizeli is something going on to allow booting and running only from SSD/Pendrive?
Raspberry Pi3, Pi3B+ and soon Pi4 are/will be able to run from an external drive without the need to use SD card at all.
The ideal setup would be to download an image, flash it to SD/SSD/Pendrive plug it in and that's all. No need to manually edit files, move partitions.

Is this even doable for HassOS?

esciara, 2019-07-12:
@Misiu : I did quite a bit of research on that, and the conclusion was that HassOS did not allow for that. Don't have the references at hand (please share if you find them), but I remember the alternative was about installing Hassio on raspbian, which has pros and cons, but is definitely more prone to having issues, since HassOS does a lot of work for you...

rbray89, 2019-07-12:
@Misiu @esciara:

Essentially, what we'd have to do is rework uboot and it's boot script `buildroot-external/board/raspberrypi/rpi3-64/uboot-boot.ush` to detect what we're booting off of and handle the two boot cases differently. You'd also need to follow this procedure on the rpi: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

I have a spare RPI 3B that I can start investigating this. Initially, I thought that this may be impossible for a (pure) USB boot, but it seems that this may not be the case on the RPI given that it has it's own bootloader in eprom that can load bootcode off the USB drive.

The crux of this, is that the USB device will need to have support in uboot. This is going to cause a LOT of support tickets for us going forward as people are going to throw the image on the USB stick and it won't boot cause the specific drive they chose isn't supported in uboot.

Misiu, 2019-07-16:
@esciara I'm aware that this is complicated if it weren't then it would already be added 😁
@rbray89 thanks for the update.
USB stick and SSD support would be awesome. I wasn't aware that this can cause so much potential trouble with all the devices.
Nonetheless, having support for a single ssd and usb stick would be a massive step forward.
Raspbian doesn't work with all possible hardware, there are many forum posts about that (for example https://www.raspberrypi.org/forums/viewtopic.php?t=210497) and people must live with that.
Personally, I can buy a specific ssd drive/usb stick if only it would be supported and I think many users have the exact same opinion 🙂

esciara, 2019-07-16:
@rbray89 
> The crux of this, is that the USB device will need to have support in uboot. This is going to cause a LOT of support tickets for us going forward as people are going to throw the image on the USB stick and it won't boot cause the specific drive they chose isn't supported in uboot.

Maybe writing a step by step somewhere explaining how to verify that the usb stick is uboot and other steps ? Then refer to this if people open an issue and asking to make sure they verified everything written before answering their issue?

I quickly searched but could not find useful links (but that was a very quick search...)

Misiu, 2019-07-16:
> Maybe writing a step by step somewhere explaining how to verify that the usb stick is uboot and other steps ? Then refer to this if people open an issue and asking to make sure they verified everything written before answering their issue?

That would be very useful and for sure would prevent a LOT of support tickets.

pvizeli, 2019-07-16:
I think a script/wizard they offload the data partition is enought because al other are readonly...

Misiu, 2019-07-17:
@pvizeli I don't mind having an SD card in the slot and having everything else on USB Thumb Drive or an SSD.
The ideal solution would be full support for booting off USB Thumb Drive or SSD drive.
Let's wait for @rbray89, maybe he'll have success with it.

rbray89, 2019-07-22:
@esciara @Misiu @pvizeli 

Was able to get standalone USB booting on my RPI 3 B V1.2 working. Don't have other boards to test, but I don't see why any of them shouldn't work. 

See https://github.com/rbray89/hassos/blob/rpi_usb_boot/Documentation/boards/raspberrypi.md#usb-boot for more info.

Pull request: https://github.com/home-assistant/hassos/pull/440
dev build: https://github.com/rbray89/hassos/releases/tag/3.4-rpi-usb-boot

pepeEL, 2019-07-22:
But good if it will be available also for RPi4.

rbray89, 2019-07-22:
@pepeEL I **think** it should work, but have to RPi4 to verify.

pepeEL, 2019-07-22:
but there is no any info about rpi 4... only to works to rpi 3b+

pepeEL, 2019-07-22:
please also build image for RPI4 and i can test it... and share this image in your repo...

rbray89, 2019-07-23:
@pepeEL Looks like the FW bootloader does not support USB for the Pi4 yet:
https://www.raspberrypi.org/forums/viewtopic.php?t=243995
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

pepeEL, 2019-07-23:
Yes I read. But please share also image for RPI3b+. I test...

rbray89, 2019-07-23:
@pepeEL , the release I posted earlier is the RPI3 beta: 
https://github.com/rbray89/hassos/releases/tag/3.4-rpi-usb-boot

Misiu, 2019-07-25:
@pepeEL did You try this release? If yes then please share Your results.

pepeEL, 2019-07-25:
No i don't try because it is for RPi3. I have a rpi4.

[131 more replies]

closed 2020-06-08 by pvizeli