home-assistant/operating-system #169: Increase available storage with VMDK

nhorvath, 2018-08-21:
How do I add storage to HassOS running in VMware?

NK553, 2018-08-27:
I use VirtualBox and I had to increase the size as well.
@pvizeli can we make the default size larger?  It requires a lot of steps to increase it later.

nhorvath, 2018-08-27:
What were the steps you used to make it usable to HassOS after you grew the
virtual disk?

mihalski, 2018-08-29:
The default size is 6GB right?
With VMware my process is as follows:
1. Download HassOS image and extract vmdk.
2. Create new VM in Fusion/Workstation with UEFI boot and the HassOS vmdk.
3. Edit settings (such as disk size) to my preferences and apply. Note: Disk must be IDE and not SCSI.
4. Export OVA (NEVER START THE VM).
5. Upload OVA to VMware ESXi server (could probably also upload via Fusion/Worstation)
6. Start VM.

If you intend to run in the desktop version of VMware (Fusion/Workstation) then just skip steps 4 and 5.

nhorvath, 2018-08-29:
Ok, but if I have already started it what can I do?

mihalski, 2018-08-29:
You loose nothing from starting again but potentially days of messing around without success if you try resizing an existing install.

mihalski, 2018-08-29:
FYI. When you restore a snapshot the add-ons will need to download from their repositories, therefore if you have added a repository and installed an add-on from it . you will first need to re-add the repository to the new install. And don't forget to give the add-ons time to download and re-install.

nhorvath, 2018-08-29:
So you mean if I snapshot and restore do I need to manually reinstall the
addons or just the repo? If I'm just using the normal community repo I just
need to restore the snapshot right?

mihalski, 2018-08-29:
Before you restore the snapshot, add any repo that you have installed an add-on from. If you are using the included repos then all you have to do is wait for the add-ons to redownload and install. The configs will be preserved. I learnt all this today when following my own instructions.

nhorvath, 2018-08-29:
Ok thanks for the help, I'll try that. I still think a method for growing
an already created disk would be beneficial though.

mihalski, 2018-08-29:
I've done it before (not in hassos though) and it's always been a major PITA. You need to resize the drive in VMware, then boot off an iso image such as parted-live and shift around partitions.. I don't think it's worth the effort as even if it was supported it would take LONGER than a snapshot and reinstall.

mihalski, 2018-08-29:
I think there at LEAST 8 partitions, and I doubt the one we want to resize is last. Therefore it's a matter of resizing the extended, moving a lot go partitions and THEN resizing the one we want to enlarge.

nhorvath, 2018-08-29:
Oh, i see. that does complicate things.

nhorvath, 2018-08-29:
It would be great if you could just mount a second disk that automatically
gets mapped to store data.

mihalski, 2018-08-29:
That MAY be something that's in the works as using a USB stick on the rpi for data is something that has been hinted at. Could translate just as easily to a virtual machine.

nhorvath, 2018-08-29:
Yeah, instead of just detecting usb, detect /dev/sdb or something like that
and mount it the same way.

mihalski, 2018-08-29:
Or just base it on volume name.

nhorvath, 2019-04-01:
I put this off because I didn't want to risk screwing up addons. I finally did it by resizing following these steps with no issues (instructions for virtualbox, but should translate to other vm hosts):
1. Go to Hass.io System Tab in web interface and update Hass.io Host System (2.11 at this writing)
2. After it comes back up, click shutdown.
3. Resize the virtual disk 
```bash
cd /opt/virtualbox/HassOS
vboxmanage clonemedium hassos_ova-1.9.vmdk hassos.vdi --format vdi
vboxmanage modifymedium hassos.vdi --resize 51200
vboxmanage clonemedium hassos.vdi hassos.vmdk --format vmdk
```
Note: change 51200 to some other number if you don't want a 50GB disk. Paths and original filename will vary with your setup.
4. In virtualbox change the hard disk from the original vmdk to the new one.
5. Download gparted live cd from here: https://gparted.org/download.php and mount the iso.
6. Turn on the VM and boot into the gparted live cd.
7. If/When asked at startup if you want to fix the GPT table to use the entire disk say "Fix".
8. Resize partition /dev/sda8 to fill the disk and click apply.
9. Reboot and verify you now see all the storage!
10. Clean up by deleting the old vmdk and intermediate vdi file.

I'd still like to keep this issue open as a feature request to be able to just add another VMDK that gets used as storage, or automatic resizing at boot by just expanding the size of the original VMDK.

stratplayer, 2019-04-05:
@nhorvath thanks for the above, worked a treat. I had the same problem and followed this guide above to fix the issue :-)

ryanupp, 2019-04-09:
Thanks @nhorvath, this worked for me as well, expanded to 40gb without any problems.

For anyone else who ends up here I'm running Hass.io as a VM in ESXi 5.5, and in my case there was no need to go to the intermediate vdi format.  I just cloned the vmdk, expanded it, then modified my VM's config to use the new disk and continued with the steps above using gparted.

SSH into ESXi
#cd into your VM's directory
*cd /vmfs/volumes/datastore/hassio/*
#clone existing vmdk
*vmkfstools -i hassio.vmdk hassio_clone.vmdk*
#resize cloned vmdk to 40GB
*vmkfstools -X 40g hassio_clone.vmdk*
#rename vmdk's if desired
*vmkfstools -E "originalname.vmdk" "newname.vmdk"*

carloscae, 2019-04-15:
For anyone who's is struggling with step 5 and 6 of @nhorvath instructions above, on how to attach and boot Gparted:

- After downloading the ISO, go to Virtual Media Manager and load it on the Optical Disk section.
- Then attach into your Hassio VM in Settings/Storage under Sata Controller.
- When you boot the Hassio VM it should boot the Partition Manager now.

There will be a couple of straight forward dialogs until you get to the GUI where is pretty simple to resize the image mentioned above.

Maybe this might seem unnecessary explanation, but since it took me a while to figure out, maybe some people also have the same issue.

Thanks for the solution, @nhorvath :)

closed 2019-06-21 by stale[bot]

fluppie, 2020-03-06:
I think the default size should be at least 20-25 GB. New installation with some add-ons and I already have a full disk, which gives issues with the log files and unifi controller.

zeljko-ciric, 2020-03-25:
I have same problem. Initial disk was 6GB. Increasing disk size in Hyper V and reboot system solve issue.

pvizeli, 2020-03-25:
Yep, it resize only on reboot

kbruurs, 2020-10-24:
> Thanks @nhorvath, this worked for me as well, expanded to 40gb without any problems.
>
> For anyone else who ends up here I'm running Hass.io as a VM in ESXi 5.5, and in my case there was no need to go [...]

I used this method using HassOS 4.14 and Home Assistant 0.116.4 and it worked well.

I didn't use gparted as I read that hassos should automatically use the max harddisk size and this worked for me (at least it reports my new harddrive size now, so I guess it works).

bobhavlin, 2020-10-29:
Thanks for documenting this!

You can also do step 3 with Virtualbox's "Virtual media manager" to first "clone" and then "increase" before resizing the partition with Gparted.  Also, no need to change formats from VDI to VMDK (confirmed, I just did it this way).  Probably slightly easier for noobs :)

> I put this off because I didn't want to risk screwing up addons. I finally did it by resizing following these steps with no issues (instructions for virtualbox, but should translate to other vm hosts):
> 
> 1. Go to Hass.io System Tab in web interface and update Hass.io Host System (2.11 at this writing)
> 2. After it comes back up, click shutdown.
> 3. Resize the virtual disk
> 
> ```shell
> cd /opt/virtualbox/HassOS
> vboxmanage clonemedium hassos_ova-1.9.vmdk hassos.vdi --format vdi
> vboxmanage modifymedium hassos.vdi --resize 51200
> vboxmanage clonemedium hassos.vdi hassos.vmdk --format vmdk
> ```
> 
> Note: change 51200 to some other number if you don't want a 50GB disk. Paths and original filename will vary with your setup.
> 4. In virtualbox change the hard disk from the original vmdk to the new one.
> 5. Download gparted live cd from here: https://gparted.org/download.php and mount the iso.
> 6. Turn on the VM and boot into the gparted live cd.
> 7. If/When asked at startup if you want to fix the GPT table to use the entire disk say "Fix".
> 8. Resize partition /dev/sda8 to fill the disk and click apply.
> 9. Reboot and verify you now see all the storage!
> 10. Clean up by deleting the old vmdk and intermediate vdi file.
> 
> I'd still like to keep this issue open as a feature request to be able to just add another VMDK that gets used as storage, or automatic resizing at boot by just expanding the size of the original VMDK.

Megachip, 2020-11-25:
> I have same problem. Initial disk was 6GB. Increasing disk size in Hyper V and reboot system solve issue.

so not necessary anymore to go into the system? Just increase vmdk and reboot?

SixFive7, 2021-05-15:
I'm experiencing the same issue on a perfectly stable host OS. Ticket:
https://github.com/home-assistant/operating-system/issues/1362

ScottG489, 2022-01-14:
It seems like these solutions assume you have VBox GUI access. I'm running it on a headless server. So I don't believe the gparted solution will work for me.

Does anyone have any suggestions for how to do a resize purely via the command line?

nhorvath, 2022-01-14:
You could enable the vbox vnc server which lets you remote console in to
the vm.

ScottG489, 2022-01-14:
Another option seems to be that you can just create a new VM and attach the storage to that.

Using a new VM may not be an option for all, but the only issue I ran into was that the IP was different so I had to update my mqtt devices to point at the new IP (broker is installed via the add-on) and then change port forwarding on my router.

That ended up being much less work for me and more importantly may be automated and as a HA user I think we can all appreciate that :)