home-assistant/operating-system #170: Unable to upgrade from 0.75.3 to 0.76.2, upgrader just auto rolls back

bagobones, 2018-08-22:
I have tried a few times now with 0.76.1 and 0.76.2 I have tried purging the DB and repacking to avoid upgrade time etc. Everything is running on a regular PI 3

However it seem to keep timing out waiting for HASS to come back up so it just rolls back the upgrade.

bagobones, 2018-08-22:
Of note it looks like it only waits 10 min which is FAR less than the DB upgrade time of 15 min? Suggested for this release.

pvizeli, 2018-08-22:
Correct, but we can detect running migration with hass.io 127 and I don't see one.
What is the ouput of `hassio ha logs` after 3-4min?

bagobones, 2018-08-23:
Looks like it was this breaking change

HTTP: Make sure use_x_forward_for and trusted_proxies must be configured together (@awarecan - #15804) (http docs) (breaking change)

```
Please check your config. @ 2018-08-22T19:57:18.740908-07:00>, new_state=<state persistent_notification.invalid_config=notifying; title=Invalid config, message=The following components and platforms could not be set up:
 - [http](https://home-assistant.io/components/http/)
 - [system-log](https://home-assistant.io/components/system_log/)
 - [auth](https://home-assistant.io/components/auth/)
 - [onboarding](https://home-assistant.io/components/onboarding/)
 - [api](https://home-assistant.io/components/api/)
 - [websocket-api](https://home-assistant.io/components/websocket_api/)
 - [frontend](https://home-assistant.io/components/frontend/)
 - [history](https://home-assistant.io/components/history/)
 - [mqtt](https://home-assistant.io/components/mqtt/)
```
Looks like checking for a valid config or errors in config validation would be good to surface to the user as well.

More so than just:
`18-08-22 09:21:17 WARNING (MainThread) [hassio.homeassistant] Don't wait anymore of HomeAssistant startup!`

bagobones, 2018-08-23:
HA will fail to start if  use_x_forward_for is configured but trusted_proxies is not present.. It might be a better fail condition to ignore use_x_forward for if trusted proxies is not present and show an error on startup similar to the embedded MQTT server password change on this release.

Going to mark this as closed as I was able to upgrade but better failure detection / noticed in the HASOS log is still probably a good idea as future releases will likely also have startup failing config changes.

closed 2018-08-23 by bagobones