home-assistant/operating-system #178: hassio commandline and Odroid C2 (beta)

iuro, 2018-08-26:
This doesn't work in the beta version.

Issuing:

core-ssh:~# hassio ho update

returns:

Error decoding json invalid character ':' after top-level value: 404: Not Found

rbray89, 2018-08-26:
are you trying to use the SSH addon or the built-in SSH (documented in [https://github.com/home-assistant/hassos/blob/dev/Documentation/configuration.md](the configuration docs)?

iuro, 2018-08-26:
64 bit. I will try reinstalling the add-on.

iuro, 2018-08-29:
I reinstalled the add-on but the issue remains. Still not possible to issue the command:

hassio ho update

iuro, 2018-08-29:
more detailed errors: `# hassio -d ho update
DEBUG [CmdHost]: action->'update', endpoint='update', serverOverride->'', GET->'false', options->'', rawjson->'false', filter->''
DEBUG [ExecCommand]: basepath->'host', endpoint->'update', serverOverride->'', get->'false', Options->'', Filter->'', RawJSON->'false'
DEBUG [GenerateURI]: basepath->'host', endpoint->'update', serverOverride->''
DEBUG [RestCall]: data->'http://hassio/host/update', GET->'false', payload->''
DEBUG [RestCall]: ResponseBody->'404: Not Found'
Error decoding json invalid character ':' after top-level value: 404: Not Found`

iuro, 2018-08-30:
Read on twitter that this issue is a general issue in hassos and not only a odroid c2 specific one.