home-assistant/operating-system #179: Let's Encrypt add-on and Odroid C2 (beta)

iuro, 2018-08-26:
Cannot renew the cert. Paths seems to be wrong.

Snippet from log (replace my domain with generic host.yourdomain.com)

starting version 3.2.4
Saving debug log to /var/log/letsencrypt/letsencrypt.log
-------------------------------------------------------------------------------
Processing /data/letsencrypt/renewal/host.yourdomain.com.conf
-------------------------------------------------------------------------------
expected /data/letsencrypt/live/host.yourdomain.com/cert.pem to be a symlink
Renewal configuration file /data/letsencrypt/renewal/host.yourdomain.com.conf is broken. Skipping.
-------------------------------------------------------------------------------
No renewals were attempted.
Additionally, the following renewal configuration files were invalid: 
  /data/letsencrypt/renewal/host.yourdomain.com.conf (parsefail)
-------------------------------------------------------------------------------
0 renew failure(s), 1 parse failure(s)

rbray89, 2018-08-26:
was this on a clean install? Or did you migrate anything from another install?

iuro, 2018-08-29:
Removing the add-on and reinstalling resolved the issue.

rbray89, 2018-08-29:
Closed-per comment. An issue should probably be created for HassIO for backup and restore.

closed 2018-08-29 by rbray89