home-assistant/operating-system #182: Configuration without a USB Stick

urish, 2018-08-27:
I installed Hass.io OS on my Pi Zero W, and didn't have an extra USB stick, so I found the following workaround for configuring the network through my windows machine:

1. Rename the boot partition from `hassos-boot` to `CONFIG`
2. Put the network configuration file inside `network/my-network` as outlined in the [documentation](https://github.com/home-assistant/hassos/blob/dev/Documentation/network.md).

I believe this workaround can be useful for other users who want to set up WiFi or a static IP address but don't have an available USB stick. If you wish, I can send a PR for the documentation suggesting this workaround.

pvizeli, 2018-08-27:
I'm open for PR's they allow to fallback into boot partition folder.

urish, 2018-08-27:
What do you think about the following procedure?

1. If /boot/config exists, mount it over /mnt/config (using bind)
2. If we have a config partition, mount it over /mnt/config as overlay mount
3. Then load all configuration files from /mnt/config as we do today

thesoftwarejedi, 2018-09-03:
@urish Thank you!  After failing to get the Pi Zero to read several USB sticks, this method worked great.

pvizeli, 2018-09-13:
@urish 

1. Check if a USB stick exists -> process
2. Otherwise look if a config partition exists -> process -> remove config

borpin, 2018-09-18:
Can I add a couple of cents worth, I am trying to change an instance of Hassio running on a VM to a static IP.  I cannot find the boot partition. I have tried putting the configuration in a 'my-networks' file in `\etc\networks\` but each time it gets deleted and nothing has changed.  I realise this is a slightly different issue, but if something was being implemented this use case might be considered.

urish, 2018-09-21:
@pvizeli thank you for the feedback! What is the recommended procedure for testing my changes? Is there some kind of QEMU setup where I can check it without building and writing an image to the Pi every single time?

PR #212 (2018-10-25): OS: Config importer support now also boot partition
fix: #182 

Device like RPi zero have no USB stick. It allow also to simple set a config with boot partition (Fat32) after write a the SD card and need no extra device.

With support config per boot partition and with a USB stick, we could support any case to setup a device unattended.
closed 2018-10-25 by pvizeli