home-assistant/operating-system #183: First boot up issue caused by NTP server

milesking, 2018-08-28:
Hi guys, I'm in China. We can't access Google NTP Server (time1.google.com stuff). Since Raspberry PI's local time is not correct, SSL Certificate failed when downloading the image. Could you please change NTP server from Google to ntp.org stuff?
Thanks!

fliaping, 2018-09-25:
@milesking you can add a hosts record: `<ntp server ip> time1.google.com` , by the way I use aliyun ntp `203.107.6.88    time1.google.com`

mkh595, 2018-11-23:
@fliaping Where this can be added?

milesking, 2018-11-23:
You can run Raspbian with GUI，then install hassio，quite easy with script. Running well for 3 month till now.

closed 2018-11-23 by milesking

pvizeli, 2018-11-23:
You are also able to set now NTP with HassOS.

fliaping, 2018-11-24:
> @fliaping Where this can be added?

`vi /etc/hosts`

truglodite, 2019-09-06:
> You are also able to set now NTP with HassOS.

How?