home-assistant/operating-system #184: Notification on HassOS updates?

rpitera, 2018-08-29:
Would be great if the hass.io dashboard would notify you of updates like it does with HA and AddOns.  Or is there anyway to create a sensor to show the latest version so an automation could be set up to notify you?  I often find out when there is an update a long time after it is released as I don't always check the system section of the hass.io panel.

rpitera, 2019-04-16:
Well, six other users upvoted the comment so it seems I'm not alone in asking.  But feel free to close this if there is no intent to add this feature.  Just let us know one way or the other, OK?

reharmsen, 2019-04-24:
I would like such a feature

DeviousPenguin, 2019-05-09:
You may be able to set a sensor to check the releases page of github, I think there may even be an example of this somewhere on the HA website

rpitera, 2019-05-09:
@DeviousPenguin - What a great idea!  I've never really thought about the github sensor much as I am not an active dev.  But this provides a wealth of info and I should be able to parse whatever I want from it to get to the goal of notifications.

For anyone else interested, I simply added the following to my sensors config:

```
- platform: github
  access_token: !secret github_access_token
  repositories:
    - path: 'home-assistant/home-assistant'
    - path: 'home-assistant/hassos'
    - path: 'home-assistant/hassio'
```

after following the [setup instructions on the component page](https://www.home-assistant.io/components/github/).

Thanks for a great alternative!