home-assistant/operating-system #187: Asus Tinkerboard S or Odroid C2 with eMMC for hassio/hassos?

felixstorm, 2018-09-02:
I would like to migrate my hassio/hassos from RPi 3 to a similar platform that is still future-proof with regards to hassio/hassos, but with a bit more (especially disk) performance, so has anyone a recommendation (maybe even @pvizeli) on whether I should buy an Asus Tinkerboard S or an Odroid C2 with eMMC module (or maybe even something completely different in a similar price range) for that purpose?

Thanks,
Felix

felixstorm, 2018-09-03:
Sorry for being a bit off topic here (in the community forums most topics about Odroid do not yet target hassos but still resionOS or manual installations) and I know that there is no scientifically correct answer here - I was just hoping for personal opinions from people that have worked with several options...

billchurch, 2018-12-20:
Any word on how you can implement eMMC on an existing HassIO on a Tinkerboard S?

rbray89, 2018-12-21:
@billchurch It should be in the latest pre-release: https://github.com/home-assistant/hassos/releases/tag/2.4

billchurch, 2018-12-21:
@rbray89 gotcha, so it's just a re-flash? nothing special to utilize the emmc?

rbray89, 2018-12-21:
@billchurch Yeah, just treat is as an SD card. The bootloader script will take care of the rest.

billchurch, 2018-12-21:
awesome, thanks!

felixstorm, 2019-01-02:
@rbray89 Thanks, good to know.

After the governor had been fixed, the performance of my RPi became *a lot* better, so I did not do any migration yet.

closed 2019-01-02 by felixstorm