home-assistant/operating-system #189: OS update fails

RealDeemer, 2018-09-03:
After updating from the frontend (0.77) the frontend does not come up again. Is this anything that can be fixed or where I can provide info to the project. Otherwise I might go ahead with a backup.

`core-ssh:~# hassio ha start
ERROR
panic: interface conversion: interface is nil, not string

goroutine 1 [running]:
panic(0x2c7050, 0x107be3a0)
	/home/travis/.gimme/versions/go1.7.6.linux.amd64/src/runtime/panic.go:500 +0x33c
github.com/home-assistant/hassio-cli/command/helpers.DisplayOutput(0x107d4200, 0x24, 0x200, 0x0)
	/home/travis/gopath/src/github.com/home-assistant/hassio-cli/command/helpers/common.go:118 +0x3c8
github.com/home-assistant/hassio-cli/command/helpers.ExecCommand(0x2f6f83, 0xd, 0x7ec7be58, 0x5, 0x0, 0x0, 0x76500, 0x0, 0x0, 0x0, ...)
	/home/travis/gopath/src/github.com/home-assistant/hassio-cli/command/helpers/common.go:155 +0x308
github.com/home-assistant/hassio-cli/command.CmdHomeassistant(0x10778420)
	/home/travis/gopath/src/github.com/home-assistant/hassio-cli/command/homeassistant.go:48 +0x540
github.com/urfave/cli.HandleAction(0x2af768, 0x324b14, 0x10778420, 0x0, 0x0)
	/home/travis/gopath/src/github.com/urfave/cli/app.go:503 +0xf0
github.com/urfave/cli.Command.Run(0x2f6f83, 0xd, 0x0, 0x0, 0x40bd10, 0x1, 0x1, 0x301e4d, 0x2f, 0x0, ...)
	/home/travis/gopath/src/github.com/urfave/cli/command.go:165 +0x6b8
github.com/urfave/cli.(*App).Run(0x107648c0, 0x1070a140, 0x3, 0x4, 0x0, 0x0)
	/home/travis/gopath/src/github.com/urfave/cli/app.go:259 +0x8b0
main.main()
	/home/travis/gopath/src/github.com/home-assistant/hassio-cli/main.go:22 +0x108
core-ssh:~# 
`

rbray89, 2018-09-03:
You'll need to provide what platform you're using.

RealDeemer, 2018-09-04:
Sure, this is a HassOs on a Pi.

RealDeemer, 2018-09-05:
There is no auth: line in my configuration.yaml.

The only special I can share about authentication, ist that I've upgraded to 0.77 a day ago which enabled the new authentication system. But that system did work fine before I've started the OS update.

cogneato, 2018-09-05:
You don't want any auth providers in there either. You could try deleting the files in .storage in /config, but leave the core.entity_registry and core.config_entries if they exist.

mcinnes01, 2018-09-06:
I've been experiencing this issue over a number of versions of hassio, I'm stuck on 1.7 of hassos. Whenever I click update, it goes through the motions, downloading version 1.10, says it's installing, reboots, then says found version 1.7 of hassos.

meichthys, 2019-01-27:
@cogneato  Does Hassos not support any authentication? I'm having similar issues updating like @mcinnes01  was having. Eventually the frontend does seem to come back up, but i can't seem to find any logs to identify why it took so long.

closed 2019-04-23 by stale[bot]