home-assistant/operating-system #191: No optimal performance on Raspberry Pi due to governor "powersafe"

tyborall, 2018-09-07:
The cpufreq governor on HassOS is set to "powersafe" which on a Raspberry Pi 3B+ leads to a cpu frequency of always 600 MHz. It will never use the higher speeds. On other Raspberry Pi version it is similar.

I changed this to governor "ondemand" for testing purposes and after that the boot time speeded up and even there are much lesser warnings about platforms which take longer than 10 seconds to start. My Raspberry Pi 3B+ then dynamically uses frequency from 600 Mhz to 1400 Mhz.

Joeboyc2, 2018-09-07:
Hey @tyborall,

How did you cange the setting please, id love to be able to update that setting becuase i constantly get the startup time warnings and reboots and upgrades seem to take ages

tyborall, 2018-09-07:
I recompiled and switched the kernel, but you can just try to add this to your /mnt/boot/config.txt (on development console):

`force_turbo=1`

But this will set waranty bit on Raspberry. I never had problems with overclocking.
See also here: https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

Joeboyc2, 2018-09-07:
Hey @tyborall,
When you say "add this to your /mnt/boot/config.txt (on development console):" how do you mean
Like via a putty session?
or from somewhere in the hassio interface?

tyborall, 2018-09-07:
You need access to development console if on Hassio: https://developers.home-assistant.io/docs/en/hassio_debugging.html

tyborall, 2018-09-07:
I also tested on stock HassOS kernel with "force_turbo=1" in config.txt. It will then stay on 1400 MHz (Raspberry Pi 3B+) and will never throttle down and governor still is set to "powersafe".
@pvizeli I suggest to switch to "ondemand" governor on Raspberry Pi platform, which is also default on Raspbian (don't know since when, but they switched that some time ago).

Joeboyc2, 2018-09-07:
Ah so you did mean via something like putty, I already had that access setup so have been able to add that string to the config file.

Thanks for your help

tyborall, 2018-09-09:
This issue has nothing to do with overclocking the RPi. This is not what I want to suggest. I just want to point out that Hassio HassOS has governor "powersafe" as default which holds the RPi always at the lowest default cpu frequency (e.g. on RPi 3B+ it is 600 MHz and on RPi 2 it is 700 MHz). This is a performance issue. If this is switched to default "ondemand" then it will dynamically switch to higher frequency if needed and falls back to lowest frequency if the power is not needed. This will do no harm to the hardware...

pvizeli, 2018-09-10:
will do that 👍

tyborall, 2018-09-10:
Don't know why, but on my Raspbian systems there is ondemand governor used. Double checked right now.

pvizeli, 2018-09-10:
Is there a cpufreq package installed?

pvizeli, 2018-09-10:
We should it now also selected as default. I read that the rpiconfig change that to `ondemand` but we have no rpiconfig on hassos.

pvizeli, 2018-09-10:
> The latest kernel has a cpufreq kernel driver with the "ondemand" governor enabled by default. It has no effect if you have no overclock settings. But when you do, the arm frequency will vary with processor load. Non default values are only used when needed according to the used governor. You can adjust the minimum values with the *_min config options or disable dynamic clocking with force_turbo=1

That is strange. The min/max should be different and in that case the ondemand should be go up/down, I need try it out on evening.

tyborall, 2018-09-10:
But when I change it to ondemand on HassOS and check current frequency when idle it says 600 MHz an when I then e.g. start addon installation it jumps to 1400 MHz.

tyborall, 2018-09-10:
And this is default on HassOS:

```
~ # cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq
600000
~ # cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq
1400000

```

pvizeli, 2018-09-11:
perfectly