home-assistant/operating-system #198: rpi-rf: failed to add edge detection to GPIO

dhzl84, 2018-09-17:
I want to use (again) the [RPi-RF Receiver AddOn from pantomax](https://github.com/pantomax/hassio-addons/tree/master/rpi-rf-receiver) with HassOS but it fails to start.

`Using base prefix '/usr'
New python executable in /data/venv/bin/python3
Also creating executable in /data/venv/bin/python
Installing setuptools, pip, wheel...done.
Running virtualenv with interpreter /usr/bin/python3
Collecting rpi-rf
  Downloading https://files.pythonhosted.org/packages/d9/a2/e4d43f773ce20a6950f6fd799cb1ede7feb679fd1c863f0360b24e7ebe97/rpi-rf-0.9.6.tar.gz
Collecting RPi.GPIO (from rpi-rf)
  Downloading https://files.pythonhosted.org/packages/e2/58/6e1b775606da6439fa3fd1550e7f714ac62aa75e162eed29dbec684ecb3e/RPi.GPIO-0.6.3.tar.gz
Building wheels for collected packages: rpi-rf, RPi.GPIO
  Running setup.py bdist_wheel for rpi-rf: started
  Running setup.py bdist_wheel for rpi-rf: finished with status 'done'
  Stored in directory: /root/.cache/pip/wheels/f8/f7/17/c4c4b5b048fd5320952b5b1e1f83cbb6378b51fb4319fbf469
  Running setup.py bdist_wheel for RPi.GPIO: started
  Running setup.py bdist_wheel for RPi.GPIO: finished with status 'done'
  Stored in directory: /root/.cache/pip/wheels/f7/79/24/a9af41176b06678ebe1c1c2e25ecdd43fdddde1a9368aae9f4
Successfully built rpi-rf RPi.GPIO
Installing collected packages: RPi.GPIO, rpi-rf
Successfully installed RPi.GPIO-0.6.3 rpi-rf-0.9.6
/data/venv/lib/python3.6/site-packages/rpi_rf/rpi_rf.py:167: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.gpio, GPIO.IN)
Traceback (most recent call last):
  File "/share/rpi-rf_receive.py", line 34, in <module>
    rfdevice.enable_rx()
  File "/data/venv/lib/python3.6/site-packages/rpi_rf/rpi_rf.py", line 168, in enable_rx
    GPIO.add_event_detect(self.gpio, GPIO.BOTH)
RuntimeError: Failed to add edge detection`

The AddOn ran flawless with ResinOS and immediately stopped working after migration to HassOS.
I saw some issues for that in the past (closed now) but even HassOS 1.10 didn't fix the problem.
I tried the 32 bit and 64 bit image, both raise the same error.

karfam, 2018-09-26:
Same here Home Assistant 0.78.3

dhzl84, 2018-10-21:
Still not working with HassOS 1.11, 32 Bit, HA 80.3, RPi 3B

dhzl84, 2018-10-29:
solved by https://github.com/fryguy04/hassio-addons.git

closed 2018-10-29 by dhzl84

riker09, 2018-11-29:
@dhzl84 How can I use that addon to *send* codes? This is only for receiving...

dhzl84, 2018-11-30:
@riker09 As the name suggest the AddOn is only made for receiving 433 MHz codes and provide them as a sensor.

For sending I just define switches like that:
`  - platform: rpi_rf
    gpio: 17
    switches:
      funksteckdose_aussen:
        code_on: 4433
        code_off: 4436`

You need to have a 433 MHz transmitter attached to one of the GPIOs.

riker09, 2018-11-30:
I'm using HassOS (not Hass.io) and I think there's a known bug that keeps me from using the GPIO of the Pi. I was under the impression that your addon was developed exactly because of that limitation?

Since the documentation is scarce and I didn't get any help in the discussion channel yesterday I was hoping for some insights here. My question is: How can I use the GPIO under HassOS? Simply adding configuration like this won't work:

```yaml
switch:
  - platform: rpi_rf
    gpio: 17
    switches:
      bedroom_light:
        code_on: 1234567
        code_off: 1234568
```

This will result in an error in the log file saying that `rf_rpi` is not supported. Here's an unhelpful discussion regarding the topic: https://community.home-assistant.io/t/is-gpio-broken/61051

dhzl84, 2018-12-03:
Maybe ask fryguy04 since he fixed this issue for the AddOn.

riker09, 2018-12-03:
Sometimes it is better to move on. I took the opportunity and removed the 433MHz sender and receiver from my RPi GPIO and hooked them up to a NodeMCU (ESP8622) board. Receiving the codes transmitted by my remotes already works, sending still has some issues I have to iron out. But I'm getting better at C along the way, so yay! :slightly_smiling_face: