home-assistant/operating-system #201: netdisco fails

welbo97, 2018-09-21:
New install hassos 1.10 on rpi 3B+. Supervisor 131. 

2018-09-21 13:56:30 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved Traceback (most recent call last): 
File "/usr/local/lib/python3.6/site-packages/homeassistant/components/discovery.py", line 168, in scan_devices results = await hass.async_add_job(_discover, netdisco) 
File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run result = self.fn(*self.args, **self.kwargs) 
File "/usr/local/lib/python3.6/site-packages/homeassistant/components/discovery.py", line 194, in _discover netdisco.scan() 
File "/usr/local/lib/python3.6/site-packages/netdisco/discovery.py", line 57, in scan self.ssdp.scan() 
File "/usr/local/lib/python3.6/site-packages/netdisco/ssdp.py", line 41, in scan self.update() 
File "/usr/local/lib/python3.6/site-packages/netdisco/ssdp.py", line 89, in update entry for entry in scan() File "/usr/local/lib/python3.6/site-packages/netdisco/ssdp.py", line 225, in scan for addr in zeroconf.get_all_addresses(): TypeError: get_all_addresses() missing 1 required positional argument: 'address_family'

Is this associated with the zeroconf dependency that was updated in 78.2?

ludeeus, 2018-09-21:
That issue should be fixed with 0.78.2

ludeeus, 2018-09-21:
You should wait for it to appear.
<details>
  <summary>But if you really need it now...</summary>
  hassio ha update -o version=0.78.2
</details>

welbo97, 2018-09-21:
Thanks @ludeeus. You are a very useful engine.

closed 2019-04-23 by stale[bot]