home-assistant/operating-system #206: time incorrect at first boot (hassos on esxi)

pergolafabio, 2018-09-30:
hi, my time is correct in bios.-/esxi, but at first start of hassos, the time is  hour ahead, since i live in ETA +1 zone

so my sensors are not working for the first hour :), so when i do a clean boot, i need to restart hass when its loaded, afterwards it has the correct time

thnx

highground88, 2018-10-25:
I think it's becuase HassOS seems to use UTP time, regardless of your time zone. 

I personally hope this gets changed, otherwise I'll need to move back to another OS.

pergolafabio, 2018-10-25:
well, updated newer build, still same issue
but i dont care, i know when i do a cold reboot, i need to restart HA from the configuration one time after first start

closed 2019-03-28 by pergolafabio