home-assistant/operating-system #211: 802.1q support

efficiosoft, 2018-10-11:
Hi,

I recently hooked up HassOS with KVM on my hypervisor. Since I've different VLAN's that Hass should work with, I set up two new network manager profiles for these, using the single virtio link as parent.

However, it seems that the 8021q kernel module is missing. Would it be possible to include that in future builds? For what it's worth, that's something really useful.

Personally, I can just add another virtio link to the VM and couple network manager profiles to their MACs, but devices with only a single physical NIC don't have that option.

Best regards
Robert

pvizeli, 2018-10-11:
Feel free to make a PR 👍

efficiosoft, 2018-10-11:
@pvizeli Ok, do you have any hint on where to start with this? I'm absolutely unfamiliar with the HassOS buildroot architecture... Wouldn't it be enough to activate the building of ``8021q`` somewhere globally for all architectures? Don't know whether userland software is missing as well, but since ``ip link`` says ``Operation not supported``, I think the only missing part is the kernel module.

rbray89, 2018-10-11:
@efficiosoft You can get started by taking a look in `buildroot-external/configs` and taking a look at one of the configs to see where things are being configured from. Among other things, the kernel config/version/patches are specified here. If you want to enable this for all platforms, you'll need to update several kernel configs.

jtru, 2018-10-11:
@efficiosoft due to somebody else's request, I wrote https://github.com/jtru/hassos/blob/getting-started-with-devel-guide/Documentation/getting_started_development.md yesterday. Maybe you find it helpful, too :) That, and knowing a bit about how the Linux' config system works, should be enough to get you started.

pvizeli, 2018-10-11:
Yeah like @jtru have done it is fine. I will split this kernel config into a generic HassOS and linux version part later before next release come out.

efficiosoft, 2018-10-12:
@jtru Thanks for that howto, found it to be very helpful!

I added the module to ``buildroot-external/kernel/hassos-4_14.config`` and it's building currently... We'll see how long that takes :)

jtru, 2018-10-12:
You're welcome :)

We arrived at the same conclusion, I think - I built tha OVA image with `CONFIG_VLAN_8021Q=y` set in the file you mentioned, and could successfully create a VLAN device wit htat kernel booted.

efficiosoft, 2018-10-12:
Ok, finally got the build run through. Creating VLAN interfaces works.

PR #216 (2018-10-13): Added 8021q kernel module for VLAN support
Closes #211
closed 2018-10-13 by pvizeli