home-assistant/operating-system #219: Doc how to login to base system lacking

scop, 2018-10-15:
https://github.com/home-assistant/hassos/blob/dev/Documentation/network.md#using-nmcli-to-set-a-static-ipv4-address

These instructions start with "Log into the HASSOS base system via a console". But that's not quite enough -- where does one connect the console to? ssh? Which port? My hassos doesn't seem to have anything but port 10010 open. Does one need to install the authorized_keys file as described in https://developers.home-assistant.io/docs/en/hassio_debugging.html#hassos-based-hassio ?

TonyApuzzo, 2019-01-02:
I think you have 2 options (I'm just a semi-clueless user, so take this with a grain of salt.)

1. Use the `authorized_keys` as you described via external USB device
2. Plug in an HDMI monitor and a USB keyboard and log in directly using the physical console.

I have not tried 2., but I use 1. all the time.

vincegre, 2019-02-03:
Succeeded to get access at host system with login command ? Login command on my system is doing absolutely nothing unhappy :(

TonyApuzzo, 2019-02-03:
If you are logging on to physical console via monitor/keyboard or serial console via TTY (depends on your base device as to what is available) you can just enter the username "root" and no password.  It logs you into an extremely simple interface that runs the hassio cmd in a loop.  If you then enter `login` at the hassio prompt then you can enter Linux shell commands to poke around.

Example from a serial console:

```sh
Welcome to HassOS
hassio login: root
  _    _                 _
 | |  | |               (_)
 | |__| | __ _ ___ ___   _  ___
 |  __  |/ _` / __/ __| | |/ _ \
 | |  | | (_| \__ \__ \_| | (_) |
 |_|  |_|\__,_|___/___(_)_|\___/

Welcome on Hass.io CLI for HassOS.

For more details use 'help' and 'exit' to close.
If you need access to host system use 'login'.

hassio > ha info
boot: true
custom: false
image: homeassistant/odroid-xu-homeassistant
last_version: 0.87.0b2
machine: odroid-xu
port: 8123
ssl: false
version: 0.86.4
wait_boot: 600
watchdog: true

hassio > login
# uptime
 18:27:46 up 11:13,  load average: 0.00, 0.04, 0.00
```

scop, 2019-04-22:
I believe this is still not addressed.

scop, 2019-06-22:
Still not addressed.

AtticusG3, 2019-06-24:
still not documented

DubhAd, 2019-06-30:
*console* means the *physical console* - you plug in a monitor and keyboard.

TonyApuzzo, 2019-07-04:
I am willing to make some documentation updates and request that the core team integrate them, but I don't really understand how to address the lack mentioned by @scop and @AtticusG3.  There is documentation although it is quite terse.

Have you tried the suggestions in this thread and been able to login?

vincegre, 2019-07-10:
> I am willing to make some documentation updates and request that the core team integrate them, but I don't really understand how to address the lack mentioned by @scop and @AtticusG3. There is documentation although it is quite terse.

Very good idea as at time I had this problem there was none documentation or I was unable to find it :(

> Have you tried the suggestions in this thread and been able to login?

You can imagine it's a long time I have reinstalled hass.io from scratch as I could never sort out the problem at that time !

closed 2019-09-15 by stale[bot]

sammcj, 2024-02-26:
I still don't think this has been addressed, it's not clear how to access the host OS on devices such as the home assistant green. This seems to be required to do basic things such as partitioning SD cards etc....