home-assistant/operating-system #223: Changing the permissions of created devices

sholzmayer, 2018-10-19:
I always get the following error in the docker container of my hassIO addon: 
`Can't open /dev/ttyACM0: Permission denied`

Changing the permissions of this device in hassOS solves this problem: `sudo chmod a+rw /dev/ttyACM0`. But after a reboot or replacement of the usb-dongle those permissions are gone.

Any idea how to define the permissions for newly created devices (`/dev/xyz`) in hassOS?

Thanks!

rbray89, 2018-10-19:
Closing. This is an option in the hassio addon config (devices)
https://developers.home-assistant.io/docs/en/hassio_addon_config.html

closed 2018-10-19 by rbray89

sholzmayer, 2018-10-19:
@rbray89  I know this option. But this option doesn't solve the problem on the hassOS device. I also have to change the permissions in the host system - because there are the wrong permissions on this device... It doesn't fix the issue when the correct permissions are set in the addon.

rbray89, 2018-10-19:
What are the full permissions of the device both inside and outside of the addon?
How are you accessing hassOS? native SSH, native shell (keyboard/mouse), or ssh addon?