home-assistant/operating-system #224: Cannot connect to host 172.30.32.1:8123 (Fails to boot sometimes - rare)

Fusseldieb, 2018-10-21:
Hi, I've been experiencing this bug since I first knew Home Assistant, so I'm not so sure if this can be resolved in HassOS, but following does happen sometimes when I restart Home Assistant:

Sometimes when I restart Home Assistant, it doesn't come up again. The page says Connection Refused and that's it. The first time this happened to me I almost freaked out, but now I know that a simple router restart fixes it. It's strange, Today it happened to me again and I wanted some concrete proof of what is happening so that I can submit this here.

So, the SSH Server still runs normally during this time, so there it is where it errored out:

```
core-ssh:~# hassio su logs
Error decoding json invalid character '\x1b' looking for beginning of value: 18-10-21 06:26:39 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
18-10-21 06:56:39 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
18-10-21 07:26:39 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
18-10-21 07:56:39 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
18-10-21 08:26:39 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
18-10-21 08:43:56 INFO (SyncWorker_9) [hassio.docker.interface] Stop homeassistant/armhf-addon-mosquitto Docker application
18-10-21 08:43:57 INFO (SyncWorker_9) [hassio.docker.interface] Clean homeassistant/armhf-addon-mosquitto Docker application
18-10-21 08:44:00 INFO (SyncWorker_19) [hassio.docker.addon] Start Docker add-on homeassistant/armhf-addon-mosquitto with version 2.0
18-10-21 08:44:34 INFO (MainThread) [hassio.api.proxy] /homeassistant/api/services access from core_configurator
18-10-21 08:44:34 INFO (MainThread) [hassio.api.proxy] /homeassistant/api/events access from core_configurator
18-10-21 08:44:34 INFO (MainThread) [hassio.api.proxy] /homeassistant/api/states access from core_configurator
18-10-21 08:56:39 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
18-10-21 09:29:43 INFO (MainThread) [hassio.api.proxy] /homeassistant/api/services/homeassistant/restart access from core_configurator
18-10-21 09:29:43 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
18-10-21 09:29:43 INFO (SyncWorker_15) [hassio.docker] Run command 'python3 -m homeassistant -c /config --script check_config' on homeassistant/raspberrypi3-homeassistant
18-10-21 09:30:30 INFO (MainThread) [hassio.homeassistant] Home Assistant config is valid
18-10-21 09:30:30 INFO (SyncWorker_6) [hassio.docker.interface] Stop homeassistant/raspberrypi3-homeassistant Docker application
18-10-21 09:31:02 INFO (SyncWorker_6) [hassio.docker.interface] Clean homeassistant/raspberrypi3-homeassistant Docker application
18-10-21 09:31:03 INFO (SyncWorker_8) [hassio.docker.homeassistant] Start homeassistant homeassistant/raspberrypi3-homeassistant with version 0.80.3
18-10-21 09:32:50 INFO (MainThread) [hassio.api.proxy] /homeassistant/api/services access from core_configurator
18-10-21 09:32:50 ERROR (MainThread) [hassio.homeassistant] Error on call http://172.30.32.1:8123/api/services: Cannot connect to host 172.30.32.1:8123 ssl:None [Connection refused]
18-10-21 09:32:50 ERROR (MainThread) [hassio.api.proxy] Error on API for request services
18-10-21 09:35:01 INFO (MainThread) [hassio.api.security] /homeassistant/logs access from core_ssh
18-10-21 09:35:13 INFO (MainThread) [hassio.api.security] /supervisor/logs access from core_ssh

```
Note: A supervisor restart doesn't resolve it, I need to restart the whole router and then restart Home Assistant (again).

janningnetworks, 2018-11-04:
#metoo ;) Did you solve this issue?

Fusseldieb, 2018-11-05:
@janningnetworks No, it's quite rare to occurr, so I don't care that much, but I still put this here, so the devs can eventually fix this.

thakkarrahul, 2018-11-06:
Hi,

I am having similar issues and have tried the above method but still the same. I am on HASSIO version 0.81.6 

Here is the link to forum page where I have explained the issue.

https://community.home-assistant.io/t/hassio-no-web-interface/77208

If you have any solution, please help.

Fusseldieb, 2019-01-10:
Bump. Still happening on 0.84.

magin1, 2019-01-12:
Bump. Still happening on 0.85.

TheZoker, 2019-01-15:
I have the same issue

SP410, 2019-01-15:
FYI, I have the issue in running HassIO in Docker on Ubuntu 18.04.1.  For me it occurs after performing a snapshot restore...

arissaraiva, 2019-01-15:
Hi same issue here openned another thread, I can´t access spotfy, darksky nd cloud all give me API error

bratanon, 2019-01-17:
Same for me since 0.85. Running  the default Hassio on rpi3

bratanon, 2019-01-17:
and same with 0.85.1

Fusseldieb, 2019-01-18:
**Quick update on my findings:**
If this happens, reconnecting the LAN interface on my router and then doing a "hassio su reload" over SSH makes it work again, so my assumption is right. 
`this can be fixed by disabling and re-enabling the ethenet port on the RPi, if that specific error happens, causing the RPi to refresh it's IP, DHCP and whatnot. Would be great if someone could implement this on a newer build.`

ahnernoch, 2019-01-20:
Same here. Sometimes Web Interface just wont come up.

Obicom, 2019-02-01:
Same issue with new beta 0.87.0b0

1iveowl, 2019-02-12:
Same issue w/ 0.87.1

PreferHardware, 2019-02-18:
Same here, hassio 8.7.1 on NUC5PPYH with duckdns and ssl:

[homeassistant.components.hassio.handler] Client error on /supervisor/ping request Cannot connect to host 172.30.32.2:80 ssl:None [Connection refused]

The problems started as soon as I moved to ssl, duckdns, etc.  No amount of messing around with yaml files made any difference, but a cold reboot magically cleared the errors.

PreferHardware, 2019-02-18:
This behaviour could also, repeatably, be triggered by including a discovered media_player (samsungtv) in the yaml.  Commenting out the line and rebooting (not simply restarting the containers) was the solution.

h4nc, 2019-02-19:
Which line exactly do you mean? Ignore it in autodiscovery?

```
discovery:
  ignore:
```

PreferHardware, 2019-02-19:
I tried adding the Samsung TV to a "Sitting Room" group by simply adding the line:

  - media_player.television

in groups.yaml.  This was the line that caused problems and was the line that I commented out before rebooting.

The process of discovery does not seem to be the problem.  The problem only comes when the discovered entity is actually introduced into the yaml.

h4nc, 2019-02-19:
Ok, I don't have my media_player's in a group.
I'm not sure we have the same issue. 

I also noticed that the samsung media player entities are not reliable. Sometimes they show up sometimes not. 
I think you onle get them when the TV is on when you start home assisant.

So if you restart ha and the tv is off, the entity does not show (I'm not a 100% sure about that, buts that'S my guess).

radionerd1, 2019-03-15:
This is driving me nuts. Why is it trying an IP that is not on my network?
`ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False `

I have scene this for months now, I search forums with no fixes. My private network is 192.168.1.0/24 WTF???

radionerd1, 2019-03-15:
Looks for invalid IP 172.30.32.1:8123, Won't update, rolls back. 
```

19-03-15 04:42:42 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-03-15 04:42:42 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-03-15 04:42:42 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-03-15 04:42:46 WARNING (MainThread) [hassio.homeassistant] Don't wait anymore of Home Assistant startup!
19-03-15 04:42:46 CRITICAL (MainThread) [hassio.homeassistant] HomeAssistant update fails -> rollback!
19-03-15 04:42:46 INFO (MainThread) [hassio.homeassistant] Update Home Assistant to version 0.88.2
19-03-15 04:42:46 INFO (SyncWorker_12) [hassio.docker.interface] Update Docker homeassistant/raspberrypi3-homeassistant:0.89.2 to homeassistant/raspberrypi3-homeassistant:0.88.2
19-03-15 04:42:46 INFO (SyncWorker_12) [hassio.docker.interface] Pull image homeassistant/raspberrypi3-homeassistant tag 0.88.2.
```

TheZoker, 2019-03-15:
@radionerd1 The IP 172.30.32.1 belongs to docker:
https://community.home-assistant.io/t/mqtt-why-172-30-32-1-gets-connected/46513/2?u=zoker

tyborall, 2019-04-29:
Just add

```
  auth_providers:
    - type: trusted_networks
      trusted_networks:
        - 172.16.0.0/12
```

under `homeassistant:`

in your configuration.yaml...

DeviousPenguin, 2019-05-09:
> Just add
> 
> ```
>   auth_providers:
>     - type: trusted_networks
>       trusted_networks:
>         - 172.16.0.0/12
> ```
> 
> under `homeassistant:`
> 
> in your configuration.yaml...
> 
> @Developers: I think this should be a default in configuration.

Will this disable authentication at all? I have my HomeAssistant port exposed to the internet, and I'd like to keep auth enabled.

tyborall, 2019-05-09:
Ah sorry, if you use nginx, yes it will. Then just allow that particular host.

DeviousPenguin, 2019-05-11:
Ok just wanted to be sure before tinkering with the settings. 

I've got it set under `trusted_proxies:` instead and and it seems to be working OK with NGINX + authentication.

rover6s, 2019-05-12:
I had the same issue and removed all the Add-ons that I had added while learning HA. I had Configurator, Samba, and APPDaemon. 

After removing the Add-ons I rebooted the RPi from the Hass.io-->System-->Host System menu and the error is gone. I have added the Configurator back and still no error in the System log. I really need the AppDaemon for the HADashboard so I'll add that back and see what happens.

closed 2019-07-20 by stale[bot]

cotwild, 2019-07-23:
After update to 0.96.3 I cannot access any dev-state, dev-info, etc. - the "developer-tools" link is new but doesn't work... So this is my System log under Hass.io -> System:

`19-07-23 06:44:02 ERROR (MainThread) [hassio.homeassistant] Error on call http://172.30.32.1:8123/api/states/sensor.printer_status: Cannot connect to host 172.30.32.1:8123 ssl:None [Connection refused]
19-07-23 06:44:02 ERROR (MainThread) [hassio.api.proxy] Error on API for request states/sensor.printer_status
19-07-23 06:44:03 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-07-23 06:44:03 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-07-23 06:44:03 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_appdaemon3
19-07-23 06:44:03 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-07-23 06:44:03 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-07-23 06:44:03 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-07-23 06:44:07 ERROR (MainThread) [hassio.homeassistant] Error on call http://172.30.32.1:8123/api/states/sensor.printer_status: Cannot connect to host 172.30.32.1:8123 ssl:None [Connection refused]
19-07-23 06:44:07 ERROR (MainThread) [hassio.api.proxy] Error on API for request states/sensor.printer_status
19-07-23 06:44:08 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-07-23 06:44:08 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-07-23 06:44:08 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-07-23 06:44:08 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_appdaemon3`

cotwild, 2019-07-23:
Stupid me.... Page refresh did the trick! sorry... how could if forget this and try to fix it for that long...

pilot1981, 2019-07-23:
I have hassio 0.96 with same issue...

Fusseldieb, 2019-07-23:
The stale bot is probably one of the most useless bots ever.
It closes the issue, essentially marking it as solved and therefore it won't appear in open issues anymore, making the devs think this is solved.
**But it isn't.** 
Can anyone open this back up, because this is clearly **not** solved!?

Chuchodavids, 2019-07-25:
I know this is an old post. But I had this problem with the last HA update. I found a solution:
The problem is caused by a conflict in the ports
1. Stop all your containers except Portainer and HA + HA supervisor
2. Check if that solved your problem. If so.. continue to point three.
3. Activate one of your Docker containers and check if the problem persist. 
4. Repeat number three until you find out which one is the container that is conflicting with HA.
5. When you find it, just stop the container and change the ports in which the container is running.
6. enjoy.

Ailme, 2019-11-07:
I have this issue. hassio 0.100.2 - 0.101.2

Antraciten, 2019-11-10:
I have the same issue. Hassio 0.101.3

pilot1981, 2019-11-10:
Also me, during reboot action

rant989, 2019-11-18:
I have the same issue. Hassio 0.101.3

sebba12, 2019-11-18:
I have the same issue. Hassio 0.101.3 and not in any versions before.

gdreelin, 2019-11-20:
Yep same here just started on version 0.101.3 never seen it before that.

PietroMB, 2019-11-21:
I have same issue. Hassio 0.100.2

RichardUUU, 2019-11-25:
I have it in Hassio 102.2. Running as a VM with no dockers.

dubbfredda, 2019-11-26:
I also have the same problem

dubbfredda, 2019-11-26:
First I could not update Hassio and when I tried to update a second time it never come back up again after the update. The Homeassistant container is started but can't get any connection with the Hassio frontend. Get following error
19-11-26 20:34:25 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].

MalagaInternet, 2019-11-28:
+1 Hassio 102.2

thomasfals, 2019-11-30:
+1 Hassio 0.102.3

matteos1, 2019-11-30:
i have the same problem
19-11-30 21:47:34 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:47:39 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:47:39 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:47:39 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:47:44 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:47:44 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:47:44 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:47:49 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:47:49 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:47:49 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:47:54 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:47:54 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:47:54 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:47:59 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:47:59 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:47:59 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:04 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:04 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:04 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:09 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:09 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:09 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:14 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:14 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:14 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:19 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:19 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:19 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:24 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:24 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:24 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:29 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:29 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:29 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:34 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:34 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:34 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:39 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:39 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:39 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:44 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:44 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:48:44 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:48:49 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:48:49 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
[... cut ...]
19-11-30 21:49:50 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
19-11-30 21:49:55 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
19-11-30 21:49:55 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
19-11-30 21:49:55 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].

anklimov, 2019-12-08:
Is some progress to fix the issue? I have same. And have no idea how to recover it, except moving back to OpenHab

pricard32, 2019-12-10:
This is the internal IP of your docker container.
I run hassio on a Ubuntu server, but every Hassio installation use docker.
Docker need two network, an external (your lan) and a internal (172.30.32.0/23).

I'm not sure why we all seeing this, maybe a docker expert can help us.
But in my particular case, exept seeing this, theres no error on my installation.

bschatzow, 2019-12-13:
I have this issue when I updated to .103 in Hassio

mpgoebel, 2019-12-13:
I just updated to .103 and am also seeing this.  I haven't seen any issues with anything not working, just this error.

caiocaminoski, 2019-12-13:
I have just updated to v103 as well and I am also getting this error...

sivastola, 2019-12-15:
same here with 103. 
Is it possible this is related to the information about python on the ReleaseNotes page? It says support for python3.6 is deprecated and we should move to python3.7
I'm running HA on unbuntu 18.04 and here python3.6 is default.
I don't like to change the default packages normally.

pricard32, 2019-12-16:
I've update python to 3.7 and the error disapear...!

soylant, 2020-01-01:
also updated to 103  and having the same problem

fillilutten, 2020-01-02:
Having the same problem running 103. Everything is running just fine though.

blamaz, 2020-01-07:
Same issue here on 0.103.6. I'm having this error since a long time but everything seem to be running ok so I ignored it. Still hope someone figure it out.

JANogueira, 2020-01-15:
Same  here with HassOS3.8 + Home Assistant 0.103.6 and the errors are always having node red related messages:
```
20-01-15 14:30:06 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-15 14:30:06 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-15 14:30:06 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-15 14:30:11 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-15 14:30:11 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-15 14:30:11 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-15 14:30:14 INFO (MainThread) [hassio.homeassistant] Home Assistant pip installation done
20-01-15 14:30:16 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-15 14:30:16 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-15 14:30:16 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-15 14:30:21 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-15 14:30:21 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-15 14:30:21 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-15 14:30:26 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-15 14:30:26 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-15 14:30:26 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-15 14:30:31 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-15 14:30:31 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
```

mattclar, 2020-01-21:
same problem here on 0.104.2

kic883, 2020-01-22:
Same here 

My log 
20-01-22 02:57:23 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:23 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:57:28 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:57:28 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:28 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:57:33 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:57:33 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:33 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:57:38 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:57:38 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:38 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:57:40 INFO (MainThread) [hassio.homeassistant] Home Assistant pip installation done
20-01-22 02:57:43 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:57:43 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:43 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:57:45 INFO (MainThread) [hassio.homeassistant] Home Assistant pip installation in progress
20-01-22 02:57:48 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:57:48 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:48 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:57:53 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:57:53 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:53 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:57:58 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:57:58 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:57:58 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:58:03 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:58:03 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:58:03 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:58:08 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:58:08 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:58:08 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:58:13 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:58:13 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:58:13 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:58:18 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:58:18 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:58:18 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:58:23 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:58:23 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:58:23 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:58:28 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:58:28 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:58:28 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:58:33 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:58:33 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:58:33 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
[... cut ...]
20-01-22 02:59:33 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize
20-01-22 02:59:33 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_nodered
20-01-22 02:59:33 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].
20-01-22 02:59:35 INFO (MainThread) [hassio.homeassistant] Detect a running Home Assistant instance
20-01-22 02:59:35 INFO (MainThread) [hassio.homeassistant] Successful run Home Assistant 0.104.3

bschatzow, 2020-01-22:
Saw it again on .104.2.  Had to reboot router to load HA.  Any setting changes to make to avoid this issue?

mattclar, 2020-01-25:
@pvizeli any chance we could reopen this issue?

[53 more replies]